lwfinger/rtw88 #89: Driver rtw88_8821ce is misbehaving

pantr19, 2022-06-14:
Hello,
I have opened this issue because I have observed abnormal behavior in driver rtw88_8821ce.
After an update to Linux kernel 5.18.3-arch1-1, I enabled rtw88_8821ce disabling open source dkms driver.
I driver is usable in this kernel release but I observed sudden drop in signal strength.
Example:Good --> weak --> Excellent (Indicated by Gnome system tray Wi-Fi icon)
The event occurred many times and sometimes the icon disappeared denoting Wi-Fi disconnected
or it denoted Wi-Fi connecting. These events lasted within 10 seconds.
Since I use hotspot from my mobile I observed multiple ghost devices connected.
The driver does not plays well with with Bluetooth.
I have since then ran my system for 10 mins multiple times and collected dmesg and journal data under four conditions.
Hotspot 2.4GHz Bluetooth Not Connected
Hotspot 2.4GHZ Bluetooth Connected
Hotsport 5GHz Bluetooth Not Connected
Hotspot 5GHz Bluetooth Connected
For each condition there are three Runs.

failed to get tx report from firmware
CTRL-EVENT-BEACON-LOSS

These two above messages, have popped up multiple times when there is some misbehavior.
I am attaching the required hardware information along with my observations.
Please read the Observation Section at last in both the files.

[rtw88_8821ce on 2.4GHz.txt](https://github.com/lwfinger/rtw88/files/8897464/rtw88_8821ce.on.2.4GHz.txt)
[rtw88_8821ce on 5GHz.txt](https://github.com/lwfinger/rtw88/files/8897468/rtw88_8821ce.on.5GHz.txt)
[Hardware Information.txt](https://github.com/lwfinger/rtw88/files/8897473/Hardware.Information.txt)

Thank You

pantr19, 2022-06-20:
Hey,
I Don't think it's a problem but since I have updated to kernel 5.18.5-arch-1-1 I have been seeing a new dbus message.
I suspect it might help

[Jun20 23:22] rtw_8821ce 0000:02:00.0: unhandled firmware c2h interrupt
[  +8.288923] rtw_8821ce 0000:02:00.0: unhandled firmware c2h interrupt
[Jun20 23:23] rtw_8821ce 0000:02:00.0: unhandled firmware c2h interrupt
[  +7.779504] rtw_8821ce 0000:02:00.0: unhandled firmware c2h interrupt
[  +0.409354] rtw_8821ce 0000:02:00.0: unhandled firmware c2h interrupt
[  +0.825306] rtw_8821ce 0000:02:00.0: unhandled firmware c2h interrupt
[ +23.139008] rtw_8821ce 0000:02:00.0: unhandled firmware c2h interrupt
[  +5.939126] rtw_8821ce 0000:02:00.0: unhandled firmware c2h interrupt
[  +3.072008] rtw_8821ce 0000:02:00.0: unhandled firmware c2h interrupt
[  +9.346021] rtw_8821ce 0000:02:00.0: unhandled firmware c2h interrupt
[Jun20 23:24] rtw_8821ce 0000:02:00.0: unhandled firmware c2h interrupt
[ +11.467496] rtw_8821ce 0000:02:00.0: unhandled firmware c2h interrupt
[  +5.116518] rtw_8821ce 0000:02:00.0: unhandled firmware c2h interrupt
[  +4.918572] rtw_8821ce 0000:02:00.0: unhandled firmware c2h interrupt
[  +2.253402] rtw_8821ce 0000:02:00.0: unhandled firmware c2h interrupt
[  +2.454552] rtw_8821ce 0000:02:00.0: unhandled firmware c2h interrupt
[  +1.228735] rtw_8821ce 0000:02:00.0: unhandled firmware c2h interrupt
[  +0.618133] rtw_8821ce 0000:02:00.0: unhandled firmware c2h interrupt
[  +2.664598] rtw_8821ce 0000:02:00.0: unhandled firmware c2h interrupt
[Jun20 23:25] rtw_8821ce 0000:02:00.0: unhandled firmware c2h interrupt
[  +3.479320] rtw_8821ce 0000:02:00.0: unhandled firmware c2h interrupt
[ +12.498029] rtw_8821ce 0000:02:00.0: unhandled firmware c2h interrupt
[ +10.646262] rtw_8821ce 0000:02:00.0: unhandled firmware c2h interrupt
[  +5.727877] rtw_8821ce 0000:02:00.0: unhandled firmware c2h interrupt
[ +14.751841] rtw_8821ce 0000:02:00.0: unhandled firmware c2h interrupt
[Jun20 23:26] rtw_8821ce 0000:02:00.0: unhandled firmware c2h interrupt
[  +3.681897] rtw_8821ce 0000:02:00.0: unhandled firmware c2h interrupt
[  +2.455483] rtw_8821ce 0000:02:00.0: unhandled firmware c2h interrupt
[  +4.304840] rtw_8821ce 0000:02:00.0: unhandled firmware c2h interrupt
[  +0.822251] rtw_8821ce 0000:02:00.0: unhandled firmware c2h interrupt
[ +10.649468] rtw_8821ce 0000:02:00.0: unhandled firmware c2h interrupt
[  +0.610316] rtw_8821ce 0000:02:00.0: unhandled firmware c2h interrupt
[  +0.006369] rtw_8821ce 0000:02:00.0: unhandled firmware c2h interrupt
[  +4.294407] rtw_8821ce 0000:02:00.0: unhandled firmware c2h interrupt
[  +2.359062] rtw_8821ce 0000:02:00.0: unhandled firmware c2h interrupt
[  +3.994852] rtw_8821ce 0000:02:00.0: unhandled firmware c2h interrupt
[  +6.143945] rtw_8821ce 0000:02:00.0: unhandled firmware c2h interrupt
[  +4.706167] rtw_8821ce 0000:02:00.0: unhandled firmware c2h interrupt
[Jun20 23:27] rtw_8821ce 0000:02:00.0: unhandled firmware c2h interrupt
[  +2.044526] rtw_8821ce 0000:02:00.0: unhandled firmware c2h interrupt
[  +0.412270] rtw_8821ce 0000:02:00.0: unhandled firmware c2h interrupt
[ +20.684040] rtw_8821ce 0000:02:00.0: unhandled firmware c2h interrupt
[  +5.121058] rtw_8821ce 0000:02:00.0: unhandled firmware c2h interrupt
[  +5.938611] rtw_8821ce 0000:02:00.0: unhandled firmware c2h interrupt
[Jun20 23:28] rtw_8821ce 0000:02:00.0: unhandled firmware c2h interrupt
[  +8.191796] rtw_8821ce 0000:02:00.0: unhandled firmware c2h interrupt

lwfinger, 2022-06-20:
It is not clear whether you are running the driver from this repo, or the one built into the kernel. If this one, did you blacklist the kernel version?

In either case, please try to other version.

pantr19, 2022-06-21:
No, I have not installed the driver from this repo, I am running the one built in kernel.
I have opened the issue here because this repo is the source for the built in kernel driver.