lwfinger/rtw88 #90: Make command fails on Fedora 36 plasma version

nestlecrunchkid, 2022-06-19:
When I execute the "make" command, I get this error:
make -C /lib/modules/5.17.5-300.fc36.x86_64/build M=/home/liveuser/rtw88 modules
make[1]: *** /lib/modules/5.17.5-300.fc36.x86_64/build: No such file or directory.  Stop.
make: *** [Makefile:83: all] Error 2

lwfinger, 2022-06-19:
You have not installed the kernel headers, which are needed to build a module outside the kernel. See README.md for instructions for Fedora.

wangd, 2022-07-01:
For some reason, I had to install `kernel-devel` on a new Fedora 36 installation(I had thought `kernel-headers` would be enough). Might give that a try... I still think `kernel-headers` should be enough.

lwfinger, 2022-07-01:
It depends on the distro. Fedora needs both. That is not his problem as missing kernel-devel would prevent the build, not the loading of the resulting .ko files.

closed 2022-12-27 by nestlecrunchkid