lwfinger/rtw88 #91: Fails on kernel 5.18.7 for fedora36

galure, 2022-07-01:
Hi, and thanks for the amazing job. 

Last kernel update breaks wifi on my Lenovo V15-ADA, and also breaks wifi Settings in gnome on Fedora 36 (wifi settings is absent from the panel and the settings window, although `nmcli con show` returns all available connections correctly). I'm able to update and pull using bluetooth tethering with my phone (so bluetooth works).

I did:
```
cd ~/rtw88
git pull
make
sudo make install
```
and it is successful (`Install rtw88 SUCCESS`)

but `modprobe rtw_8822ce` returns
```
modprobe: ERROR: could not insert 'rtw_8822ce': Exec format error
```
I wrote a post in askFedora explaining my issue in detail: [
Lost wifi and wifi settings after last update ](https://ask.fedoraproject.org/t/lost-wifi-and-wifi-settings-after-last-update/23959)

As a say in that post, 5.17.5 and 5.18.6 had `rtw_8822c`, `rtw_pci` and `rtw_core` built in (although not `rtw_8822ce`, which is why I followed your instructions in the first place). But in 5.18.7, `lsmod | grep rtw` returns nothing, which, if I understand well, would mean they discarded `rtw_*` altogether?

I'm sorry if what I'm saying makes no sense, I don't understand a lot of what the problem is, but I followed a lot of discussions with similar problems and this is where I'm stuck.

I looked at [that post](https://github.com/lwfinger/rtw88/issues/37) but I'm not sure I can reproduce the steps and diagnose the problem.

galure, 2022-07-01:
The return of `dmesg` is :
```
rtw_core: version magic '5.18.6-200.fc36.x86_64 SMP preempt mod_unload ' 
should be '5.18.7-200.fc36.x86_64 SMP preempt mod_unload '
```
Again, I don't know what I'm doing, but it seems to be the reason for the `modprobe: ERROR`

lwfinger, 2022-07-01:
Your problem is that you built the driver with 5.18.6 headers, but are running a 5.18.7 kernel. THEY MUST MATCH. No exceptions.

Using dnf to update kernel-headers  and kernel-devel should do the trick.

You can check that at build time. The first line logged after "make" is
make -C /lib/modules/<Kernel Version>/build

Make sure that the <Kernel Version> part of that line exactly matches the output of 'uname -r'. I could probably automate that part of the Makefile, but unlikely to do so. :)

galure, 2022-07-01:
Thank you for your reply
I don't understand everything, should I just do a `dnf update kernel-headers kernel-devel` ?

On build time, `make` displays :
```
make -C /lib/modules/5.18.7-200.fc36.x86_64/build M=/home/bidulorama/rtw88 modules
make[1]: Entering directory '/usr/src/kernels/5.18.7-200.fc36.x86_64'
make[1]: Leaving directory '/usr/src/kernels/5.18.7-200.fc36.x86_64'
```
So it seems to build the driver with 5.18.7 headers?
`uname -r` returns
```
5.18.7-200.fc36.x86_64
```
 So it seems to match

galure, 2022-07-01:
Or maybe I previously built it with the wrong headers and I have to suppress the old built driver before building the new one?

But I don't know how to do that. Sorry if I'm not understanding things correctly.

galure, 2022-07-01:
Thank you!
I was starting to do nasty things like running `make uninstall` or locating the drivers with `make -n install `and manually trying to remove them one by one before redoing `make` and `make install`. 
`make clean` is more subtle. And now wifi works again.

I still don't understand what happened, because I did the update, rebooted and built the driver, so I should have been on the new kernel with the new kernel-headers.

In order to prevent those mistakes in the future, I'll run a `make clean` everytime. Good to know, I'm learning a lot of things.

Thank you again for your time and for maintaining those life-saving drivers.