lwfinger/rtw88 #92: Should I use this driver on newer kernel like  5.18.13

shyamjos, 2022-08-04:
Hi 
I am running fedora 36 with Kernel  5.18.13, and I have RTL8822CE Card, Is there any additional fixes in this driver which is not available in the one that shipped with fedora.

lwfinger, 2022-08-04:
If the in-kernel driver is working, then use it.

The repo essentially is up to date with kernel 5.19, and will soon be up to date with kernel 6.0, which will not be released for about 2 months.

shyamjos, 2022-08-04:
I kinda felt this driver connects instantly after waking from sleep compared to the one shipped with fedora, that's why I specifically asked if there is any additional tweaking done in this driver.

Keep up the good work, Thank you

closed 2022-08-04 by shyamjos