lwfinger/rtw88 #93: module launch error

TreeFall1, 2022-08-09:
1. When I try to run the rtw_8723de module, I get an error:
`sudo modprobe rtw_8723de
`
`modprobe: ERROR: could not insert 'rtw_8723de': Exec format error`

2. The output of lspci -k says that the module is loaded, but not the driver:
`02:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8723DE 802.11b/g/n PCIe Adapter
	DeviceName: Tiffany
	Subsystem: Hewlett-Packard Company Device 8319
	Kernel modules: rtw_8723de
`
How i can fix it?

TreeFall1, 2022-08-09:
Forgot to add: Arch linux (Manjaro) on kernel 5.4.209-1

lwfinger, 2022-08-09:
The reason for it not loading will be given in the dmesg log. Are you using secure boot? If so, the modules need to be signed.

TreeFall1, 2022-08-09:
> The reason for it not loading will be given in the dmesg log. Are you using secure boot? If so, the modules need to be signed.
Im not using secure boot.
[dmesg.txt](https://github.com/lwfinger/rtw88/files/9294670/dmesg.txt)

lwfinger, 2022-08-09:
The clue is the line:
[   12.655384] rtw_core: version magic '5.4.209-1-lts54 SMP mod_unload ' should be '5.4.209-1-MANJARO SMP preempt mod_unload modversions '

You have built the driver with different headers than the kernel uses. The 'version magic' must match.

TreeFall1, 2022-08-10:
And how fix it?

lwfinger, 2022-08-10:
Install the correct headers.

shanmukhateja, 2023-01-12:
Hello,

I ran into the same issue. Can we somehow detect this at compile time and throw error?

lwfinger, 2023-01-12:
Not really. If your distro is doing things right, then installing the kernel headers should get the correct ones. If it does not, complain to them.