lwfinger/rtw88 #95: RTL8822BE: Wifi stops working after resume on Linux 5.19.2

UtkarshVerma, 2022-08-20:
```
[ 4914.877037] rtw_8822be 0000:04:00.0: Unable to change power state from D3cold to D0, device inaccessible
[ 4914.886333] rtw_8822be 0000:04:00.0: mac power on failed
[ 4914.886342] rtw_8822be 0000:04:00.0: failed to power on mac
[ 4914.886344] rtw_8822be 0000:04:00.0: failed to setup chip efuse info
[ 4914.886348] rtw_8822be 0000:04:00.0: failed to setup chip information
[ 4914.892027] rtw_8822be 0000:04:00.0: Firmware version 27.2.0, H2C version 13
[ 4914.922150] rtw_8822be: probe of 0000:04:00.0 failed with error -114
```

lwfinger, 2022-08-20:
There are two things to suggest:
1. Create a file named /etc/modprobe.d/50-rtw-core.conf containing a single line "options rtw_core". After you create that file, do a 'sudo modprobe -rv rtw_8822be && sudo modprobe -v rtw_8852be'. You should see the option on the rtw_core line of the second modprobe command.
2. Implement the script that unloads the module before sleep and reloads it after resume. Do a git pull and them 'sudo cp suspend_rtw8822be /usr/lib/systemd/system-sleep/'.

linas, 2022-09-15:
Ah, Heh. I just realized you are not a realtek engineer, and that this is not an "official" staging area for beta device drivers!  I will contact the linux-wireless mailing list, and will try to pursue there. Thanks for your patience!

UtkarshVerma, 2023-06-08:
seems to work with the provided module loading/unloading script.

closed 2023-06-08 by UtkarshVerma