lwfinger/rtw88 #99: Debian 11.5 Kernel 5.10.0-18-amd64 is stuck at rebooting after driver installation

mvedovato, 2022-09-28:
Hi!
I recently installed this driver in a clean Debian 11.5 installation.
The WiFi works fine, but after the reboot the system is stuck. After uninstallation in rescue mode the system starts normally.
Is there any workaround for my problem?
Thank you so much!
My apologies for misreading the install procedure and then making the wrong query elsewhere.

Mariano

lwfinger, 2022-09-28:
If you want the in-kernel version, then you should not be using this repo at all! Full stop!

From your other E-mail, it appears that you have this repo and the in-kernel drivers available. This situation may cause kernel instability including the possibility of crashes.

The wireless mailing list is for problems that I cannot solve. This situation is not one of them.

mvedovato, 2022-09-28:
Hi Larry, I want the driver of your repo. I always use it since several years and it was always the salvation. I don't fear the crash, re installation is not a problem.
In the recongnition of hardware during the installation of Debian, a message appears:
Firmware files missing: rtw88/rtw8723d_fw.bin
I tried to load but without luck.
So, is there nothing else that can be done? 
Thank you!

mvedovato, 2022-09-28:
Sorry, I did'nt read this:
The wireless mailing list is for problems that I cannot solve. This situation is not one of them.
Anything you need, please ask me.

lwfinger, 2022-09-28:
You have obviously not installed this repo using 'sudo make install'. That firmware should have been installed with the operating system, but it is part of this repo.

mvedovato, 2022-09-28:
Manually I have installed with sudo (like the README says).
The bin file that I mentioned, I had it from another try (several re installation I have tried before I realized the rescue mode login).

mvedovato, 2022-09-28:
I forgot try to blacklist rtw88_8723de, I did this:

$:cat /etc/modprobe.d/blacklist.conf 
blacklist rtw88_8723de
$:

I did the make and sudo make install the repo, then WiFi is OK. Reboot and stuck. Again rescue mode, uninstall repo with make uninstall, reboot and the system starts. Now:

$:
$:lsmod | grep rtw
$:

The same result of "lsmod | grep rtw" was before to blacklist rtw88_8723de after of uninstall the repo.

BR

Mariano

lwfinger, 2022-09-29:
If your blacklist attempt did not keep rtw88_8723de from loading, then it is in the wrong place. Placing your file in /etc/modprobe.d/ is the traditional location, but some distros have moved it to /usr/lib/modprobe.d/.

Run the command 'sudo /etc/modprobe.d/blacklist.conf /usr/lib/modprobe.d/50-rtw8723de.conf' and see if that helps.

mvedovato, 2022-09-29:
Hi Larry, it didn't work.
Anyway take a look to lsmod command:

$:sudo lsmod | grep rtw
$:

After uninstall the driver of this repo. It seems like there is no module for WiFi. Or am I missing something?

lwfinger, 2022-09-29:
Did you remove the blacklist?

mvedovato, 2022-09-29:
No,
$:ls -l /usr/lib/modprobe.d/50-rtw8723de.conf
-rw-r--r-- 1 root root 23 sep 29 13:37 /usr/lib/modprobe.d/50-rtw8723de.conf
$:
$:cat /usr/lib/modprobe.d/50-rtw8723de.conf
blacklist rtw88_8723de
$:

lwfinger, 2022-09-29:
Well, if you uninstall this repo, and prevent the kernel from loading its version with a blacklist, what do you expect?

You can always 'sudo modprove -v rtw88_8723de'. That will manually load the kernel version.

mvedovato, 2022-09-29:
I don't know if I mentioned, but before to create any blacklist and without installation of your repo the result of lsmod command was:
$:sudo lsmod | grep rtw
$:

Ok, now:

$:sudo modprobe -v rtw88_8723de
modprobe: FATAL: Module rtw88_8723de not found in directory /lib/modules/5.10.0-18-amd64
$:

BR
Mariano

lwfinger, 2022-09-29:
Somehow, your drivers must have been deleted. Reinstall the kernel.

mvedovato, 2022-09-29:
Ok:
$:sudo apt-get install --reinstall linux-image-5.10.0-18-amd64
Leyendo lista de paquetes... Hecho
Creando árbol de dependencias... Hecho
Leyendo la información de estado... Hecho
0 actualizados, 0 nuevos se instalarán, 1 reinstalados, 0 para eliminar y 4 no actualizados.
Se necesita descargar 55,4 MB de archivos.
Se utilizarán 0 B de espacio de disco adicional después de esta operación.
Des:1 http://deb.debian.org/debian bullseye/main amd64 linux-image-5.10.0-18-amd64 amd64 5.10.140-1 [55,4 MB]
Descargados 55,4 MB en 6s (9.434 kB/s)                  
(Leyendo la base de datos ... 201743 ficheros o directorios instalados actualmente.)
Preparando para desempaquetar .../linux-image-5.10.0-18-amd64_5.10.140-1_amd64.deb ...
Desempaquetando linux-image-5.10.0-18-amd64 (5.10.140-1) sobre (5.10.140-1) ...
Configurando linux-image-5.10.0-18-amd64 (5.10.140-1) ...
depmod: ERROR: failed to load symbols from /lib/modules/5.10.0-18-amd64/kernel/drivers/net/wireless/realtek/rtw88/rtw_8821c.ko: Invalid argument
depmod: ERROR: failed to load symbols from /lib/modules/5.10.0-18-amd64/kernel/drivers/net/wireless/realtek/rtw88/rtw_8723d.ko: Invalid argument
depmod: ERROR: failed to load symbols from /lib/modules/5.10.0-18-amd64/kernel/drivers/net/wireless/realtek/rtw88/rtw_8821ce.ko: Invalid argument
depmod: ERROR: failed to load symbols from /lib/modules/5.10.0-18-amd64/kernel/drivers/net/wireless/realtek/rtw88/rtw_8822ce.ko: Invalid argument
depmod: ERROR: failed to load symbols from /lib/modules/5.10.0-18-amd64/kernel/drivers/net/wireless/realtek/rtw88/rtw_core.ko: Invalid argument
depmod: ERROR: failed to load symbols from /lib/modules/5.10.0-18-amd64/kernel/drivers/net/wireless/realtek/rtw88/rtw_8822b.ko: Invalid argument
depmod: ERROR: failed to load symbols from /lib/modules/5.10.0-18-amd64/kernel/drivers/net/wireless/realtek/rtw88/rtw_8822be.ko: Invalid argument
depmod: ERROR: failed to load symbols from /lib/modules/5.10.0-18-amd64/kernel/drivers/net/wireless/realtek/rtw88/rtw_pci.ko: Invalid argument
depmod: ERROR: failed to load symbols from /lib/modules/5.10.0-18-amd64/kernel/drivers/net/wireless/realtek/rtw88/rtw_8822c.ko: Invalid argument
depmod: ERROR: failed to load symbols from /lib/modules/5.10.0-18-amd64/kernel/drivers/net/wireless/realtek/rtw88/rtw_8723de.ko: Invalid argument
/etc/kernel/postinst.d/dkms:
dkms: running auto installation service for kernel 5.10.0-18-amd64:
Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' all KVER=5.10.0-18-amd64.......(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.10.0-18-amd64 (x86_64)
Consult /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/make.log for more information.
.
/etc/kernel/postinst.d/initramfs-tools:
update-initramfs: Generating /boot/initrd.img-5.10.0-18-amd64
W: Possible missing firmware /lib/firmware/i915/skl_huc_2.0.0.bin for module i915
W: Possible missing firmware /lib/firmware/i915/skl_guc_33.0.0.bin for module i915
W: Possible missing firmware /lib/firmware/i915/bxt_huc_2.0.0.bin for module i915
W: Possible missing firmware /lib/firmware/i915/bxt_guc_33.0.0.bin for module i915
W: Possible missing firmware /lib/firmware/i915/kbl_huc_4.0.0.bin for module i915
W: Possible missing firmware /lib/firmware/i915/kbl_guc_33.0.0.bin for module i915
W: Possible missing firmware /lib/firmware/i915/glk_huc_4.0.0.bin for module i915
W: Possible missing firmware /lib/firmware/i915/glk_guc_33.0.0.bin for module i915
W: Possible missing firmware /lib/firmware/i915/kbl_huc_4.0.0.bin for module i915
W: Possible missing firmware /lib/firmware/i915/kbl_guc_33.0.0.bin for module i915
W: Possible missing firmware /lib/firmware/i915/cml_huc_4.0.0.bin for module i915
W: Possible missing firmware /lib/firmware/i915/cml_guc_33.0.0.bin for module i915
W: Possible missing firmware /lib/firmware/i915/icl_huc_9.0.0.bin for module i915
W: Possible missing firmware /lib/firmware/i915/icl_guc_33.0.0.bin for module i915
W: Possible missing firmware /lib/firmware/i915/ehl_huc_9.0.0.bin for module i915
W: Possible missing firmware /lib/firmware/i915/ehl_guc_33.0.4.bin for module i915
W: Possible missing firmware /lib/firmware/i915/tgl_huc_7.5.0.bin for module i915
W: Possible missing firmware /lib/firmware/i915/tgl_guc_35.2.0.bin for module i915
W: Possible missing firmware /lib/firmware/i915/tgl_huc_7.5.0.bin for module i915
W: Possible missing firmware /lib/firmware/i915/tgl_guc_35.2.0.bin for module i915
W: Possible missing firmware /lib/firmware/i915/bxt_dmc_ver1_07.bin for module i915
W: Possible missing firmware /lib/firmware/i915/skl_dmc_ver1_27.bin for module i915
W: Possible missing firmware /lib/firmware/i915/kbl_dmc_ver1_04.bin for module i915
W: Possible missing firmware /lib/firmware/i915/glk_dmc_ver1_04.bin for module i915
W: Possible missing firmware /lib/firmware/i915/cnl_dmc_ver1_07.bin for module i915
W: Possible missing firmware /lib/firmware/i915/icl_dmc_ver1_09.bin for module i915
W: Possible missing firmware /lib/firmware/i915/tgl_dmc_ver2_08.bin for module i915
W: Possible missing firmware /lib/firmware/i915/rkl_dmc_ver2_02.bin for module i915
/etc/kernel/postinst.d/zz-update-grub:
Generating grub configuration file ...
Found background image: /usr/share/images/desktop-base/desktop-grub.png
Found linux image: /boot/vmlinuz-5.10.0-18-amd64
Found initrd image: /boot/initrd.img-5.10.0-18-amd64
Warning: os-prober will be executed to detect other bootable partitions.
Its output will be used to detect bootable binaries on them and create new boot entries.
done
$:

In the next post I will send you the content of that log

mvedovato, 2022-09-29:
And the content of log:
mariano@debian-hp-blanca:~$ cat /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/make.log
DKMS make.log for rtl8723de-5.1.1.8_21285.20171026_COEX20170111-1414 for kernel 5.10.0-18-amd64 (x86_64)
jue 29 sep 2022 18:25:22 -03
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.10.0-18-amd64/build M=/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build  modules
make[1]: se entra en el directorio '/usr/src/linux-headers-5.10.0-18-amd64'
  CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_cmd.o
In file included from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/osdep_service.h:42,
                 from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/drv_types.h:32,
                 from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_cmd.c:22:
/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/osdep_service_linux.h: In function ‘_init_timer’:
/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/osdep_service_linux.h:287:8: error: ‘_timer’ {aka ‘struct timer_list’} has no member named ‘data’
  287 |  ptimer->data = (unsigned long)cntx;
      |        ^~
/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/osdep_service_linux.h:288:2: error: implicit declaration of function ‘init_timer’; did you mean ‘_init_timer’? [-Werror=implicit-function-declaration]
  288 |  init_timer(ptimer);
      |  ^~~~~~~~~~
      |  _init_timer
In file included from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/drv_types.h:35,
                 from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_cmd.c:22:
/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/wifi.h: At top level:
/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/wifi.h:1006: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
 1006 | #define IEEE80211_MAX_AMPDU_BUF 0x40
      | 
In file included from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/osdep_service_linux.h:86,
                 from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/osdep_service.h:42,
                 from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/drv_types.h:32,
                 from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_cmd.c:22:
/usr/src/linux-headers-5.10.0-18-common/include/linux/ieee80211.h:1656: note: this is the location of the previous definition
 1656 | #define IEEE80211_MAX_AMPDU_BUF  0x100
      | 
cc1: some warnings being treated as errors
make[3]: *** [/usr/src/linux-headers-5.10.0-18-common/scripts/Makefile.build:291: /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_cmd.o] Error 1
make[2]: *** [/usr/src/linux-headers-5.10.0-18-common/Makefile:1856: /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build] Error 2
make[1]: *** [/usr/src/linux-headers-5.10.0-18-common/Makefile:185: __sub-make] Error 2
make[1]: se sale del directorio '/usr/src/linux-headers-5.10.0-18-amd64'
make: *** [Makefile:1884: modules] Error 2
mariano@debian-hp-blanca:~$

mvedovato, 2022-09-29:
After reboot:
$:sudo modprobe -v rtw88_8723de
insmod /lib/modules/5.10.0-18-amd64/kernel/drivers/net/wireless/realtek/rtw88/rtw88_core.ko 
insmod /lib/modules/5.10.0-18-amd64/kernel/drivers/net/wireless/realtek/rtw88/rtw88_pci.ko 
insmod /lib/modules/5.10.0-18-amd64/kernel/drivers/net/wireless/realtek/rtw88/rtw88_8723d.ko 
insmod /lib/modules/5.10.0-18-amd64/kernel/drivers/net/wireless/realtek/rtw88/rtw88_8723de.ko 
$

And now I have WiFi. Let me reboot again. I will be back soon hehehehe

mvedovato, 2022-09-29:
Ok, this is the situation:
- Kernel reinstalled
- The driver of your repo is NOT installed
- command "lsmod | grep rtw doesn't show anything"
- command "sudo modprobe -v rtw88_8723de" activates the WiFi
- reboot the system, no stuck, starts normally but without WiFi
-  command "lsmod | grep rtw doesn't show anything"
- command "sudo modprobe -v rtw88_8723de" activates the WiFi:
$:sudo modprobe -v rtw88_8723de
[sudo] password for mariano: 
insmod /lib/modules/5.10.0-18-amd64/kernel/lib/crypto/libarc4.ko 
insmod /lib/modules/5.10.0-18-amd64/kernel/net/wireless/cfg80211.ko 
insmod /lib/modules/5.10.0-18-amd64/kernel/net/mac80211/mac80211.ko 
insmod /lib/modules/5.10.0-18-amd64/kernel/drivers/net/wireless/realtek/rtw88/rtw88_core.ko 
insmod /lib/modules/5.10.0-18-amd64/kernel/drivers/net/wireless/realtek/rtw88/rtw88_pci.ko 
insmod /lib/modules/5.10.0-18-amd64/kernel/drivers/net/wireless/realtek/rtw88/rtw88_8723d.ko 
insmod /lib/modules/5.10.0-18-amd64/kernel/drivers/net/wireless/realtek/rtw88/rtw88_8723de.ko 
$:

So, if I understand, is there a problem with in-built Kernel module? Could I use your driver and remove completily the built-in one?

lwfinger, 2022-09-29:
Yes, you can do that OR you can delete the blacklist that is keeping rtw88_8723de from automatically loading. Your choice.

There is no probl;em other than the one you created!

mvedovato, 2022-09-29:
The blacklist is commented, so it hasn't effect in the loading. My original driver has problem but I don't understand why when I want to use your driver the system is stuck. 
Note that I am usign WiFi with the original driver is due to your indication to probe with "sudo modprobe -v rtw88_8723de" (even so after a reboot the Wifi is lost).
Really I think that I didn't create any problem, I want to fix the Wifi issue (again, like several Debian versions ago I fixed with your driver)

closed 2022-09-30 by mvedovato