lwfinger/rtw88 #103: Problem updating kernel to 6.0.0 or latest with RTL8723DE module

Noroxi, 2022-10-24:
I had 5.15 kernel, i installed the module with out problems, didn´t work so i think it was the kernel, so i update the kernel but when i retry the install i get stuck in here:
[image]
Please help me, the module works before, but i don´t know when stops working after a reboot, so now i don´t have WIFI, bluetooth is working.
Kernel version: 6.0.3-060003-generic

Noroxi, 2022-10-24:
Ok I solved the problem, i will show response of code
sudo make into rtw88 folder returns:
[image]
[image]
sudo make install returns:
[image]
sudo modprobe rtw_8723de doesn´t returns anything, and wifi still not working, bluetooth is working but wifi isn´t.

lwfinger, 2022-10-24:
The Bluetooth device is USB and the wifi device is PCI. They only inhabit the same chip, but they do not share a driver.

The command 'sudo modprobe rtw_8723de' ONLY responds with output if it is unable to load the module in question.

Please send the following outputs. I suppose that you will need to take a screen shot:
1. 'lspci -nn'
2. 'lsmod | grep rtw'

Noroxi, 2022-10-24:
lspci -nn returns:
[image]
lsmod | grep rtw  I didn't have return to this comand

lwfinger, 2022-10-24:
If 'lsmod | grep rtw' returned nothing, your modules are not loaded. Please do 'sudo modprobe -v rtw_8723de'.

Your 'lspci -nn' command shows that you have an RTW8723DE, thus the driver should have been loaded after reboot. With Kernel 5.15, there is a potential problem of the in-kernel driver interacting badly with drivers from this repo, but there is a possibility that everything is blacklisted.

Noroxi, 2022-10-24:
`sudo modprobe -v rtw_8723de` returns:
[image]
i will reboot

Noroxi, 2022-10-24:
`sudo modprobe -v rtw_8723de` returns:
[image]
after reboot nothing happened

Noroxi, 2022-10-24:
i think that module is blacklisted. What do i need to do?
[image]

lwfinger, 2022-10-24:
Check rtl8723de.conf and rtw_8723de.conf. They should blacklist rtw88_8723de, but not rtw_8723de. The first one comes from the kernel.

After you did the 'sudo modprobe -v rtw88723de', did wifi work before you rebooted?

Noroxi, 2022-10-24:
rtl8723de.conf i think it define antenna setting:
[image]
same with rtw_8723de.conf
[image]
inside of blacklist-8723de.conf:

[image]

lwfinger, 2022-10-24:
You can delete rtl8723de.conf. That antenna setting is not used. Keep rtw_8723de.conf. Delete blacklist-8723de.conf.

Noroxi, 2022-10-24:
I have to reboot after that?

lwfinger, 2022-10-24:
Do you mean that it is still blacklisted?

Run the following command: 'sudo grep 8723 /etc/modprobe.d/*' and report the output.
Repeat for 'sudo grep 8723 /usr/lib/modprobe.d/*'

Noroxi, 2022-10-24:
[image]

Noroxi, 2022-10-24:
Into every folder i check all the files, only find 8723 in the next files:
[image]

lwfinger, 2022-10-25:
I have no idea why your system is blacklisting the files. You will need to 'sudo modprobe -v rtw_8723de' after every boot. Does wifi work after that?

Noroxi, 2022-10-25:
Look this now:
[image]

lwfinger, 2022-10-25:
Did you change kernels and forget to 'make' and 'sudo make install'? That has to be done every time that the kernel changes!!!!!!!!!

Noroxi, 2022-10-25:
I didn't change the kernel! When i passed from 6 to 5 i installed again...

Noroxi, 2022-10-25:
[image]

Noroxi, 2022-10-25:
Sorry, i did it, don´t work also. 
[image]

lwfinger, 2022-10-25:
That is probably not a failure. Use 'lsmod | grep rtw' to see if the modules are loaded.

If you do not have wifi, you will next need to check the output of the dmesg command to see if it has errors. That one will be too big for you to post photos.

Noroxi, 2022-10-26:
I don't know exactly what i'm looking for, but:
[image]
[image]
[image]
[image]

lwfinger, 2022-10-26:
This driver does not know about the 'ant_sel' parameter. The file in /etc/modrobe.d/ that sets it can be deleted. Otherwise, I do not see any errors. It found the firmware. The fact that the driver was loaded at 4.6 seconds means that the device and driver are recognized, and not blacklisted.

Do the following:
sudo rfkill list
sudo modprobe -rv rtw_8723de
sudo modprobe -v rtw_8723de

Send the output of the first command, then the very end of the dmesg output.

Noroxi, 2022-10-26:
[image]
What do you mean with  the very end of the dmesg output?

Noroxi, 2022-10-26:
I think that the problem is in core. Look at this:
[image]
[image]

lwfinger, 2022-10-26:
Enter 'dmesg' command. When it finishes, that is the end.

Noroxi, 2022-10-26:
[image]

lwfinger, 2022-10-26:
\I have no idea what you are unplugging and replugging to get those USB disconnects and reconnects, but they have nothing to do with rtw_8723de. There are no errors with it. The only thing missing is a rename of wlan0, but your system may not do that. It seems that the problem has to be with your user-space setup of a wireless device.

What does 'iw dev' show?

Noroxi, 2022-10-26:
[image]

lwfinger, 2022-10-26:
If you do 'sudo iw dev wlan0 scan | egrep "signal|SSID', do you see output? Send me a picture.

Noroxi, 2022-10-26:
[image]

Noroxi, 2022-10-26:
Look this!!
[image]

lwfinger, 2022-10-26:
Your software that controls the wifi has not enabled the device. That has nothing to do with this driver. As I do not know what distro you are using, I cannot help you with that. You will need to search for the method needed to bring up wifi on your setup.

From the code that I control, everything is working!

Noroxi, 2022-10-26:
Ok thanks!

closed 2022-10-26 by Noroxi