lwfinger/rtw88 #105: Make fails on Kernel 6.0.7-301

triakshunn, 2022-11-22:
Hi!

I am trying to use this for RTL8822be.
I have installed the kernel-devel and kernel-header package. 

[image]
Now when I try `make` command in the rtw88 directory,I get the following error:

[image]

I saw on other fedora forum [here](https://ask.fedoraproject.org/t/wifi-adapter-not-found-rtl8723de/4056/15) that its a kernel-devel and kernel-header dependency issue but here its showing that its installed,so idk whats the issue.
 Thanks!

lwfinger, 2022-11-22:
I do not know. I created a new Fedora 37 virtual machine, updated it, and install kernel-headers and kernel-devel just as the README.md says for Fedora. After that, the necessary link in /lib/modules/$(uname -r)/build was there, and I could compile rtw88. My kernel is 6.0.9 after the update, but that should not make any difference.

triakshunn, 2022-11-23:
Thanks for the reply!
Even I was able to build today! Idk what changed[I didnt do anything] but the build was successful. Anyways thanks for your time!

closed 2022-11-23 by triakshunn