lwfinger/rtw88 #106: Issue with rtw_8821ce

CSParadiso, 2022-12-07:
Hello everyone.
I've installed the last release of Ubuntu recently and cant fix an issue with this driver. Im not really sure is that is the problem neither.
I've followed the steps from this repository but the wifi keeps on failing sometimes.
I think that the screenshot shows that the driver is working but i dont know for sure (im learning).
If you have some advice, plis let me know.
Regards, Simon.
[image]

CSParadiso, 2022-12-07:
[image]

lwfinger, 2022-12-07:
Your screenshots show that you have an 8821ce and that the driver is loaded. Nothing more than that.

Does the output of dmesg contain any information about your wifi?

Does it show up in the output of 'iw dev'?

lwfinger, 2022-12-07:
These outputs say it is working.

CSParadiso, 2022-12-08:
Ok.
Thanks for the help. I will keep on trying, maybe there are some bad places for the Wifi signal.
Thanks again.

CSParadiso, 2022-12-09:
Thank you very much @lwfinger. 
The best performance ive reach so far is:
signal: -48.00 dBm
I will keep on searching the best spot. Thanks again!

lwfinger, 2022-12-09:
Anything larger than -60 should work. A signal of -48 dBm is nearly ideal. For various reasons, I run 4 different AP's at a distance of 1-2 m from my desk. The range of strengths ranges from -35 to -52 dBm.

closed 2022-12-09 by CSParadiso