lwfinger/rtw88 #108: Laptop overheating with rtw_8723de

apoorvpotnis, 2022-12-21:
I have installed this wifi driver for my chip rtl8723de. I have to blacklist it from loading at boot as the system does not boot if I don’t blacklist it. Is there a way to figure what goes wrong?

 If I manually load the driver after I boot, it works and I can use wifi. But after some time, the computer overheats for some reason and the fan goes crazy. I am sure this has something to do with this driver as unloading the driver does not cause this issue. 

Also, the bluetooth and wifi don’t work properly when used simultaneously. The bluetooth does get connected but often disconnects abruptly and when it does connect, the wifi stops working or becomes extremely slow.

My output of `journalctl -b | grep rtw` after I load the driver is as follows.

```
Dec 21 16:58:49 apoorvlaptop dbus-daemon[448]: [system] Activating via systemd: service name='org.freedesktop.home1' unit='dbus-org.freedesktop.home1.service' requested by ':1.65' (uid=0 pid=1484 comm="sudo modprobe rtw_8723de")
Dec 21 16:58:51 apoorvlaptop sudo[1484]: apoorvpotnis : TTY=pts/1 ; PWD=/home/apoorvpotnis ; USER=root ; COMMAND=/usr/bin/modprobe rtw_8723de
Dec 21 16:58:51 apoorvlaptop kernel: rtw_core: loading out-of-tree module taints kernel.
Dec 21 16:58:51 apoorvlaptop kernel: rtw_core: module verification failed: signature and/or required key missing - tainting kernel
Dec 21 16:58:51 apoorvlaptop kernel: rtw_8723de 0000:02:00.0: Firmware version 48.0.0, H2C version 0
Dec 21 16:58:51 apoorvlaptop kernel: rtw_8723de 0000:02:00.0 wlo1: renamed from wlan0
Dec 21 16:58:51 apoorvlaptop NetworkManager[457]: <info>  [1671622131.7414] rfkill1: found Wi-Fi radio killswitch (at /sys/devices/pci0000:00/0000:00:1c.5/0000:02:00.0/ieee80211/phy0/rfkill1) (driver rtw_8723de)
Dec 21 16:58:54 apoorvlaptop kernel: rtw_8723de 0000:02:00.0: failed to poll offset=0x6 mask=0x2 value=0x2
Dec 21 16:58:54 apoorvlaptop kernel: rtw_8723de 0000:02:00.0: mac power on failed
Dec 21 16:58:54 apoorvlaptop kernel: rtw_8723de 0000:02:00.0: failed to power on mac
Dec 21 16:58:54 apoorvlaptop kernel: rtw_8723de 0000:02:00.0: leave idle state failed
Dec 21 16:58:54 apoorvlaptop kernel: rtw_8723de 0000:02:00.0: failed to leave ips state
Dec 21 16:58:54 apoorvlaptop kernel: rtw_8723de 0000:02:00.0: failed to leave idle state
Dec 21 16:58:55 apoorvlaptop kernel: rtw_8723de 0000:02:00.0: timed out to flush pci tx ring[5]
Dec 21 16:58:59 apoorvlaptop kernel: rtw_8723de 0000:02:00.0: failed to poll offset=0x6 mask=0x2 value=0x2
Dec 21 16:58:59 apoorvlaptop kernel: rtw_8723de 0000:02:00.0: mac power on failed
Dec 21 16:58:59 apoorvlaptop kernel: rtw_8723de 0000:02:00.0: failed to power on mac
Dec 21 16:58:59 apoorvlaptop kernel: rtw_8723de 0000:02:00.0: leave idle state failed
Dec 21 16:58:59 apoorvlaptop kernel: rtw_8723de 0000:02:00.0: failed to leave ips state
Dec 21 16:58:59 apoorvlaptop kernel: rtw_8723de 0000:02:00.0: failed to leave idle state
Dec 21 16:58:59 apoorvlaptop kernel: rtw_8723de 0000:02:00.0: S1 TX IQK isn't done
Dec 21 16:58:59 apoorvlaptop kernel: rtw_8723de 0000:02:00.0: S1 RX IQK isn't done
Dec 21 16:58:59 apoorvlaptop kernel: rtw_8723de 0000:02:00.0: S1 RX IQK isn't done
Dec 21 16:58:59 apoorvlaptop kernel: rtw_8723de 0000:02:00.0: S1 RX IQK isn't done
Dec 21 16:58:59 apoorvlaptop kernel: rtw_8723de 0000:02:00.0: S1 RX IQK isn't done
Dec 21 16:58:59 apoorvlaptop kernel: rtw_8723de 0000:02:00.0: S0 TX IQK isn't done
Dec 21 16:58:59 apoorvlaptop kernel: rtw_8723de 0000:02:00.0: S0 RX IQK isn't done
Dec 21 16:58:59 apoorvlaptop kernel: rtw_8723de 0000:02:00.0: S0 RX IQK isn't done
Dec 21 16:58:59 apoorvlaptop kernel: rtw_8723de 0000:02:00.0: S0 RX IQK isn't done
Dec 21 16:58:59 apoorvlaptop kernel: rtw_8723de 0000:02:00.0: S0 RX IQK isn't done
Dec 21 16:58:59 apoorvlaptop kernel: rtw_8723de 0000:02:00.0: S1 TX IQK isn't done
Dec 21 16:58:59 apoorvlaptop kernel: rtw_8723de 0000:02:00.0: S1 RX IQK isn't done
Dec 21 16:58:59 apoorvlaptop kernel: rtw_8723de 0000:02:00.0: S1 RX IQK isn't done
Dec 21 16:58:59 apoorvlaptop kernel: rtw_8723de 0000:02:00.0: S1 RX IQK isn't done
Dec 21 16:58:59 apoorvlaptop kernel: rtw_8723de 0000:02:00.0: S1 RX IQK isn't done
Dec 21 16:58:59 apoorvlaptop kernel: rtw_8723de 0000:02:00.0: S0 TX IQK isn't done
Dec 21 16:58:59 apoorvlaptop kernel: rtw_8723de 0000:02:00.0: S0 RX IQK isn't done
Dec 21 16:58:59 apoorvlaptop kernel: rtw_8723de 0000:02:00.0: S0 RX IQK isn't done
Dec 21 16:58:59 apoorvlaptop kernel: rtw_8723de 0000:02:00.0: S0 RX IQK isn't done
Dec 21 16:58:59 apoorvlaptop kernel: rtw_8723de 0000:02:00.0: S0 RX IQK isn't done
Dec 21 16:59:01 apoorvlaptop kernel: rtw_8723de 0000:02:00.0: failed to send h2c command
Dec 21 16:59:02 apoorvlaptop kernel: rtw_8723de 0000:02:00.0: failed to send h2c command
Dec 21 16:59:02 apoorvlaptop kernel: rtw_8723de 0000:02:00.0: timed out to flush pci tx ring[5]
Dec 21 16:59:02 apoorvlaptop kernel: rtw_8723de 0000:02:00.0: failed to send h2c command
```

The output of `lsmod | grep rtw` after loading the driver is as follows.
```
rtw_8723de             16384  0
rtw_8723d              61440  1 rtw_8723de
rtw_pci                36864  1 rtw_8723de
rtw_core              278528  2 rtw_8723d,rtw_pci
mac80211             1298432  2 rtw_core,rtw_pci
cfg80211             1118208  2 rtw_core,mac80211
```

Earlier, the `rtw88_8723de` driver used to work fine. But it didn't work for some reason, I uninstalled it and installed this driver. I cannot install it again it gives a bad build error. I am on Manjaro KDE with kernel 6.0.14-1 but I have these problems with other kernels as well. 

Thank you.

lwfinger, 2022-12-21:
Without the details on the "bad build error", there is nothing I can do. You also did not list your kernel version.

The power to the chip problem is mostly when the machine is dual-booting Windows and Linux. Windows has a quick shutdown mode that does not power off the hardware. That will lead to the "mac power on" problem.

The contents of rtw88 match the driver in kernel 6.2.0-rcX, thus it is newer than the kernel you are likely running.

apoorvpotnis, 2022-12-22:
Right now, I am on Manjaro Linux 6.0.14-1. The contents of the log file `/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/make.log` are as follows. I followed the instructions given on https://github.com/smlinux/rtl8723de in order to build the `rtw88_8723de` driver (which I guess is provided by default in the kernel). 

```
DKMS make.log for rtl8723de-5.1.1.8_21285.20171026_COEX20170111-1414 for kernel 6.0.10-1-MANJARO (x86_64)
Thursday 22 December 2022 05:16:54 PM IST
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.0.10-1-MANJARO/build M=/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build  modules
make[1]: Entering directory '/usr/lib/modules/6.0.10-1-MANJARO/build'
CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_cmd.o
CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_security.o
CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_debug.o
CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_io.o
CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_ioctl_query.o
CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_ioctl_set.o
CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_ieee80211.o
CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_mlme.o
CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_mlme_ext.o
/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_mlme_ext.c: In function ‘mgt_dispatcher’:
/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_mlme_ext.c:1464:20: warning: this statement may fall through [-Wimplicit-fallthrough=]
1464 |                 if (check_fwstate(pmlmepriv, WIFI_AP_STATE) == _TRUE)
|                    ^
/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_mlme_ext.c:1469:9: note: here
1469 |         case WIFI_ASSOCREQ:
|         ^~~~
CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_mi.o
CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_wlan_util.o
CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_vht.o
CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_pwrctrl.o
CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_rf.o
CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_recv.o
CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_sta_mgt.o
/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_sta_mgt.c: In function ‘rtw_mfree_stainfo’:
/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_sta_mgt.c:344:25: warning: the comparison will always evaluate as ‘true’ for the address of ‘lock’ will never be NULL [-Waddress]
344 |         if (&psta->lock != NULL)
|                         ^~
In file included from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/drv_types.h:110,
from /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_sta_mgt.c:22:
/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/include/sta_info.h:199:17: note: ‘lock’ declared here
199 |         _lock   lock;
|                 ^~~~
CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_ap.o
CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_xmit.o
CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_p2p.o
CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_tdls.o
CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_br_ext.o
/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_br_ext.c:25:18: fatal error: net/ipx.h: No such file or directory
25 |         #include <net/ipx.h>
|                  ^~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:249: /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_br_ext.o] Error 1
make[1]: *** [Makefile:1852: /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.0.10-1-MANJARO/build'
make: *** [Makefile:1886: modules] Error 2
```

While closing Windows, I always press the Shift key while clicking on Shut Down to perform a complete shutdown.

apoorvpotnis, 2022-12-22:
Also, when the laptop overheats, the screen shuts off and I have to press to power button to power off and restart.

lwfinger, 2022-12-22:
The error message 
CC [M]  /var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_br_ext.o
/var/lib/dkms/rtl8723de/5.1.1.8_21285.20171026_COEX20170111-1414/build/core/rtw_br_ext.c:25:18: fatal error: net/ipx.h: No such file or directory
25 |         #include <net/ipx.h>
|                  ^~~~~~~~~~~

indicates that you ARE NOT running the driver from rtw88. I know the driver, and I do not suppport it.

closed 2022-12-22 by lwfinger

apoorvpotnis, 2022-12-22:
I think there is some misunderstanding. I have correctly installed the driver available in repository. I wrote about the other driver just to say that I cannot use it as well. My issue of overheating is with the driver available in this repo.

apoorvpotnis, 2022-12-22:
By repository, I mean **this** repository in the preceding message.

lwfinger, 2022-12-22:
When you quote an error message that my repository cannot posssibly produce, what am I to think? My time for dealing with these problems is limited.

Have you used compressed ait to clean the CPU fan? That can cause a lot of overheating problems.

Before you start wifi, open a terminal and start 'top'. Then when it starts to get hot, take a screen shot of the 'top' screen and post it.