lwfinger/rtw88 #109: Make failure on Pop!_OS kernel 6.0.12-76060006-generic

lbkNhubert, 2023-01-02:
`make -C /lib/modules/6.0.12-76060006-generic/build M=/the/path/rtw88 modules`
`make[1]: Entering directory '/usr/src/linux-headers-6.0.12-76060006-generic'`
`Warning: the compiler differs from the one used to build the kernel`
`  The kernel was built by: x86_64-linux-gnu-gcc-11 (Ubuntu 11.2.0-19ubuntu1) 11.2.0`
`  You are using:           gcc (Ubuntu 11.3.0-1ubuntu1-22.04) 11.3.0`
`make[1]: Leaving directory '/usr/src/linux-headers-6.0.12-76060006-generic'`

So far I have not been able to figure out how to work around this. Any help appreciated, thank you!

lwfinger, 2023-01-02:
If it loads and works, then ignore the warning. Otherwise, you need to ask Pop!_OS why they are delivering an unusable system.

closed 2023-01-02 by lwfinger

lbkNhubert, 2023-01-02:
Thank you for the quick reply. It's not loading, for now I reverted the blacklisting and am using the bundled modules. I agree that it doesn't make sense to have the kernel built on a version of the compiler that is not what is used by the system. Fortunately that is not my primary machine, so I can let it sit for a while and see if things get sorted out.

Thank you for your work on this, and have a great day.

ohquait, 2023-01-03:
@lbkNhubert 
give a report to the packagers of Pop!_OS distro, 
MAYBE...
they have to include pieces ( ... between ... ) of the line below:
`MAKE="... export IGNORE_CC_MISMATCH=1; 'make'....`
into the used dkms.conf or in %pkgbuild: for packaging
to prevent the warning's,
otherwise all users must have the same GCC/CC version's such the Packager?
Pop!OS have a other base-system or build-system than Ubuntu ?
With the driver source from here all is okay,

ohquait, 2023-01-05:
**JUST TO INFORM NOW it WORKS**

First: **THANK YOU LARRY FOR YOUR HARD WORK! I BELIEVE ON YOU!** 

i becomes the same error before , now with patched Makefile
the Build works perfectly, also the install with dkms works exactly,...
btw: the Makefile don't accepted a patch on rpm-build so i must rewrite and past it to the source-tarball
before packaging, maybe i have a miss-matching codings from used texteditor???
I will not publishing my makefile, she is edited to work on my loved distribution and
i knowing that's is not amazing to handle for the short times Larry have on day's. 

```
 apt-get install dkms-rtlwifi-rtw88-0.1.2-1pclos2023.x86_64.rpm
Reading Package Lists... Done
Building Dependency Tree... Done
Selecting dkms-rtlwifi-rtw88 for 'dkms-rtlwifi-rtw88-0.1.2-1pclos2023.x86_64.rpm'
The following NEW packages will be installed:
   dkms-rtlwifi-rtw88 (0.1.2-1pclos2023)
0 upgraded, 1 newly installed, 0 removed and 0 not upgraded.
Need to get 0B/529kB of archives.
After unpacking 4519kB of additional disk space will be used.
Committing changes...
Preparing                                ############################## [100%]
Updating / installing
  dkms-rtlwifi-rtw88-0.1.2-1pclos2023.x8 ############################## [100%]

Creating symlink /var/lib/dkms/rtlwifi-rtw88/0.1.2-1pclos2023/source ->
                 /usr/src/rtlwifi-rtw88-0.1.2-1pclos2023

DKMS: add Completed.

Preparing kernel 6.1.2-pclos1 for module build:
(This is not compiling a kernel, just preparing kernel symbols)
Storing current .config to be restored when complete
Running Generic preparation routine
make clean....(bad exit status: 2)
using /proc/config.gz
make oldconfig....
make prepare....

Building module:
cleaning build area....
export IGNORE_CC_MISMATCH=1; 'make' clean KSRC=/lib/modules/6.1.2-pclos1/build SKIP_STACK_VALIDATION=1 all..........
cleaning build area....
cleaning kernel tree (make clean)....(bad exit status: 2)

DKMS: build Completed.

rtw_pci.ko.xz:
 - Installation
   - Installing to /lib/modules/6.1.2-pclos1/kernel/3rdparty/rtlwifi-rtw88/

rtw_8723de.ko.xz:
 - Installation
   - Installing to /lib/modules/6.1.2-pclos1/kernel/3rdparty/rtlwifi-rtw88/

rtw_8723d.ko.xz:
 - Installation
   - Installing to /lib/modules/6.1.2-pclos1/kernel/3rdparty/rtlwifi-rtw88/

rtw_8822be.ko.xz:
 - Installation
   - Installing to /lib/modules/6.1.2-pclos1/kernel/3rdparty/rtlwifi-rtw88/

rtw_8822b.ko.xz:
 - Installation
   - Installing to /lib/modules/6.1.2-pclos1/kernel/3rdparty/rtlwifi-rtw88/

rtw_8822ce.ko.xz:
 - Installation
   - Installing to /lib/modules/6.1.2-pclos1/kernel/3rdparty/rtlwifi-rtw88/

rtw_8822c.ko.xz:
 - Installation
   - Installing to /lib/modules/6.1.2-pclos1/kernel/3rdparty/rtlwifi-rtw88/

rtw_8821ce.ko.xz:
 - Installation
   - Installing to /lib/modules/6.1.2-pclos1/kernel/3rdparty/rtlwifi-rtw88/

rtw_8821c.ko.xz:
 - Installation
   - Installing to /lib/modules/6.1.2-pclos1/kernel/3rdparty/rtlwifi-rtw88/

rtw_core.ko.xz:
 - Installation
   - Installing to /lib/modules/6.1.2-pclos1/kernel/3rdparty/rtlwifi-rtw88/

rtw_8723du.ko.xz:
 - Installation
   - Installing to /lib/modules/6.1.2-pclos1/kernel/3rdparty/rtlwifi-rtw88/

rtw_8822bu.ko.xz:
 - Installation
   - Installing to /lib/modules/6.1.2-pclos1/kernel/3rdparty/rtlwifi-rtw88/

rtw_8822cu.ko.xz:
 - Installation
   - Installing to /lib/modules/6.1.2-pclos1/kernel/3rdparty/rtlwifi-rtw88/

rtw_8821cu.ko.xz:
 - Installation
   - Installing to /lib/modules/6.1.2-pclos1/kernel/3rdparty/rtlwifi-rtw88/

rtw_usb.ko.xz:
 - Installation
   - Installing to /lib/modules/6.1.2-pclos1/kernel/3rdparty/rtlwifi-rtw88/

depmod.....

DKMS: install Completed.
Reloading network configuration ... 
Done.
```

ohquait, 2023-01-06:
?