lwfinger/rtw88 #110: DHCP Offer packet issue.

soragaoanakash, 2023-01-13:
with RTL8822BE WiFi card we facing DHCP offer packet loss and we can't able capture even in the air.

Test scenario 1: 
When Client DHCP Discover packet in the unicast mode  [FLAG : 0x0000 ], we able to see OFFER packet transmitted via antenna and we able to capture packet with WIFI analyser tool.  

Packet flow
Client  --------[Discover]-------->  server
Server --------[OFFER] ---------> client
Client ----------[REQUEST]--------> Server
Server ---------[ACK]------------> client 

Test scenario 2: 
When Client DHCP Discover packet is the Broadcast mode [FLAG : 0x8000], we able to see OFFER packet generated in the server side but not transmitted via TX path.

Client keep on sending DISCOVER packet and server keep on generating OFFER packet, It never reach to the client device and keeps on looping until time out. 

Packet flow

Client --------[Discover]----------> Server
Server ---------[OFFER]------/.Not Transmitted//  Client
Client --------[Discover]----------> Server
Server ---------[OFFER]------/.Not Transmitted//  Client
Client --------[Discover]----------> Server
Server ---------[OFFER]------/.Not Transmitted//  Client

soragaoanakash, 2023-01-13:
```
root@OpenWrt:/# tcpdump -i wlan0 port 67 and 68 -vv
tcpdump: listening on wlan0, link-type EN10MB (Ethernet), capture size 262144 bytes
23:28:14.898433 IP (tos 0x10, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 328)
    0.0.0.0.bootpc > 255.255.255.255.bootps: [udp sum ok] BOOTP/DHCP, Request from c4:93:00:23:cd:10 (oui Unknown), length 300, xid 0x2812741, secs 5, Flags [Broadcast] (0x8000)
          Client-Ethernet-Address c4:93:00:23:cd:10 (oui Unknown)
          Vendor-rfc1048 Extensions
            Magic Cookie 0x63825363
            DHCP-Message Option 53, length 1: Discover
            Hostname Option 12, length 7: "Komikan"
            Parameter-Request Option 55, length 13: 
              Subnet-Mask, BR, Time-Zone, Default-Gateway
              Domain-Name, Domain-Name-Server, Option 119, Hostname
              Netbios-Name-Server, Netbios-Scope, MTU, Classless-Static-Route
              NTP
23:28:14.898459 IP (tos 0x10, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 328)
    0.0.0.0.bootpc > 255.255.255.255.bootps: [udp sum ok] BOOTP/DHCP, Request from c4:93:00:23:cd:10 (oui Unknown), length 300, xid 0x2812741, secs 5, Flags [Broadcast] (0x8000)
          Client-Ethernet-Address c4:93:00:23:cd:10 (oui Unknown)
          Vendor-rfc1048 Extensions
            Magic Cookie 0x63825363
            DHCP-Message Option 53, length 1: Discover
            Hostname Option 12, length 7: "Komikan"
            Parameter-Request Option 55, length 13: 
              Subnet-Mask, BR, Time-Zone, Default-Gateway
              Domain-Name, Domain-Name-Server, Option 119, Hostname
              Netbios-Name-Server, Netbios-Scope, MTU, Classless-Static-Route
              NTP
23:28:14.898811 IP (tos 0xc0, ttl 64, id 29749, offset 0, flags [none], proto UDP (17), length 328)
    OpenWrt.lan.bootps > 255.255.255.255.bootpc: [udp sum ok] BOOTP/DHCP, Reply, length 300, xid 0x2812741, secs 5, Flags [Broadcast] (0x8000)
          Your-IP 192.168.1.119
          Server-IP OpenWrt.lan
          Client-Ethernet-Address c4:93:00:23:cd:10 (oui Unknown)
          Vendor-rfc1048 Extensions
            Magic Cookie 0x63825363
            DHCP-Message Option 53, length 1: Offer
            Server-ID Option 54, length 4: OpenWrt.lan
            Lease-Time Option 51, length 4: 43200
            RN Option 58, length 4: 21600
            RB Option 59, length 4: 37800
            Subnet-Mask Option 1, length 4: 255.255.255.0
            BR Option 28, length 4: 192.168.1.255
            Default-Gateway Option 3, length 4: OpenWrt.lan
            Domain-Name-Server Option 6, length 4: OpenWrt.lan
            Domain-Name Option 15, length 3: "lan"
23:28:21.905809 IP (tos 0x10, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 328)
    0.0.0.0.bootpc > 255.255.255.255.bootps: [udp sum ok] BOOTP/DHCP, Request from c4:93:00:23:cd:10 (oui Unknown), length 300, xid 0x2812741, secs 12, Flags [Broadcast] (0x8000)
          Client-Ethernet-Address c4:93:00:23:cd:10 (oui Unknown)
          Vendor-rfc1048 Extensions
            Magic Cookie 0x63825363
            DHCP-Message Option 53, length 1: Discover
            Hostname Option 12, length 7: "Komikan"
            Parameter-Request Option 55, length 13: 
              Subnet-Mask, BR, Time-Zone, Default-Gateway
              Domain-Name, Domain-Name-Server, Option 119, Hostname
              Netbios-Name-Server, Netbios-Scope, MTU, Classless-Static-Route
              NTP
23:28:21.905826 IP (tos 0x10, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 328)
    0.0.0.0.bootpc > 255.255.255.255.bootps: [udp sum ok] BOOTP/DHCP, Request from c4:93:00:23:cd:10 (oui Unknown), length 300, xid 0x2812741, secs 12, Flags [Broadcast] (0x8000)
          Client-Ethernet-Address c4:93:00:23:cd:10 (oui Unknown)
          Vendor-rfc1048 Extensions
            Magic Cookie 0x63825363
            DHCP-Message Option 53, length 1: Discover
            Hostname Option 12, length 7: "Komikan"
            Parameter-Request Option 55, length 13: 
              Subnet-Mask, BR, Time-Zone, Default-Gateway
              Domain-Name, Domain-Name-Server, Option 119, Hostname
              Netbios-Name-Server, Netbios-Scope, MTU, Classless-Static-Route
              NTP
23:28:21.906073 IP (tos 0xc0, ttl 64, id 30135, offset 0, flags [none], proto UDP (17), length 328)
    OpenWrt.lan.bootps > 255.255.255.255.bootpc: [udp sum ok] BOOTP/DHCP, Reply, length 300, xid 0x2812741, secs 12, Flags [Broadcast] (0x8000)
[... 132 lines cut ...]
          Vendor-rfc1048 Extensions
            Magic Cookie 0x63825363
            DHCP-Message Option 53, length 1: Discover
            Parameter-Request Option 55, length 7: 
              Subnet-Mask, BR, Time-Zone, Default-Gateway
              Domain-Name, Domain-Name-Server, Hostname
23:29:54.673202 IP (tos 0xc0, ttl 64, id 32691, offset 0, flags [none], proto UDP (17), length 328)
    OpenWrt.lan.bootps > 255.255.255.255.bootpc: [udp sum ok] BOOTP/DHCP, Reply, length 300, xid 0xdd905359, secs 26, Flags [Broadcast] (0x8000)
          Your-IP 192.168.1.119
          Server-IP OpenWrt.lan
          Client-Ethernet-Address c4:93:00:23:cd:10 (oui Unknown)
          Vendor-rfc1048 Extensions
            Magic Cookie 0x63825363
            DHCP-Message Option 53, length 1: Offer
            Server-ID Option 54, length 4: OpenWrt.lan
            Lease-Time Option 51, length 4: 43200
            RN Option 58, length 4: 21600
            RB Option 59, length 4: 37800
            Subnet-Mask Option 1, length 4: 255.255.255.0
            BR Option 28, length 4: 192.168.1.255
            Default-Gateway Option 3, length 4: OpenWrt.lan
            Domain-Name-Server Option 6, length 4: OpenWrt.lan
            Domain-Name Option 15, length 3: "lan"

```

soragaoanakash, 2023-01-13:
DHCP Flow during the unicast mode.
```
:53:53.380538 IP (tos 0x10, ttl 64, id 0, offset 0, flags [DF], proto UDP (17), length 326)
	0.0.0.0.bootpc > 255.255.255.255.bootps: [udp sum ok] BOOTP/DHCP, Request from ce:d4:15:d6:23:7d (oui Unknown), length 298, xid 0xa7c6f59d, secs 6, Flags [none] (0x0000)
		  Client-Ethernet-Address ce:d4:15:d6:23:7d (oui Unknown)
		  Vendor-rfc1048 Extensions
			Magic Cookie 0x63825363
			DHCP-Message Option 53, length 1: Discover
			Client-ID Option 61, length 7: ether ce:d4:15:d6:23:7d
			MSZ Option 57, length 2: 1500
			Vendor-Class Option 60, length 15: "android-dhcp-13"
			Hostname Option 12, length 7: "AKASH"
			Parameter-Request Option 55, length 12: 
			  Subnet-Mask, Default-Gateway, Domain-Name-Server, Domain-Name
			  MTU, BR, Lease-Time, RN
			  RB, Vendor-Option, URL, Option 108
22:53:53.380559 IP (tos 0x10, ttl 64, id 0, offset 0, flags [DF], proto UDP (17), length 326)
	0.0.0.0.bootpc > 255.255.255.255.bootps: [udp sum ok] BOOTP/DHCP, Request from ce:d4:15:d6:23:7d (oui Unknown), length 298, xid 0xa7c6f59d, secs 6, Flags [none] (0x0000)
		  Client-Ethernet-Address ce:d4:15:d6:23:7d (oui Unknown)
		  Vendor-rfc1048 Extensions
			Magic Cookie 0x63825363
			DHCP-Message Option 53, length 1: Discover
			Client-ID Option 61, length 7: ether ce:d4:15:d6:23:7d
			MSZ Option 57, length 2: 1500
			Vendor-Class Option 60, length 15: "android-dhcp-13"
			Hostname Option 12, length 7: "AKASH"
			Parameter-Request Option 55, length 12: 
			  Subnet-Mask, Default-Gateway, Domain-Name-Server, Domain-Name
			  MTU, BR, Lease-Time, RN
			  RB, Vendor-Option, URL, Option 108
22:53:53.380872 IP (tos 0xc0, ttl 64, id 31615, offset 0, flags [none], proto UDP (17), length 328)
	OpenWrt.lan.bootps > 192.168.1.205.bootpc: [udp sum ok] BOOTP/DHCP, Reply, length 300, xid 0xa7c6f59d, secs 6, Flags [none] (0x0000)
		  Your-IP 192.168.1.205
		  Server-IP OpenWrt.lan
		  Client-Ethernet-Address ce:d4:15:d6:23:7d (oui Unknown)
		  Vendor-rfc1048 Extensions
			Magic Cookie 0x63825363
			DHCP-Message Option 53, length 1: Offer
			Server-ID Option 54, length 4: OpenWrt.lan
			Lease-Time Option 51, length 4: 43200
			RN Option 58, length 4: 21600
			RB Option 59, length 4: 37800
			Subnet-Mask Option 1, length 4: 255.255.255.0
			BR Option 28, length 4: 192.168.1.255
22:53:53.395145 IP (tos 0x10, ttl 64, id 0, offset 0, flags [DF], proto UDP (17), length 338)
	0.0.0.0.bootpc > 255.255.255.255.bootps: [udp sum ok] BOOTP/DHCP, Request from ce:d4:15:d6:23:7d (oui Unknown), length 310, xid 0xa7c6f59d, secs 6, Flags [none] (0x0000)
		  Client-Ethernet-Address ce:d4:15:d6:23:7d (oui Unknown)
		  Vendor-rfc1048 Extensions
			Magic Cookie 0x63825363
			DHCP-Message Option 53, length 1: Request
			Client-ID Option 61, length 7: ether ce:d4:15:d6:23:7d
			Requested-IP Option 50, length 4: 192.168.1.205
			Server-ID Option 54, length 4: OpenWrt.lan
			MSZ Option 57, length 2: 1500
			Vendor-Class Option 60, length 15: "android-dhcp-13"
			Hostname Option 12, length 7: "AKASH"
			Parameter-Request Option 55, length 12: 
			  Subnet-Mask, Default-Gateway, Domain-Name-Server, Domain-Name
			  MTU, BR, Lease-Time, RN
			  RB, Vendor-Option, URL, Option 108
22:53:53.395156 IP (tos 0x10, ttl 64, id 0, offset 0, flags [DF], proto UDP (17), length 338)
	0.0.0.0.bootpc > 255.255.255.255.bootps: [udp sum ok] BOOTP/DHCP, Request from ce:d4:15:d6:23:7d (oui Unknown), length 310, xid 0xa7c6f59d, secs 6, Flags [none] (0x0000)
		  Client-Ethernet-Address ce:d4:15:d6:23:7d (oui Unknown)
		  Vendor-rfc1048 Extensions
			Magic Cookie 0x63825363
			DHCP-Message Option 53, length 1: Request
			Client-ID Option 61, length 7: ether ce:d4:15:d6:23:7d
			Requested-IP Option 50, length 4: 192.168.1.205
			Server-ID Option 54, length 4: OpenWrt.lan
			MSZ Option 57, length 2: 1500
			Vendor-Class Option 60, length 15: "android-dhcp-13"
			Hostname Option 12, length 7: "AKASH"
			Parameter-Request Option 55, length 12: 
			  Subnet-Mask, Default-Gateway, Domain-Name-Server, Domain-Name
			  MTU, BR, Lease-Time, RN
			  RB, Vendor-Option, URL, Option 108
22:53:53.395416 IP (tos 0xc0, ttl 64, id 31616, offset 0, flags [none], proto UDP (17), length 328)
	OpenWrt.lan.bootps > 192.168.1.205.bootpc: [udp sum ok] BOOTP/DHCP, Reply, length 300, xid 0xa7c6f59d, secs 6, Flags [none] (0x0000)
		  Your-IP 192.168.1.205
		  Server-IP OpenWrt.lan
		  Client-Ethernet-Address ce:d4:15:d6:23:7d (oui Unknown)
		  Vendor-rfc1048 Extensions
			Magic Cookie 0x63825363
			DHCP-Message Option 53, length 1: ACK
			Server-ID Option 54, length 4: OpenWrt.lan
			Lease-Time Option 51, length 4: 43200
			RN Option 58, length 4: 21600
			RB Option 59, length 4: 37800
			Subnet-Mask Option 1, length 4: 255.255.255.0
			BR Option 28, length 4: 192.168.1.255
			Default-Gateway Option 3, length 4: OpenWrt.lan
			Domain-Name-Server Option 6, length 4: OpenWrt.lan
			Domain-Name Option 15, length 3: "lan"
22:53:53.864966 IP (tos 0x0, ttl 64, id 0, offset 0, flags [none], proto UDP (17), length 328)
```

lwfinger, 2023-01-13:
Your problem is one that I will be unlikely to solve, and I will likely need to send it to the program authors in Taiwan.

Before I do that, tell me a little more about your setup. Is the RTL8822BE being used as an AP, or just as a DHCP server? What is the hardware and CPU type? I see openWRT mentioned. Is the 8822BE on that hardware?

soragaoanakash, 2023-01-13:
@lwfinger , Thanks for the reply.

Hardware details : 
SOC :  LS1046AFRWY (ARM 644 bit)
WIFI PCI Module : RTL8822BE 

Operation system : Openwrt 22.03 Version.

soragaoanakash, 2023-01-13:
With other WiFi vendor module we not observing such issue, only with RTL8822BE we facing weired issue.

lwfinger, 2023-01-18:
The Realtek group will do some tests, but with the Chinese New Year, it will take some time.