lwfinger/rtw88 #111: TX power not able to modify.

soragaoanakash, 2023-01-13:
Hi,

We trying to adjust the TX power using /etc/config/wireless configuration, but we can't able to change TX power of RTL8822BE.

In the RTL driver code we observed the TXSCAL table, we not getting what exact the TXSCAL meant for, is possible to modify for the TXSCALE table to achieve low TX power.  

RTL8822b.c 
```
#define RTW_TXSCALE_SIZE 37
static const u32 rtw8822b_txscale_tbl[RTW_TXSCALE_SIZE] = {
	0x081, 0x088, 0x090, 0x099, 0x0a2, 0x0ac, 0x0b6, 0x0c0, 0x0cc, 0x0d8,
	0x0e5, 0x0f2, 0x101, 0x110, 0x120, 0x131, 0x143, 0x156, 0x16a, 0x180,
	0x197, 0x1af, 0x1c8, 0x1e3, 0x200, 0x21e, 0x23e, 0x261, 0x285, 0x2ab,
	0x2d3, 0x2fe, 0x32b, 0x35c, 0x38e, 0x3c4, 0x3fe
};
```
Thank you
Akash

lwfinger, 2023-01-13:
I did not write the driver. I merely take the kernel code, adapt it to build on older kernels, and put it here. I have no idea what that table does.

What are you trying to do?

soragaoanakash, 2023-01-13:
@lwfinger , Thanks for the reply.

We  are trying to reduce TX power.

lwfinger, 2023-01-13:
I will need to send an inquiry to the driver's authors. It will be at least Monday before I get an answer.

I looked through the code and found ".max_power_index = 0x3f," at line 2536 od rtw8822b.c. You might try what happens if yiu change that 0x3f to 0x1f. I do not have the chip so I cannot test the effect.

soragaoanakash, 2023-01-13:
Yes sure, I will update you on Monday.

soragaoanakash, 2023-01-17:
@lwfinger , Thanks for the Input.

soragaoanakash, 2023-01-17:
When I tried to update the Kernel to 5.15, started facing other issues. 

`    9.367853] ------------[ cut here ]------------
[    9.372484] write RF mode table fail
[    9.376081] WARNING: CPU: 2 PID: 3542 at backports-6.1-rc8/drivers/net/wireless/realtek/rtw88/rtw8822b.c:805 0xffff800000eac23c
[    9.387569] Modules linked in: rtw88_8822ce rtw88_8822c rtw88_8822be rtw88_8822b rtw88_8723de rtw88_8723d pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet rtw88_pci rtw88n
[    9.452539] CPU: 2 PID: 3542 Comm: iw Not tainted 5.15.88 #0
[    9.458200] Hardware name: LS1046A FRWY Board (DT)
[    9.462987] pstate: 80400005 (Nzcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[    9.469950] pc : 0xffff800000eac23c
[    9.473433] lr : 0xffff800000eac23c
[    9.476916] sp : ffff80000ae837b0
[    9.480225] x29: ffff80000ae837b0 x28: ffff000800cf4180 x27: ffff80000ae83dd0
[    9.487365] x26: 0000000000000001 x25: ffff800000eae2a8 x24: 0000000000000005
[    9.494504] x23: ffff800000eae010 x22: 0000000000000000 x21: 00000000000aeaea
[    9.501644] x20: ffff0008050269f0 x19: ffff000805022060 x18: ffff800008e7fa10
[    9.508782] x17: 0000000000000000 x16: 0000000000000000 x15: 000000000000063c
[    9.515920] x14: 0000000000000214 x13: ffff80000ae834d8 x12: ffff800008ed7a10
[    9.523060] x11: fffffffffffe99b0 x10: ffff800008ed7a10 x9 : 0000000000000000
[    9.530198] x8 : ffff800008e7f9c0 x7 : ffff800008e7fa10 x6 : 00000000ffffefff
[    9.537337] x5 : ffff00087bfd5770 x4 : 0000000000000000 x3 : 0000000000000027
[    9.544475] x2 : 0000000000000027 x1 : ffff000800cf4180 x0 : 0000000000000018
[    9.551613] Call trace:
[    9.554053]  0xffff800000eac23c
[    9.557188]  0xffff800000eac63c
[    9.560323]  rtw_set_txrx_1ss+0x24c/0x17b0 [rtw88_core]
[    9.565556]  ieeee80211_obss_color_collision_notify+0x19a8/0x2e3c [mac80211]
[    9.572622]  nl80211_parse_chandef+0x1638/0x2e54 [cfg80211]
[    9.578201]  genl_family_rcv_msg_doit+0xb8/0x11c
[    9.582823]  genl_rcv_msg+0xd4/0x1cc
[    9.586395]  netlink_rcv_skb+0x5c/0x130
[    9.590228]  genl_rcv+0x38/0x50
[    9.593366]  netlink_unicast+0x184/0x280
[    9.597286]  netlink_sendmsg+0x1a0/0x3dc
[    9.601205]  ____sys_sendmsg+0x1b8/0x220
[    9.605125]  ___sys_sendmsg+0x84/0xf0
[    9.608783]  __sys_sendmsg+0x48/0xb0
[    9.612354]  __arm64_sys_sendmsg+0x24/0x30
[    9.616447]  invoke_syscall.constprop.0+0x50/0xf0
[    9.621149]  do_el0_svc+0x50/0x140
[    9.624547]  el0_svc+0x18/0x54
[    9.627597]  el0t_64_sync_handler+0xe8/0x114
[    9.631863]  el0t_64_sync+0x184/0x188
[    9.635520] ---[ end trace 5fc4dc9a29135c5f ]---
`

soragaoanakash, 2023-01-17:
I don't what exactly happening over there.

lwfinger, 2023-01-17:
First of all, switch from the in-kernel driver to the one from this repo. That will update the driver from kernel 5.15 to 6.2. It is possible, even likely, that the problem is fixed.

The warning happens because the driver is unable to write to the RF control registers. I have never seen that before.

Why are the 8821c and 8723d modules loaded if you have an 8822b module?

soragaoanakash, 2023-01-23:
Hi @lwfinger ,

Sorry for the late reply. 

In openwrt build default when we select rtw88 package all the driver will be part of the build and loaded during the kernel bootup. 

error remains same even after removing unused driver.   

`[    8.947089] ------------[ cut here ]------------
[    8.951729] write RF mode table fail
[    8.955324] WARNING: CPU: 3 PID: 3536 at backports-6.1-rc8/drivers/net/wireless/realtek/rtw88/rtw8822b.c:805 0xffff800000e9423c
[    8.966814] Modules linked in: rtw88_8822be rtw88_8822b pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet rtw88_pci rtw88_core pppox ppp_generic nft_reject_ipv6 nft_rejectn
[    9.027430] CPU: 3 PID: 3536 Comm: iw Not tainted 5.15.88 #0
[    9.033090] Hardware name: LS1046A FRWY Board (DT)
[    9.037877] pstate: 80400005 (Nzcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
 `

theaudio, 2024-08-09:
Hi @soragaoanakash and @lwfinger , how did you manage with the power setting? 
I'm currently facing the same issue with the RTL8822CE chipset. Adjusting the power using the iw txpower command doesn't work (the command executes without errors, but the power remains the same). 
I tried adjusting it using sar, which allows reducing the power, but I couldn't find a good description of how to use it effectively.
Thanks, Tom

darash, 2024-09-03:
device: awus036ach, module: rtw_8812au

dkms status
> rtw88/r488.764a1ee, 6.10.7-zen1-1-zen, x86_64: installed

lsusb -d 0bda:8812
> Bus 002 Device 002: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter

modinfo rtw_8812au
> filename:       /lib/modules/6.10.7-zen1-1-zen/updates/dkms/rtw_8812au.ko.zst
> ...
> description:   Realtek 802.11ac wireless 8812au driver
> depends:       rtw_8821a,rtw_usb
> name:            rtw_8812au

sudo iw wlan1 set txpower fixed 3000
> command failed: Operation not supported (-95)

**Is the ability to set the txpower option supported  in this module?**

dubhater, 2024-09-04:
@darash It looks like you can't do it that way. The sar thing is the only way supported by this driver. And most likely you can only decrease the TX power.

darash, 2024-09-04:
@dubhater Thanks for the clarification.

wangwillian0, 2024-12-31:
Doesn't seem to work with RTL8812au too. Is this feature possible within the current architecture? if yes, can you suggest where to look for? also, the SAR thing is basically changing the driver itself and hardcode the wanted value?

wangwillian0, 2025-01-01:
Another question: does it make sense to change txpower in monitor mode? mac80211 doesn't seem to allow it now

Im thinking about the possibility of the TX power parameter in the radiotap header to not be fully respected.

dubhater, 2025-01-01:
rtw88 ignores the txpower command from iw, no matter what chip you are using.

The SAR thing is this command from iw:
```
iw phy <phyname> set sar_specs <sar type> <range index:sar power>*
```

sar type can be only 0 (NL80211_SAR_TYPE_POWER) right now.

range index is 0, 1, 2, or 3, depending on which frequency range you are working with:

```
[0] = { .start_freq = 2412, .end_freq = 2484, },
[1] = { .start_freq = 5180, .end_freq = 5320, },
[2] = { .start_freq = 5500, .end_freq = 5720, },
[3] = { .start_freq = 5745, .end_freq = 5825, },
```

And sar power is in units of 0.25 dBm.

You can see the effect of the command via debugfs:

```
# head /sys/kernel/debug/ieee80211/phy1/rtw88/tx_pwr_tbl

channel: 106                                                                                                                                                               
bandwidth: 2                                                                                                                                                               
regulatory: ETSI                                                                                                                                                           
path rate       pwr       base      (byr  lmt  sar ) rem                                                                                                                   
   A  OFDM_6M    26(0x1a)   24    2 (   4    2   63)    0                                                                                                                  
   A  OFDM_9M    26(0x1a)   24    2 (   4    2   63)    0                                                                                                                  
   A  OFDM_12M   26(0x1a)   24    2 (   4    2   63)    0                                                                                                                  
   A  OFDM_18M   26(0x1a)   24    2 (   4    2   63)    0                                                                                                                  
   A  OFDM_24M   26(0x1a)   24    2 (   4    2   63)    0                                                                                                                  
   A  OFDM_36M   26(0x1a)   24    2 (   4    2   63)    0                                                                                                                  

# iw phy1 set sar_specs 0 2:50

# head /sys/kernel/debug/ieee80211/phy1/rtw88/tx_pwr_tbl

channel: 106
bandwidth: 2
regulatory: ETSI
path rate       pwr       base      (byr  lmt  sar ) rem 
   A  OFDM_6M    19(0x13)   24   -5 (   4    2   -5)    0
   A  OFDM_9M    19(0x13)   24   -5 (   4    2   -5)    0
   A  OFDM_12M   19(0x13)   24   -5 (   4    2   -5)    0
   A  OFDM_18M   19(0x13)   24   -5 (   4    2   -5)    0
   A  OFDM_24M   19(0x13)   24   -5 (   4    2   -5)    0
   A  OFDM_36M   19(0x13)   24   -5 (   4    2   -5)    0
```

> Another question: does it make sense to change txpower in monitor mode? mac80211 doesn't seem to allow it now

It makes sense to me to change the TX power in monitor mode if you want to do packet injection.

> Im thinking about the possibility of the TX power parameter in the radiotap header to not be fully respected.

I think rtw88 ignores that completely. I think the only outside influences on the TX power are the kernel's regulatory database, which sets limits, and this sar_specs command, which will not help you exceed the limits.

wangwillian0, 2025-01-01:
Thank you @dubhater 

There seems to be an additional stopper for setting txpower to not even reach the driver, at https://github.com/torvalds/linux/blob/v6.12/net/mac80211/cfg.c#L3056-L3064 related to the WANT_MONITOR_VIF flag. I'm not sure I fully understood the meaning of the flag but I tried enabling it anyway, but I don't have experience with the network drivers and couldnt get the txpower change event to even reach the driver without crashing.

Do you know if an approach like https://github.com/svpcom/rtl8812au/commit/e208e19fa14393849e67d690dc691c5c0d58453d, which adds a module parameter to override the TX, would be acceptable in the kernel? Seems to be the easiest option to implement from my experience of one day.

dubhater, 2025-01-01:
I think something like that would be rejected, because the wireless maintainers don't like module parameters, and they don't like overriding the regulatory limits.

Do you want to lower the TX power or to increase it? Because if you just want to lower it, you can do it right now with the sar_specs command.

wangwillian0, 2025-01-02:
I want to primary lower the TX power, but Im open to try to contribute to make the driver better.

Anyway, I'm trying the command and I don't think I fully understood the values there:
- how did the 50 value became -5?
- what are byr and lmt?
- I can max 42 with rtl8812au at the field `pwr`, does it means a max of 10.5dbm? seems quite low
- iw dev doesn't update with the SAR value change?

i486, 2025-06-13:
Wish I'd found this thread sooner - I was going a bit crazy trying to test `txpower` based on `iw dev` output.

Initial test:
* My card is locked to the 00 (world) regulatory domain, which caps the output at 20 dBm per iw and regdb.
* I created an AP while still in the world domain — `iw dev` showed txpower at 20 dBm.
* The range was weak, just judging by the WiFi bar on my Android phone while moving around the room.
* I then switched to the `US` domain, which showed 30 dBm in `iw dev`, and recreated the AP.
* `iw dev` showed txpower gradually rising and eventually settling at 30 dBm, which aligns with `regdb`.

But the range was still weak.

So I started testing with more care:

* Disabled WiFi throttling on my phones.
* Placed them 6–8 meters away in another room.
* Used the WiFi Analyzer app with faster polling to monitor signal.
* Waited for the RSSI to stabilize.

Then I repeated tests above again. But no noticeable difference in signal strength was observed in the monitor app across any of the regions.

At that point, I started suspecting that the txpower shown in `iw dev` is **not the actual output power**, but merely the *maximum allowed* per `regdb`.

To verify that, I manually decreased the txpower using:

```bash
iw dev wlan0 set txpower fixed 500
```

This set the txpower to 5.00 dBm according to `iw dev`:

```
Interface wlan0
  ...
  txpower 5.00 dBm
```

Yet again, no change in signal strength on the monitoring app.

Then I found this thread, and finally tried:

```bash
iw phy6 set sar_specs 0 3:20
```

The signal dropped instantly in the analyzer app. Finally some real feedback!. It was beautiful to see, lol.

```
A  OFDM_6M    10(0x0a)   28  -18 (   4    8  -18)    0
...
```

My Takeaways (Please correct if anything’s wrong):

1. Don’t trust `txpower` values from `iw dev` — they fluctuate and may mislead you into thinking changes are applied.
2. Don’t use `iw dev wlan0 set txpower` — it has no real effect. https://github.com/lwfinger/rtw88/issues/111#issuecomment-2567159184
3. To actually decrease txpower, use `sar_specs`.
   Multiple desired dBm by **4** to get the `sar_power` value.
   *Example*: To set 12 dBm, use **48**

   ```bash
   iw phy6 set sar_specs 0 3:**48**
   ```
4. To estimate actual transmit power, divide the `pwr` column in `/sys/kernel/debug/ieee80211/phy6/rtw88/tx_pwr_tbl` by 2.
   Not sure how reliable it is or if there is a better method, but it roughly matches the dBm I set via SAR.

Example 1: channel 149, 5 dBm target

```bash
iw phy6 set sar_specs 0 3:20
```

```
A  OFDM_6M    11(0x0b)   ...   → 11 / 2 = ~5.5 dBm (close enough)
```

**Example 2: channel 149, 12 dBm target**

```bash
iw phy6 set sar_specs 0 3:48
```

```
A  OFDM_6M    25(0x19)   ...   → 25 / 2 = ~12.5 dBm (close enough)
```

Also worth noting:
The default maximum txpower on 5GHz (haven’t tested 2.4GHz yet) seems capped at **16 dBm**, regardless of the region setting.

Anyway, I’m happy that I can finally control and verify actual txpower now. Next, I want to try testing higher power (like 20 dBm, which is allowed in my region or any region) across different modes (AP, Monitor, etc.). **@dubhater** if you’ve got any insight on how to do that, I would really appreciate it. Thanks!

morrownr, 2025-09-27:
@cybersnow02 

> putting 'REGDOMAIN=<COUNTRY_2_LETTERS>' in file '/etc/default/crda' is not enough for many linux distro like Debian.

I would say that statement should apply to almost all distros in this day in time as crda was sunset with kernel 4.15 if I remember correctly.

> It is possible that a router return a REG from another country that will cause the TXPower to be reduced without you know what is going on...

I'll just say that older hardware and drivers can be problematic because many different people have been involved in ways to get wifi products in accordance with laws that change a lot.

> But it seems that the 80211h frame will supersede your choice, like I can see in the log:
wlxxxxxxxxxxxxx: Limiting TX power to 20 (20 - 0) dBm as advertised by xx:xx:xx:xx:xx:xx
rtw88_8812au x-x:x.x: failed to get tx report from firmware

Yes, the kernel makes that decision.

> That is a huge problem to me.

Did you mention what country you are in?

My country code is US.

```
$ iw reg get
global
country US: DFS-FCC
(902 - 904 @ 2), (N/A, 30), (N/A)
(904 - 920 @ 16), (N/A, 30), (N/A)
(920 - 928 @ 8), (N/A, 30), (N/A)
(2400 - 2472 @ 40), (N/A, 30), (N/A)
(5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
(5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
(5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
(5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
(5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, PASSIVE-SCAN
(5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR, PASSI
```

When I run the following...

$ iw list

... I get:

```
Frequencies:
* 5180.0 MHz [36] (23.0 dBm)
* 5200.0 MHz [40] (23.0 dBm)
* 5220.0 MHz [44] (23.0 dBm)
* 5240.0 MHz [48] (23.0 dBm)
* 5260.0 MHz [52] (24.0 dBm) (radar detection)
* 5280.0 MHz [56] (24.0 dBm) (radar detection)
* 5300.0 MHz [60] (24.0 dBm) (radar detection)
* 5320.0 MHz [64] (24.0 dBm) (radar detection)
* 5500.0 MHz [100] (24.0 dBm) (radar detection)
* 5520.0 MHz [104] (24.0 dBm) (radar detection)
* 5540.0 MHz [108] (24.0 dBm) (radar detection)
* 5560.0 MHz [112] (24.0 dBm) (radar detection)
* 5580.0 MHz [116] (24.0 dBm) (radar detection)
* 5600.0 MHz [120] (24.0 dBm) (radar detection)
* 5620.0 MHz [124] (24.0 dBm) (radar detection)
* 5640.0 MHz [128] (24.0 dBm) (radar detection)
* 5660.0 MHz [132] (24.0 dBm) (radar detection)
* 5680.0 MHz [136] (24.0 dBm) (radar detection)
* 5700.0 MHz [140] (24.0 dBm) (radar detection)
* 5720.0 MHz [144] (24.0 dBm) (radar detection)
* 5745.0 MHz [149] (30.0 dBm)
* 5765.0 MHz [153] (30.0 dBm)
* 5785.0 MHz [157] (30.0 dBm)
* 5805.0 MHz [161] (30.0 dBm)
* 5825.0 MHz [165] (30.0 dBm)

Frequencies:
* 2412.0 MHz [1] (30.0 dBm)
* 2417.0 MHz [2] (30.0 dBm)
* 2422.0 MHz [3] (30.0 dBm)
* 2427.0 MHz [4] (30.0 dBm)
* 2432.0 MHz [5] (30.0 dBm)
* 2437.0 MHz [6] (30.0 dBm)
* 2442.0 MHz [7] (30.0 dBm)
* 2447.0 MHz [8] (30.0 dBm)
* 2452.0 MHz [9] (30.0 dBm)
* 2457.0 MHz [10] (30.0 dBm)
* 2462.0 MHz [11] (30.0 dBm)
* 2467.0 MHz [12] (disabled)
* 2472.0 MHz [13] (disabled)
* 2484.0 MHz [14] (disabled)
```

I would appreciate it if you would run the above two commands and post the results so that we can compare results.

> If for example I'm far away from an access point, then your adapter might not be able to communicate and establish a connection with the access point because your TX power is set to a lower value (20dBm instead of 30dBm). It might have an impact there. Is that really true?

If there is a problem with the country code then the system may have no choice but to use the global default which will likely reduce the txpower max setting. On the other hand, if the country code is set properly (generally this is automatic) then you will get max allowable txpower accordingly the laws of your country. With 5 GHz., you can see above that I will allowed 24 dBM to 30 dBm depending on which channel I am using. If I can get you to run the above commands that I ran, I will know more.

>  I did some tests and I clearly see that the first attempt is sometime not connecting. I have to retry a second time to get it connected. So, if the REG is UNSET, as soon as it connects, I obtain the REG from the access point, then the txpower is modified to a higher value (20 to 30). I don't like this behavior.

There is a way to handle this situation:

Go to:

https://github.com/morrownr/USB-WiFi

That is the Main Menu of my site. Scroll down to:

12 - Frequently Asked Questions

Open it and see index item 1:

Question: It appears that the Wireless Regulatory information is not correct in my system. How can I fix this?

Go down to the part where you see the following line:

sudo nano /etc/modprobe.d/cfg80211.conf

That section shows you how to set a default country code that should work until you are connected and get the packet from your router.

> I will ask the maintainer for this driver if he could implement such a thing...

See above. 

Let me know what happens. You really want to be using the new rtw88 driver, not the old out-of-kernel driver.

cybersnow02, 2025-10-01:
@dubhater @i486 @wangwillian0 I've discovered something very interesting. If you put:
iw phy phy3 set sar_specs 0 0:-127  (-127 to 19 inclusively, you will get pwr column to 63 (31.5dB))

If you put 20 to 216, you will get a low pwr column with 63 sar column. What all this means?!

kae90, 2026-02-01:
Hi,

I noticed some unusual behavior in the management of SAR limits when changing channels in AP mode.

Measurements without any SAR limits applied:
CH 36: -48 / -51 dBm
CH 48: -47 / -48 dBm
CH 149: -55 / -57 dBm
CH 165: -60 / -63 dBm

I then apply the SAR limits with the following commands:

sudo iw phy2 set sar_specs 0 1:20
sudo iw phy2 set sar_specs 0 2:20
sudo iw phy2 set sar_specs 0 3:20

Result with SAR enabled:
CH 36: -47 / -49 dBm
CH 48: -47 / -48 dBm
CH 149: -67 / -69 dBm
CH 165: -69 / -71 dBm

Further tests show that each call to set sar_specs completely overwrites previous values, rather than adding or combining them.

For example, applying only:

sudo iw phy2 set sar_specs 0 1:20

CH 36 is correctly limited (~-66 dBm).

However, as soon as set sar_specs is executed with a different index (2 or 3), CH 36 reverts to its original values.

This suggests that only the last SAR profile set remains active, while the previous ones are canceled.

My question is:
Is there a supported way to apply SAR limits to all subbands/profiles simultaneously?

Thanks for the support.

kae90, 2026-02-08:
> [@kae90](https://github.com/kae90) If you are using Aircrack driver, then you should be able to change your TXPower using this simple command without modifying the source code each time (for 20dB):
> 
> `iw <your_device_id> set txpower fixed 2000`
> 
> Get your device ID with 'iwconfig' or 'ifconfig'
> 
> I don't know which linux/distribution you are using but if you are using NetworkManager, you could set a dispatcher script that invoke this command every time you connect/disconnect:
> 
> Go in : /etc/NetworkManager/dispatcher.d
> 
> Edit a file there with the name you want and add this: #!/usr/bin/env bash iw <your_device_id> set txpower fixed 2000
> 
> Then, do : chmod 755 <your_script_file>

A while ago I tried the command iw wlan0 set txpower fixed 2000. Unfortunately it had no effect on the driver although the terminal accepts the command and iw wlan0 info correctly indicates 20 dBm, the transmission power is not actually reduced.

I also had to block kernel updates on Parrot to use this driver

cybersnow02, 2026-02-08:
> A while ago I tried the command iw wlan0 set txpower fixed 2000. Unfortunately it had no effect on the driver although the terminal accepts the command and iw wlan0 info correctly indicates 20 dBm, [...]

This mean there is a program or something else that try to reset it to the default regulation. You could try to force the module to ignore the regulation, then try again the `iw` command

Go in /etc/modules.d
Edit 8812au.conf

You will see the line `options 8812au  ....`
append these at the end:  `rtw_tx_pwr_by_rate=0 rtw_tx_pwr_lmt_enable=0`

REBOOT your computer.

Then install 'wavemon' utility to look at the txpower if it remains fixed.