lwfinger/rtw88 #112: 8822bu : Hostapd hotspot is not visible (for most of the wifi cards) in 5Ghz

Samt43, 2023-01-17:
Hi all,
I compiled the rtw88 (14cbcf9a98eb6df4f2a1d3f6c4aeb3732faff0e8) on a raspberrypi with kernel 5.15.88.
The module loads correctly with my 8822bu adaptor, and the adaptor seems to work correctly.

I'm using hostapd to create a 5Ghz hotspot.
Unfortunatly, 5Ghz master mode support seems to not work correctly.
The hotspot starts correctly according to the command line, but remains undetectable (I have 5 devices compatible with 5Ghz, and only one seems to be able to receive the beacons and detect the hotspot). I tried different channels, and it's the same issue for all.
If i put the adaptor to 2.4Ghz, the hotspot is detected by all my devices (although speed seems quite slow).

Also, please note that 5Ghz support works correctly for me using the old 88x2bu driver.
Here are the logs comparing 88x2bu (working) with rtw88 (not detected) for the tested hostapd.conf joined.
[hostapd_logs.zip](https://github.com/lwfinger/rtw88/files/10437142/hostapd_logs.zip)

Any idea ?
Thanks a lot in advance,
Samt43

lwfinger, 2023-01-18:
Have you installed crda and set a country code? The default setting for rtw88sets the "No IR" flag, which prevents Intital Radiation, and blocks broadcasting an SSID.

If you have iw installed, 'iw list' will show the flags on the various 5 GHz channels. Using 'sudo iw reg set XX' will set the country code.

Samt43, 2023-01-26:
Unfortunatly crda was installed already and setting it to FR region did not change anything. Still only one of my device is able to detect the hotspot in 5Ghz mode.
Here is a log of iw list command :
[pi home   rtw88   iw list.txt](https://github.com/lwfinger/rtw88/files/10512612/pi.home.rtw88.iw.list.txt)

lwfinger, 2023-01-26:
The iw list looks normal, and the fact that one of your devices sees the beacons means that the device is transmitting.

I need you to run wireshark on one of your failing Linux hosts, and collect the over-the-air broadcasts. You only need to collect long enough for a couple of beacon intervals. 

If the device that works is also Linux, then collect the wireshark data there as well.

Samt43, 2023-01-27:
Hi @lwfinger !
Here is a wireshark dump on a host (running linux) that does not show the SSID (hacksterwifi2). The hotpot is using the repo rtw88 driver at 14cbcf9a98eb6df4f2a1d3f6c4aeb3732faff0e8.
It seems that in monitor mode, the card is still able to capture some packets !
[capture_wsl1_channel36.zip](https://github.com/lwfinger/rtw88/files/10521953/capture_wsl1_channel36.zip)
I can't give you a capture of the host that is able to connect to the hotspot as it's running android, and root mode is nearly possible anymore on that platform.
Tanks in advance,
Samt43

lwfinger, 2023-01-27:
A very good explanation of the association process is at 
https://documentation.meraki.com/MR/Wi-Fi_Basics_and_Best_Practices/802.11_Association_Process_Explained

Your capture shows the first 3 exchanges from Cisco's diagram. Your station (STA) sends the probe request, the AP answers, and the STA acknoledges the response. The next step would be for the STA to send the authentication open sequence1. It does not, thus there is something in the capabilities offered by the AP that do not match the requirements of the STA. TGhis keeps the association process from advancing.
I will need some time to decode the various fields of the Probe Request and the Probe Response. These are not something I do every day.

lwfinger, 2023-01-27:
One thing that may be helpful is your hostapd.conf. You will need to create a hostapd.conf.txt version to upload it here.

Samt43, 2023-01-27:
[hostapd.conf.txt](https://github.com/lwfinger/rtw88/files/10522945/hostapd.conf.txt)
Here is the hostapd.conf used (expect password).
Please note that the only packets in the capture are exchanged between my phone (that only device that is able to detect) and the AP. After some seconds, my phone can connect.
I'm surprised on my side about the missing beacons that seems to explain the fact i can't detect the ssid correctly...
Please tell me if you need another capture with a simpler configuration. As soon as the AP work at 5Ghz, it remains not detectable by most of the device.

Samt43, 2023-01-29:
I could reproduce the same issue using another linux computer (kernel ubuntu 5.19.0.29) as hotspot (on https://github.com/lwfinger/rtw88/commit/14cbcf9a98eb6df4f2a1d3f6c4aeb3732faff0e8 ) instead of a raspberrypi. This remove possible power issues, that can occurs using a Pi. Playing with all different types of param for beacons, tx power, power management failed. Hotspot remains indetectable by nearly all my wifi cards (except my phone). Seems to be a driver issue.

ValdikSS, 2023-02-04:
I can confirm no-IR flag for 5 GHz network on rtl8811cu on Debian Bookworm. Changing regdb does nothing, unfortunately.
Realtek's https://github.com/morrownr/8821cu-20210916 driver allows to set up AP in 5 GHz.

ValdikSS, 2023-02-04:
I fixed it. First you need to set reg (country), and only then you need to load the kernel module.
Changing reg after the kernel module is loaded would not bring any effect, even if you unplug the Wi-Fi card.

lwfinger, 2023-02-05:
Yes, the reg must be set before the wireless driver runs its initialization code. The way to ensure that is to create a file named /usr/lib/modprobe.d/70-cfg80211.conf and add a line "options cfg80211 ieee80211_regdom=US" to it. Change the US to match your country's code.

closed 2023-02-05 by lwfinger

Samt43, 2023-02-05:
@ValdikSS : Thanks for your fix, I will try to apply the same when I go back home. For now, don't really understand how the hotspot can be visible on my phone without emitting anything !

Samt43, 2023-02-12:
@lwfinger @ValdikSS : 

> Yes, the reg must be set before the wireless driver runs its initialization code. The way to ensure that is to create a file named /usr/lib/modprobe.d/70-cfg80211.conf and add a line "options cfg80211 ieee80211_regdom=US" to it. Change the US to match your country's code.

I did some further tests, being sure to enable the right region before lorading the module, and no this solution does not solve the issue. It seem that with my 8822bu chip, hotspot in 5ghz remain invisible for nearly all my devices.
Maybe we could reopen the issue as it does not solve the issue i'm currently observing ?

Samt43, 2023-05-01:
@lwfinger @ValdikSS 
I compiled latest version 166b8393a0205beaa88f94897c4f59ad6362e054 and try to get AP 5Ghz working with my adapter.
Bus 001 Device 011: ID 0bda:b812 Realtek Semiconductor Corp. RTL88x2bu [AC1200 Techkey]
My phone still always detect the AP and can connect. All of my other devices seems to detect it in some exceptional times (if you are very lucky). At that moment, some can connect, some can not.
Setting power management to off seems to improve the issue a bit, also decrease tx power.
I suppose a clock or a flag is not set correctly in order to correctly broadcast the signal. Some receivers could have more tolerency than other to such issue in the signal.
I have no way to confirm this hypotesis, but be AP seems to not work correctly for RTL88x2bu in 5GHz mode.
Could we reopen the issue as it is not solved and other people may seen it ?

ValdikSS, 2023-05-01:
I have only 8811cu, which is a different chip and different family. My device seem to work fine.

Samt43, 2023-05-01:
@ValdikSS : Yes, i suppose the chip makes a difference here. Mine in a 8822bu, and it doesn't work correctly for sure.

tkuester, 2023-05-01:
> difference here. Mine is a 8822bu, and it doesn't work correctly for sure.

Are you able to check the supported channels with iw, and see if any of
them are marked "No IR"?

Also, I have a handful of 8822bu's I'm sure I could donate to the cause!

Samt43, 2023-05-01:
[iwlist.txt](https://github.com/lwfinger/rtw88/files/11365924/iwlist.txt)
@tkuester  : Yes, none channel is marked No IR. Emission is confirmed by the fact AP is visible constantly by only one of my devices (all of them are compatible with 5Ghz band).

lwfinger, 2023-05-01:
On a Linux system that cannot connect to the AP, run the command
sudo iw dev <device_name> scan | egrep "signal|SSID|freq'

Change the device_name to wlan0, or whatever your wifi is called on that system.

The chip makes a HUGE diufference.

Samt43, 2023-05-02:
Hi everyone, I just compiled and installed wireless-next kernel (6.3.0-rc7+), and I can confirm the same issue is on Linux master branch.
I could not see again the created hotspot on my Linux machine, but I took a screenshot of my phone details, seeing the hotspot correctly. Hope it helps :
[image]

Samt43, 2023-05-02:
Please note that changing channels doesn't make any difference

Samt43, 2023-05-02:
By the way it seems : https://lore.kernel.org/all/1609294366.3846.3.camel@realtek.com/t/
Was an issue near mine, for the 8821AE chip

dubhater, 2024-07-24:
@Samt43 

> @dubhater @morrownr : I tried 4a1ee64475672a9186e10d87ac5e251448f86914 (beacon fix) with a 8822bu device. In my case, unfortunatly i still can't see my hotspot on 5GHz band using an intel wifi card (windows 10) and also my lenovo duet device (chromeos). Using realtek driver, i have no issue. My phone (realtek) and my android box (mediatek) can both see it (for these two devices hotspot seems to work). Not sure your fix fix beacon issue for 8822bu device unfortunatly :'(

Since you already opened this issue about this exact problem, let's discuss it here.

I don't have RTL8822BU, but I tested RTL8811CU, which is from the same generation. It is transmitting beacons. Can you use another device in monitor mode to check if it's transmitting any beacons?

And to make sure you are using the latest commit, unload any rtw* modules and load the newly compiled ones by hand:
```sh
rmmod rtw......
rmmod rtw......
....
insmod ./rtw_core.ko
insmod ./rtw_usb.ko
insmod ./rtw_8822b.ko
insmod ./rtw_8822bu.ko
```