lwfinger/rtw88 #113: Loading driver problem

rubenOrtz, 2023-01-22:
Hey! I have a problem trying to load the driver rtw_8822ce. Just after shutting down. 
Here is dmesg logs:
```
# sudo dmesg | grep rtw
[    3.186335] rtw_8822ce 0000:01:00.0: enabling device (0000 -> 0003)
[    3.206606] rtw_8822ce 0000:01:00.0: Firmware version 9.9.13, H2C version 15
[    3.210595] rtw_8822ce 0000:01:00.0: Firmware version 9.9.4, H2C version 15
[    3.234092] rtw_8822ce 0000:01:00.0: failed to download firmware
[    3.234355] rtw_8822ce 0000:01:00.0: failed to setup chip efuse info
[    3.234358] rtw_8822ce 0000:01:00.0: failed to setup chip information
[    3.262169] rtw_8822ce: probe of 0000:01:00.0 failed with error -22
[  551.567764] rtw_8822ce 0000:01:00.0: Firmware version 9.9.4, H2C version 15
[  551.569080] rtw_8822ce 0000:01:00.0: Firmware version 9.9.13, H2C version 15
[  551.598213] rtw_8822ce 0000:01:00.0: failed to download firmware
[  551.598471] rtw_8822ce 0000:01:00.0: failed to setup chip efuse info
[  551.598472] rtw_8822ce 0000:01:00.0: failed to setup chip information
[  551.629276] rtw_8822ce: probe of 0000:01:00.0 failed with error -22
[  584.127664] rtw_8822ce 0000:01:00.0: Firmware version 9.9.4, H2C version 15
[  584.127759] rtw_8822ce 0000:01:00.0: Firmware version 9.9.13, H2C version 15
[  584.159587] rtw_8822ce 0000:01:00.0: failed to download firmware
[  584.159865] rtw_8822ce 0000:01:00.0: failed to setup chip efuse info
[  584.159866] rtw_8822ce 0000:01:00.0: failed to setup chip information
[  584.181427] rtw_8822ce: probe of 0000:01:00.0 failed with error -22
```

I tried several times restarting the module with modprobe but didn't work. Any suggestions?

rubenOrtz, 2023-01-22:
Also there is some additional info:
```
# lshw -class network
  network UNCLAIMED
       description: Network controller
       product: RTL8822CE 802.11ac PCIe Wireless Network Adapter
       vendor: Realtek Semiconductor Co., Ltd.
       physical id: 0
       bus info: pci@0000:01:00.0
       version: 00
       width: 64 bits
       clock: 33MHz
       capabilities: pm msi pciexpress cap_list
       configuration: latency=0
       resources: ioport:3000(size=256) memory:80600000-8060ffff```

SeanMollet, 2023-07-13:
I'm seeing this same issue with an 8811cu. I've add some telemetry to the drive and am working through it.
I also see a fork having a similar issue with my same chip:
https://github.com/ulli-kroll/rtw88-usb/issues/11

lwfinger, 2023-07-13:
That other thread is 3 years old, and predates the development of the USB drivers for rtw88.

You added a complaint about a USB device onto a thread started for a PCIe device. The liklihood of the same answer applying to both devices is vanishingly small.

Run the command 'sudo dmesg > dmesg.txt' and attach dmesg.txt to a new issue. Also run the command 'lsusb' and attach the output.

SeanMollet, 2023-07-13:
The thread I linked is to a fork of the rtw88 driver pre mainline merge. It's not exactly the same code as this, but it's definitely part of the same family tree. 

As I mentioned, I added a bunch of telemetry to the driver.

It's failing check_hw_ready in download_firmware_validate, which I believe the same problem being discussed in that other thread.

My dmesg is the same as this one and the linked one, except for the telemetry I added to mine.

```
[   28.486954] rtx88: Loading firmware rtw88/rtw8821c_fw.bin
[   28.492907] rtw_8821cu 1-1.5:1.0: Firmware version 24.11.0, H2C version 12
[   28.988012] check_hw_ready failed
[   28.991624] rtx_88 failed in download_firmware_validate
[   28.998626] rtw_8821cu 1-1.5:1.0: failed to download firmware
[   29.012373] rtw_8821cu 1-1.5:1.0: failed to setup chip efuse info
[   29.018749] rtw_8821cu 1-1.5:1.0: failed to setup chip information
[   29.029496] rtw_8821cu: probe of 1-1.5:1.0 failed with error -22
```

SeanMollet, 2023-07-14:
I added an output to check_hw_ready that prints the register value it finds. The board I'm testing on has two radios and conveniently, sometimes one works and the other doesn't.

Looks like bit 15 in REG_MCUFW_CTRL is the magic one. If that's set, everything is good.

[   29.321310] check_hw_ready failed 4078
[   29.326157] rtx_88 failed in download_firmware_validate
[   29.333281] rtw_8821cu 1-1.3:1.0: failed to download firmware
[   29.352831] rtw_8821cu 1-1.3:1.0: failed to setup chip efuse info
[   29.360832] rtw_8821cu 1-1.3:1.0: failed to setup chip information
[   29.367945] rtw_8821cu: probe of 1-1.3:1.0 failed with error -22
[   29.374607] rtx88: Loading firmware rtw88/rtw8821c_fw.bin
[   29.381187] rtw_8821cu 1-1.5:1.0: Firmware version 24.11.0, H2C version 12
[   29.568852] check_hw_ready succeeded c078
[   29.900281] usbcore: registered new interface driver rtw_8821cu
[   29.959648] bcmgenet fd580000.ethernet: configuring instance for external RGMII (RX delay)
[   29.971374] bcmgenet fd580000.ethernet eth0: Link is Down
[   30.008740] rtw_8821cu 1-1.5:1.0 wifiext: renamed from wlan0
[   31.843284] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.
[   33.053107] bcmgenet fd580000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
[   33.061756] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   33.610664] check_hw_ready succeeded c078
[   35.216285] check_hw_ready succeeded c078
[   39.394604] EXT4-fs (mmcblk0p4): mounted filesystem with ordered data mode. Quota mode: none.
[   40.877433] wifiext: authenticate with 26:5a:4c:1f:67:72
[   41.054068] check_hw_ready succeeded c078

lwfinger, 2023-07-14:
As you can see in lines 283 and following in file mac.c, value 0xc048 in REG_MCUFW_CTRL indicates that the firmware is running. Unfortunately, it does not tell us why the firmware load failed.

All of the 802.11ac (Wifi 5) chips have two radios - one for 2.4 GHz and one for 5GHz. My Wifi 5 router uses different SSIDs for the two bands, thus I can control which radio is active.

I just pushed a change that will give us some logging when firmware upload fails. Do a pull and try again.

SeanMollet, 2023-07-14:
Thanks! I added prints to those before and didn't get any errors. 

I believe I have found the culprit. I'm testing a minimized patch now. 

What's the normal flow for patches to this driver? Should I submit it upstream first?

lwfinger, 2023-07-14:
I am glad for the news.

The flow for patches to rtw88 is to prepare them for the wireless-next source and send them to Kalle Vaio (the maintainer for wireless-next), Ping-Ke Shih (the codes author) and linux-wireless@vger.kernel.org. The subject should be "wifi: rtw88: ....". As to other requirements, look at process/submitting-patches.rst in the source tree. Usually I do not pick up such packages for rtw88 until they have been accepted into wireless-next, but I do make exceptions.

Once you have the patch ready, send it to me at Larry.Finger@lwfinger.net. I will check it for problems in formatting, etc.

SeanMollet, 2023-07-14:
Thanks a bunch, I'll put it together this weekend probably and get it submitted.

Carbrex, 2024-03-08:
I am having the same problem, 
```
sudo dmesg | grep rtw
[    2.986397] rtw_8822ce 0000:01:00.0: enabling device (0000 -> 0003)
[    2.986786] rtw_8822ce 0000:01:00.0: firmware: direct-loading firmware rtw88/rtw8822c_fw.bin
[    2.986817] rtw_8822ce 0000:01:00.0: firmware: direct-loading firmware rtw88/rtw8822c_wow_fw.bin
[    2.986955] rtw_8822ce 0000:01:00.0: WOW Firmware version 9.9.4, H2C version 15
[    2.986979] rtw_8822ce 0000:01:00.0: Firmware version 9.9.13, H2C version 15
[    3.012335] rtw_8822ce 0000:01:00.0: failed to download firmware
[    3.012445] rtw_8822ce 0000:01:00.0: failed to setup chip efuse info
[    3.012447] rtw_8822ce 0000:01:00.0: failed to setup chip information
[    3.033710] rtw_8822ce: probe of 0000:01:00.0 failed with error -22
```

```
sudo lshw -C network
  *-network UNCLAIMED       
       description: Network controller
       product: RTL8822CE 802.11ac PCIe Wireless Network Adapter
       vendor: Realtek Semiconductor Co., Ltd.
       physical id: 0
       bus info: pci@0000:01:00.0
       version: 00
       width: 64 bits
       clock: 33MHz
       capabilities: pm msi pciexpress cap_list
       configuration: latency=0
       resources: ioport:3000(size=256) memory:84000000-8400ffff
```
```
lsmod | grep rtw
rtw_8822ce             12288  0
rtw_8822c             491520  1 rtw_8822ce
rtw_pci                32768  1 rtw_8822ce
rtw_core              303104  2 rtw_8822c,rtw_pci
mac80211             1400832  2 rtw_core,rtw_pci
cfg80211             1343488  3 wl,rtw_core,mac80211
```

lwfinger, 2024-03-08:
Do you have a Lenovo or HP laptop? You need to read the section from README.md that describes adding the options to rtw_pci, and do those actions.

Carbrex, 2024-03-08:
Yes, I have an HP Laptop. Ok thanks I will try and revert back.

Carbrex, 2024-03-08:
Can you tell me which part of the readme are you talking about?

lwfinger, 2024-03-08:
Section #4. Option information

Carbrex, 2024-03-08:
I tried adding  `options rtw_pci disable_msi=1 disable_aspm=1` and `options rtw_pci disable_msi=1`
and `options rtw_pci disable_aspm=1` one by one to `/etc/modprobe.d/rtl8822ce.conf` and then reboot. But none of these seem to work. Thanks for your help though.

lwfinger, 2024-03-08:
Run the commands 'sudo modprobe -rv rtw_8822ce' and 'sudo modprobe -v rtw_8822ce'. Post the output of the second command.

Carbrex, 2024-03-08:
```
$ sudo modprobe -v rtw_8822ce
insmod /lib/modules/6.5.0-13parrot1-amd64/kernel/drivers/net/wireless/realtek/rtw88/rtw_pci.ko disable_aspm=1 
insmod /lib/modules/6.5.0-13parrot1-amd64/kernel/drivers/net/wireless/realtek/rtw88/rtw_8822c.ko 
insmod /lib/modules/6.5.0-13parrot1-amd64/kernel/drivers/net/wireless/realtek/rtw88/rtw_8822ce.ko 
```

lwfinger, 2024-03-09:
There is something wrong with your /etc/modprobe.d/rtl8822ce.conf. The load message above only shows that disable_aspm is set, but nothing of disable_msi.

Carbrex, 2024-03-09:
No, I tried all possible configurations as I mentioned above but none works.

Carbrex, 2024-03-09:
```
$ sudo modprobe -v rtw_8822ce
insmod /lib/modules/6.5.0-13parrot1-amd64/kernel/drivers/net/wireless/realtek/rtw88/rtw_pci.ko disable_msi=1 disable_aspm=1 
insmod /lib/modules/6.5.0-13parrot1-amd64/kernel/drivers/net/wireless/realtek/rtw88/rtw_8822c.ko 
insmod /lib/modules/6.5.0-13parrot1-amd64/kernel/drivers/net/wireless/realtek/rtw88/rtw_8822ce.ko 
```
This is the output after adding both disable_msi and disable_aspm

lwfinger, 2024-03-10:
We have one other option: Add this line
options rtw_core disable_lps_deep=1

Then reboot.

Carbrex, 2024-03-10:
I tried adding this to `/etc/modprobe.d/rtl8822ce.conf` but it still doesnt work or I had to put it in any other file. Thanks for your help.

lwfinger, 2024-03-10:
Adding it to /etc/modprobe.d/rtl8822ce.conf is fine.

I am sorry, but I have no more suggestions,

Carbrex, 2024-03-10:
No problem, thanks for your help!