lwfinger/rtw88 #115: `rtw_8723de` frequent disconnections and slow speeds

tshrpl, 2023-02-07:
for the first 5 mins or so, signal hovers around `-50 dBm`, but it disconnects and the list of APs becomes empty (and sometime signal is `0 dBm`), until i restart the laptop or reload the driver, in rare cases it would work for a whole day or two with very slow speeds (<`1mbps` while my phone get <`30mbps` which is my isp limit)

both `rtw_8723de` (this) and `rtw88_8723de` (kernel module) behave the same way, and it was never working before
i have a `rtl8723de` card, its one of those that came with a single antenna, and i had to use `ant_sel=2` with the older driver to get it working, now that parameter is removed, is this responsible for the frequent disconnections ?
i am on `6.0.12-arch1-1` because the latest kernel doesn't boot for me

i don't know what type of log/debug info would be useful, so here's some basic system info
<details>

<summary>

### system information

</summary>

signal is always between -54 to -47, and sometimes 0
```sh
# iwconfig wlo1
wlo1      IEEE 802.11  ESSID:"***Fiber-*****"
          Mode:Managed  Frequency:2.437 GHz  Access Point: A8:DA:0C:D4:74:42
          Bit Rate=65 Mb/s   Tx-Power=20 dBm
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:on
          Link Quality=60/70  Signal level=-50 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:21   Missed beacon:0
```
```sh
# lspci -k
03:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8723DE 802.11b/g/n PCIe Adapter
	DeviceName: WLAN
	Subsystem: Hewlett-Packard Company Device 8319
	Kernel driver in use: rtw_8723de
	Kernel modules: rtw88_8723de, rtw_8723de
```
```
# uname -a
Linux pot 6.0.12-arch1-1 #1 SMP PREEMPT_DYNAMIC Thu, 08 Dec 2022 11:03:38 +0000 x86_64 GNU/Linux
```
```sh
# pacman -Q | ^grep -E "gcc|linux"
gcc 12.2.0-1
gcc-libs 12.2.0-1
lib32-gcc-libs 12.2.1-1
linux 6.0.12.arch1-1
linux-api-headers 5.18.15-1
linux-firmware 20230117.7e4f0ed-1
linux-firmware-whence 20230117.7e4f0ed-1
linux-headers 6.0.12.arch1-1
linux-lts 5.15.90-1
util-linux 2.38.1-1
util-linux-libs 2.38.1-1
```
</details>

lwfinger, 2023-02-07:
This is the kind of problem that I cannot help you with. I have no knowledge on the internals of the card or the rtw88 driver for it.

To help you, I recreated https://github.com/lwfinger/rtl8723de.git so that you can have access to the old driver. It will build on kernels through 6.2.

PmicroTO, 2023-02-22:
> This is the kind of problem that I cannot help you with. I have no knowledge on the internals of the card or the rtw88 driver for it.
> 
> To help you, I recreated https://github.com/lwfinger/rtl8723de.git so that you can have access to the old driver. It will build on kernels through 6.2.

Are this repo's drivers the same ones mainlined at 6.2 kernel? I'm facing a similar problem with an AC1300 8822bu, frequent disconnects and stop working after few minutes.

lwfinger, 2023-02-22:
In general, no. This repo copies the drivers found in the wireless-next repo, which is generally one version ahead of kernel mainline. Thus, we currently have the drivers that will be in kernel 6.3.

The USB drivers are new, and are likely full of problems. The proper place to discuss them is the mailing list at
linux-wireless@vger.kernel.org. You will need to provide details of what is happening, and anything that is logged. Your E-mails MUST be text only. Any html-based message will be rejected.

PmicroTO, 2023-03-20:
Btw, I'm no longer having issues after upgrading to 6.2.2, maybe OP problem was solved too.

tshrpl, 2023-03-20:
doesn't work even after upgrading, but windows 10 works fine

i'll try a fresh install just to be sure

kyoobey, 2023-05-08:
the latest arch kernel doesn't has any issues, wifi works even after 2 days

closed 2023-05-08 by kyoobey