lwfinger/rtw88 #116: rtw_8822ce frequent disconnection issues

TerranFenrir, 2023-02-09:
**Background:**
- OS: Ubuntu 22.04.1 LTS x86_64
- ASUS TUF Gaming A17 FA706IH_
- 5.15.0-60-lowlatency

**Problem:**
- Sometimes wifi disconnects suddenly. It also fails to detect any networks when opened from settings.
- At other times, it does detect networks, but fails to connect to them.
- At times it remains connected to networks without actually having any internet access.

**On running dmesg, here is what I get:**
`================================================================================
[   27.787744] UBSAN: array-index-out-of-bounds in /home/fenrir/rtw88/rtw8822c.c:4449:20
[   27.787745] index 3 is out of range for type 'u8 [2][4]'
[   27.787746] CPU: 6 PID: 276 Comm: kworker/u32:4 Tainted: P        W  OE     5.15.0-60-lowlatency #66-Ubuntu
[   27.787747] Hardware name: ASUSTeK COMPUTER INC. ASUS TUF Gaming A17 FA706IH_FA706IH/FA706IH, BIOS FA706IH.316 03/12/2021
[   27.787748] Workqueue: phy0 rtw_watch_dog_work [rtw_core]
[   27.787753] Call Trace:
[   27.787754]  <TASK>
[   27.787754]  show_stack+0x52/0x5c
[   27.787755]  dump_stack_lvl+0x4a/0x63
[   27.787757]  dump_stack+0x10/0x16
[   27.787759]  ubsan_epilogue+0x9/0x49
[   27.787761]  __ubsan_handle_out_of_bounds.cold+0x44/0x49
[   27.787763]  ? rtw8822c_phy_cck_pd_set+0x40c/0x820 [rtw_8822c]
[   27.787765]  rtw8822c_phy_cck_pd_set+0x5c6/0x820 [rtw_8822c]
[   27.787767]  ? rtw_write32_mask+0x6c/0xa0 [rtw_8822c]
[   27.787771]  rtw_phy_dynamic_mechanism+0x347/0x720 [rtw_core]
[   27.787777]  rtw_watch_dog_work+0x1dd/0x270 [rtw_core]
[   27.787782]  process_one_work+0x222/0x400
[   27.787784]  worker_thread+0x50/0x3f0
[   27.787785]  ? process_one_work+0x400/0x400
[   27.787786]  kthread+0x13b/0x160
[   27.787788]  ? set_kthread_struct+0x50/0x50
[   27.787790]  ret_from_fork+0x22/0x30
[   27.787792]  </TASK>
[   27.787793] ================================================================================
[   27.790852] rtw_8822ce 0000:03:00.0: failed to send h2c command
[   27.793920] rtw_8822ce 0000:03:00.0: failed to send h2c command
[   27.797062] rtw_8822ce 0000:03:00.0: failed to send h2c command
[   28.097975] rtw_8822ce 0000:03:00.0: AMD-Vi: Event logged [IO_PAGE_FAULT domain=0x000c address=0xaef140b0 flags=0x0000]
[   29.828972] rtw_8822ce 0000:03:00.0: failed to send h2c command
[   29.832059] rtw_8822ce 0000:03:00.0: failed to send h2c command
[   29.835154] rtw_8822ce 0000:03:00.0: failed to send h2c command
[   29.838290] rtw_8822ce 0000:03:00.0: failed to send h2c command
[   31.812918] rtw_8822ce 0000:03:00.0: failed to send h2c command
[   31.816009] rtw_8822ce 0000:03:00.0: failed to send h2c command
[   31.819164] rtw_8822ce 0000:03:00.0: failed to send h2c command
[   33.796964] rtw_8822ce 0000:03:00.0: failed to send h2c command
[   33.800052] rtw_8822ce 0000:03:00.0: failed to send h2c command
[   33.903942] rtw_8822ce 0000:03:00.0: failed to send h2c command
[   33.907009] rtw_8822ce 0000:03:00.0: failed to send h2c command
[   35.844931] rtw_8822ce 0000:03:00.0: failed to send h2c command
[   35.848009] rtw_8822ce 0000:03:00.0: failed to send h2c command
[   35.851084] rtw_8822ce 0000:03:00.0: failed to send h2c command
[   35.854209] rtw_8822ce 0000:03:00.0: failed to send h2c command
[   37.828947] rtw_8822ce 0000:03:00.0: failed to send h2c command
[   37.832046] rtw_8822ce 0000:03:00.0: failed to send h2c command
[   37.835174] rtw_8822ce 0000:03:00.0: failed to send h2c command
[   39.813929] rtw_8822ce 0000:03:00.0: failed to send h2c command
[   39.817015] rtw_8822ce 0000:03:00.0: failed to send h2c command
[   39.820153] rtw_8822ce 0000:03:00.0: failed to send h2c command
[   41.796942] rtw_8822ce 0000:03:00.0: failed to send h2c command
[   41.800045] rtw_8822ce 0000:03:00.0: failed to send h2c command
[   41.903932] rtw_8822ce 0000:03:00.0: failed to send h2c command
[   43.844919] rtw_8822ce 0000:03:00.0: failed to send h2c command
[   43.848000] rtw_8822ce 0000:03:00.0: failed to send h2c command
[   43.851059] rtw_8822ce 0000:03:00.0: failed to send h2c command
[   43.854180] rtw_8822ce 0000:03:00.0: failed to send h2c command
[   45.828967] rtw_8822ce 0000:03:00.0: failed to send h2c command
[   45.832056] rtw_8822ce 0000:03:00.0: failed to send h2c command
[   45.835201] rtw_8822ce 0000:03:00.0: failed to send h2c command
[   47.812945] rtw_8822ce 0000:03:00.0: failed to send h2c command
[   47.816020] rtw_8822ce 0000:03:00.0: failed to send h2c command
[   47.819153] rtw_8822ce 0000:03:00.0: failed to send h2c command`

In other times, I also get errors in the dmesg like:
`mac power on failed` or `failed to poll offset something something`.

TerranFenrir, 2023-02-09:
Done. At least I did not receive a timeout error yet. However, I am still getting this error:

[   93.827857] ================================================================================
[   93.827858] UBSAN: array-index-out-of-bounds in /home/fenrir/rtw88/rtw8822c.c:4449:20
[   93.827859] index 3 is out of range for type 'u8 [2][4]'
[   93.827860] CPU: 9 PID: 273 Comm: kworker/u32:4 Tainted: P        W  OE     5.15.0-60-lowlatency #66-Ubuntu
[   93.827861] Hardware name: ASUSTeK COMPUTER INC. ASUS TUF Gaming A17 FA706IH_FA706IH/FA706IH, BIOS FA706IH.316 03/12/2021
[   93.827862] Workqueue: phy0 rtw_watch_dog_work [rtw_core]
[   93.827867] Call Trace:
[   93.827868]  <TASK>
[   93.827868]  show_stack+0x52/0x5c
[   93.827870]  dump_stack_lvl+0x4a/0x63
[   93.827872]  dump_stack+0x10/0x16
[   93.827873]  ubsan_epilogue+0x9/0x49
[   93.827875]  __ubsan_handle_out_of_bounds.cold+0x44/0x49
[   93.827877]  ? rtw8822c_phy_cck_pd_set+0x40c/0x820 [rtw_8822c]
[   93.827880]  rtw8822c_phy_cck_pd_set+0x5c6/0x820 [rtw_8822c]
[   93.827882]  ? rtw_write32_mask+0x6c/0xa0 [rtw_8822c]
[   93.827886]  rtw_phy_dynamic_mechanism+0x347/0x720 [rtw_core]
[   93.827892]  rtw_watch_dog_work+0x1dd/0x270 [rtw_core]
[   93.827897]  process_one_work+0x222/0x400
[   93.827899]  worker_thread+0x50/0x3f0
[   93.827900]  ? process_one_work+0x400/0x400
[   93.827901]  kthread+0x13b/0x160
[   93.827903]  ? set_kthread_struct+0x50/0x50
[   93.827905]  ret_from_fork+0x22/0x30
[   93.827908]  </TASK>
[   93.827908] ================================================================================
[   93.830955] rtw_8822ce 0000:03:00.0: failed to send h2c command
[   93.834070] rtw_8822ce 0000:03:00.0: failed to send h2c command
[   95.808408] rtw_8822ce 0000:03:00.0: failed to send h2c command
[   95.811512] rtw_8822ce 0000:03:00.0: failed to send h2c command
[   95.814671] rtw_8822ce 0000:03:00.0: failed to send h2c command
[   95.827455] rtw_8822ce 0000:03:00.0: timed out to flush pci tx ring[0]
[   95.827510] rtw_8822ce 0000:03:00.0: timed out to flush pci tx ring[1]
[   95.827567] rtw_8822ce 0000:03:00.0: timed out to flush pci tx ring[2]
[   95.827622] rtw_8822ce 0000:03:00.0: timed out to flush pci tx ring[3]
[   95.827678] rtw_8822ce 0000:03:00.0: timed out to flush pci tx ring[5]
[   95.827734] rtw_8822ce 0000:03:00.0: timed out to flush pci tx ring[6]
[   96.431399] rtw_8822ce 0000:03:00.0: failed to send h2c command
[  126.434450] rtw_8822ce 0000:03:00.0: AMD-Vi: Event logged [IO_PAGE_FAULT domain=0x000c address=0xaf0b4a40 flags=0x0000]
[  126.445327] rtw_8822ce 0000:03:00.0: firmware failed to report density after scan
[  127.006403] rtw_8822ce 0000:03:00.0: timed out to flush pci tx ring[1]
[  127.006495] rtw_8822ce 0000:03:00.0: timed out to flush pci tx ring[2]
[  127.006583] rtw_8822ce 0000:03:00.0: timed out to flush pci tx ring[3]
[  127.006670] rtw_8822ce 0000:03:00.0: timed out to flush pci tx ring[5]
[  127.007140] rtw_8822ce 0000:03:00.0: timed out to flush pci tx ring[1]
[  127.007224] rtw_8822ce 0000:03:00.0: timed out to flush pci tx ring[2]
[  127.007318] rtw_8822ce 0000:03:00.0: timed out to flush pci tx ring[3]
[  127.007403] rtw_8822ce 0000:03:00.0: timed out to flush pci tx ring[5]

For some reason, the failed to send h2c is always present whenever the wifi goes down. Is this a driver issue from Realtek's side? Also, thank you so much for helping out haha!

lwfinger, 2023-02-10:
I told you that I had not fixed the UBSAN warning, but that I was contacting Realtek about it.
I wrote the following to him:
=============================================================
One of the users of my rtw88 repo at GitHub has reported a UBSAN error as follows:

UBSAN: array-index-out-of-bounds in /home/fenrir/rtw88/rtw8822c.c:4449:20
index 3 is out of range for type 'u8 [2][4]'

In the statement,
bw = (u8)rtw_read32_mask(rtwdev, 0x9b0, 0xc);

I can see that the 2-bit mask gets values for bw as large as 3, but should the
mask be reduced to a single bit, or should the array dm_info->cck_pd_lv be
increased to [4][4]?
=============================================================

His reply was as follows:
=============================================================

CCK PD is short for CCK packet detection. It should work on 2GHz only, but only
20M or 40M is possible.

static void rtw_phy_cck_pd(struct rtw_dev *rtwdev)
{
	...
	if (rtwdev->hal.current_band_type != RTW_BAND_2G)
		return;
	...
}

Could you share the user scenario you met UBSAN error?
=============================================================

To answer his question, what is your setup? What make/model for your AP? What channel are you using? Please post the output of 'iw list'.

His proposed change would fix the issue, but it should never have happened. Perhaps, we can find the real cause.

TerranFenrir, 2023-02-10:
Hey there!
Firstly, thank you so much for taking interest into solving this issue! Now, regarding my problem, I think I may have found a fix to it/ the origin of the issue. Please let  me know if you think I might be right. I believe that this was a hardware issue. Why do I say so? 

- I have a dual booted system. I logged into Windows to see if I faced wifi problems there too. Turns out, windows was not even detecting my wifi card. 
- I thus, opened my laptop and reinserted the same card. 
- It has been around 17 hours from doing this, and not once have I faced an issue.

I did not immediately report this, as I was not sure if this was a permanent fix (I still am not really). I plan to observe this at least for a week before coming to a concrete conclusion.

As to the iw list output, here it is now:
Wiphy phy0
	wiphy index: 0
	max # scan SSIDs: 4
	max scan IEs length: 323 bytes
	max # sched scan SSIDs: 4
	max # match sets: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Device supports T-DLS.
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP-128 (00-0f-ac:4)
		* CCMP-256 (00-0f-ac:10)
		* GCMP-128 (00-0f-ac:8)
		* GCMP-256 (00-0f-ac:9)
		* CMAC (00-0f-ac:6)
		* CMAC-256 (00-0f-ac:13)
		* GMAC-128 (00-0f-ac:11)
		* GMAC-256 (00-0f-ac:12)
	Available Antennas: TX 0x3 RX 0x3
	Configured Antennas: TX 0x3 RX 0x3
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
		 * mesh point
	Band 1:
		Capabilities: 0x19ef
			RX LDPC
			HT20/HT40
			SM Power Save disabled
			RX HT20 SGI
			RX HT40 SGI
			TX STBC
			RX STBC 1-stream
			Max AMSDU length: 7935 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 2 usec (0x04)
		HT Max RX data rate: 300 Mbps
		HT TX/RX MCS rate indexes supported: 0-15, 32
		Bitrates (non-HT):
			* 1.0 Mbps
			* 2.0 Mbps
			* 5.5 Mbps
			* 11.0 Mbps
			* 6.0 Mbps
			* 9.0 Mbps
			* 12.0 Mbps
			* 18.0 Mbps
			* 24.0 Mbps
			* 36.0 Mbps
			* 48.0 Mbps
			* 54.0 Mbps
		Frequencies:
[... 162 lines cut ...]
		 * mesh point: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-client: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-GO: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-device: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
	Supported RX frame types:
		 * IBSS: 0x40 0xb0 0xc0 0xd0
		 * managed: 0x40 0xb0 0xd0
		 * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * mesh point: 0xb0 0xc0 0xd0
		 * P2P-client: 0x40 0xd0
		 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * P2P-device: 0x40 0xd0
	Supported extended features:
		* [ RRM ]: RRM
		* [ SET_SCAN_DWELL ]: scan dwell setting
		* [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
		* [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
		* [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
		* [ SCAN_RANDOM_SN ]: use random sequence numbers in scans
		* [ CAN_REPLACE_PTK0 ]: can safely replace PTK 0 when rekeying
		* [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
		* [ DEL_IBSS_STA ]: deletion of IBSS station support
		* [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
		* [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support

lwfinger, 2023-02-11:
If I understand the question, you merely reseated the card and it worked better! If that is true, you had a contact problem.

TerranFenrir, 2023-02-21:
Yep! That exactly seems to be the issue. It has been more than a week, and I haven't faced any wifi issues at all. This was clearly a hardware issue, and had nothing to do with this driver. 

Again, thank you so much for helping me and for maintaining this repo! Hope you have a great day! :)

closed 2023-02-21 by TerranFenrir