lwfinger/rtw88 #119: Slow Wifi Speed [RTW8821ce]

mkocagoel, 2023-02-16:
I am running Dualboot and my wifi Speed in Linux is very slow compared to Windows.
I dont know which information i could give currently, so if you need any information i will try to provide information

lwfinger, 2023-02-16:
First, run the following command:
sudo find /sys/kernel/debug/ieee80211 -name rtw88
Copy the patch of that file. Then when running a test through speedtest.net, run the following command:

sudo cat <file from above>/phy_info
Post that info as well as the output of 'iw info'.

mkocagoel, 2023-02-16:
- Output of the Command:
`==========[Common Info]========
Is link = Y
Current CH(fc) = 42
Current BW = 2
Current IGI = 0x2f
TP {Tx, Rx} = {0, 13}Mbps
1SS for TX and RX = N

==========[Tx Phy Info]========
[Tx Rate] =  VHT1SMCS3(0x2f)

==========[Rx Phy Info]========
[Rx Beacon Count] = 19
[Rx Rate] =  VHT1SMCS7(0x33)
[Rx Rate Count]:
 * CCK = {0, 0, 0, 0}
 * OFDM = {19, 0, 0, 0, 0, 0, 0, 0}
 * HT_MCS[0:7] = {0, 0, 0, 0, 0, 0, 0, 0}
 * VHT_MCS-1ss MCS[0:9] = {0, 0, 0, 0, 0, 0, 0, 11, 438, 14}
[RSSI(dBm)] = {-57, -100}
[Rx EVM(dB)] = {-0, -0}
[Rx SNR] = {0, 0}
[CFO_tail(KHz)] = {0, 0}
[Rx Average Status]:
 * OFDM, EVM: {-0}, SNR: {0}
 * 1SS, EVM: {-0}, SNR: {0}
 * 2SS, EVM: {-0, -0}, SNR: {0, 0}
[Rx Counter]:
 * CCA (CCK, OFDM, Total) = (0, 10499, 10499)
 * False Alarm (CCK, OFDM, Total) = (0, 1917, 1917)
 * CCK cnt (ok, err) = (0, 0)
 * OFDM cnt (ok, err) = (7684, 24)
 * HT cnt (ok, err) = (0, 0)
 * VHT cnt (ok, err) = (4245, 561)
`
iw dev interface info Output : 
`Interface wlo1
        ifindex 3
        wdev 0x1
        addr dc:a2:66:7c:68:8b
        ssid Muhi
        type managed
        wiphy 0
        channel 36 (5180 MHz), width: 80 MHz, center1: 5210 MHz
        txpower 20.00 dBm
        multicast TXQ:
                qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                0       0       0       0       0       0       0       0               0
`
Settings i have done:
- Used the given Suspend script
- blacklisted the rtw88 modules

lwfinger, 2023-02-16:
Please repeat the 'sudo cat' command while you are transmitting to speedtest.net. This test showed the bulk of the RX operations were done ar VHT1MCS8, which should be about 170 Mbps. The transmit was at VHT1MCS3, so I thing that was only sending acks. What are the values that speedtest.net shows? What do you get from Windows?

Are you aware that you only have 1 antenna connected?

mkocagoel, 2023-02-16:
Output during Upload:
`==========[Common Info]========
Is link = Y
Current CH(fc) = 42
Current BW = 2
Current IGI = 0x3b
TP {Tx, Rx} = {1, 1}Mbps
1SS for TX and RX = N

==========[Tx Phy Info]========
[Tx Rate] =  VHT1SMCS3(0x2f)

==========[Rx Phy Info]========
[Rx Beacon Count] = 19
[Rx Rate] =  VHT1SMCS9(0x35)
[Rx Rate Count]:
 * CCK = {0, 0, 0, 0}
 * OFDM = {19, 0, 0, 0, 0, 0, 0, 0}
 * HT_MCS[0:7] = {0, 0, 0, 0, 0, 0, 0, 0}
 * VHT_MCS-1ss MCS[0:9] = {0, 0, 0, 0, 0, 0, 0, 0, 18, 116}
[RSSI(dBm)] = {-48, -100}
[Rx EVM(dB)] = {-0, -0}
[Rx SNR] = {0, 0}
[CFO_tail(KHz)] = {0, 0}
[Rx Average Status]:
 * OFDM, EVM: {-0}, SNR: {0}
 * 1SS, EVM: {-0}, SNR: {0}
 * 2SS, EVM: {-0, -0}, SNR: {0, 0}
[Rx Counter]:
 * CCA (CCK, OFDM, Total) = (0, 4673, 4673)
 * False Alarm (CCK, OFDM, Total) = (0, 621, 621)
 * CCK cnt (ok, err) = (0, 0)
 * OFDM cnt (ok, err) = (3023, 3)
 * HT cnt (ok, err) = (0, 0)
 * VHT cnt (ok, err) = (135, 25)
`
During Download
`==========[Common Info]========
Is link = Y
Current CH(fc) = 42
Current BW = 2
Current IGI = 0x3b
TP {Tx, Rx} = {1, 2}Mbps
1SS for TX and RX = N

==========[Tx Phy Info]========
[Tx Rate] =  VHT1SMCS3(0x2f)

==========[Rx Phy Info]========
[Rx Beacon Count] = 20
[Rx Rate] =  VHT1SMCS9(0x35)
[Rx Rate Count]:
 * CCK = {0, 0, 0, 0}
 * OFDM = {20, 0, 0, 0, 0, 0, 0, 0}
 * HT_MCS[0:7] = {0, 0, 0, 0, 0, 0, 0, 0}
 * VHT_MCS-1ss MCS[0:9] = {0, 0, 0, 0, 0, 0, 0, 0, 13, 119}
[RSSI(dBm)] = {-49, -100}
[Rx EVM(dB)] = {-0, -0}
[Rx SNR] = {0, 0}
[CFO_tail(KHz)] = {0, 0}
[Rx Average Status]:
 * OFDM, EVM: {-0}, SNR: {0}
 * 1SS, EVM: {-0}, SNR: {0}
 * 2SS, EVM: {-0, -0}, SNR: {0, 0}
[Rx Counter]:
 * CCA (CCK, OFDM, Total) = (0, 6185, 6185)
 * False Alarm (CCK, OFDM, Total) = (0, 917, 917)
 * CCK cnt (ok, err) = (0, 0)
 * OFDM cnt (ok, err) = (3738, 3)
 * HT cnt (ok, err) = (0, 0)
 * VHT cnt (ok, err) = (126, 23)
`
- On windows im getting around 100Mb download and 20Mb Upload
- On Linux im getting around 15Mb download and 2Mb Upload  

No i was not aware of the fact that i am only using one antenna. I am using a Laptop so i did not think of that kind of things

lwfinger, 2023-02-17:
Send me the output of 'iw list' as well.
When I connect my 8852be, a wifi 6 device, to my wifi 5 router, it drives the lines at the full speed of my 500 Mbps broadband connection. The TX and RX speeds are VHT2SSMC9.

mkocagoel, 2023-02-17:
`Wiphy phy0
        wiphy index: 0
        max # scan SSIDs: 4
        max scan IEs length: 2243 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Device supports T-DLS.
        Supported Ciphers:
                * WEP40 (00-0f-ac:1)
                * WEP104 (00-0f-ac:5)
                * TKIP (00-0f-ac:2)
                * CCMP-128 (00-0f-ac:4)
                * CCMP-256 (00-0f-ac:10)
                * GCMP-128 (00-0f-ac:8)
                * GCMP-256 (00-0f-ac:9)
                * CMAC (00-0f-ac:6)
                * CMAC-256 (00-0f-ac:13)
                * GMAC-128 (00-0f-ac:11)
                * GMAC-256 (00-0f-ac:12)
        Available Antennas: TX 0x1 RX 0x1
        Configured Antennas: TX 0x1 RX 0x1
        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
                 * mesh point
        Band 1:
                Capabilities: 0x196e
                        HT20/HT40
                        SM Power Save disabled
                        RX HT20 SGI
                        RX HT40 SGI
                        RX STBC 1-stream
                        Max AMSDU length: 7935 bytes
                        DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: 2 usec (0x04)
                HT Max RX data rate: 150 Mbps
                HT TX/RX MCS rate indexes supported: 0-7, 32
                Bitrates (non-HT):
                        * 1.0 Mbps
                        * 2.0 Mbps
                        * 5.5 Mbps
                        * 11.0 Mbps
                        * 6.0 Mbps
                        * 9.0 Mbps
                        * 12.0 Mbps
                        * 18.0 Mbps
                        * 24.0 Mbps
                        * 36.0 Mbps
                        * 48.0 Mbps
                        * 54.0 Mbps
                Frequencies:
                        * 2412 MHz [1] (20.0 dBm)
                        * 2417 MHz [2] (20.0 dBm)
                        * 2422 MHz [3] (20.0 dBm)
                        * 2427 MHz [4] (20.0 dBm)
                        * 2432 MHz [5] (20.0 dBm)
                        * 2437 MHz [6] (20.0 dBm)
                        * 2442 MHz [7] (20.0 dBm)
                        * 2447 MHz [8] (20.0 dBm)
                        * 2452 MHz [9] (20.0 dBm)
                        * 2457 MHz [10] (20.0 dBm)
                        * 2462 MHz [11] (20.0 dBm)
                        * 2467 MHz [12] (20.0 dBm)
[... 139 lines cut ...]
                 * mesh point: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
                 * P2P-client: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
                 * P2P-GO: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
                 * P2P-device: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
        Supported RX frame types:
                 * IBSS: 0x40 0xb0 0xc0 0xd0
                 * managed: 0x40 0xb0 0xd0
                 * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
                 * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
                 * mesh point: 0xb0 0xc0 0xd0
                 * P2P-client: 0x40 0xd0
                 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
                 * P2P-device: 0x40 0xd0
        Supported extended features:
                * [ RRM ]: RRM
                * [ SET_SCAN_DWELL ]: scan dwell setting
                * [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
                * [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
                * [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
                * [ SCAN_RANDOM_SN ]: use random sequence numbers in scans
                * [ CAN_REPLACE_PTK0 ]: can safely replace PTK 0 when rekeying
                * [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
                * [ DEL_IBSS_STA ]: deletion of IBSS station support
                * [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
                * [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support`

lwfinger, 2023-02-17:
I do not have a fix yet, but I have found a symptom.

I installed an 8821CE chip in my computer and tested the 5 GHz band to my various routers. The results are as follows:

Router level     Rx Speed     Tx Speed
Wifi 4                108 Mbps     87
Wifi 5                  15                1.4
Wifi 6                  36                37

For some reason, the 802.11ac chip performs worst when connected to an 802.11ac AP! It seems that the driver is not handling the VHT speeds properly. If you have access to a Wifi 4 router, then you can improve your performance. In the meantime, I will be reporting this to Realtek. I will let you know when I have an update.

mkocagoel, 2023-02-20:
Thanks for reaching out to Realtelk. 
How do you know that i am only using one Antenna?

lwfinger, 2023-02-22:
I am making some progress. I booted openSUSE Leap 15.4 and Leap 15.5, both of which use a 5.14.21 kernel. I have not been able to get either to fail using the rtw88 driver built into the kernel. Under Leap 15.4 and using a 5GHz connection, I get the following:

Wifi 6 365/262 Mbs Rx/Tx
Wifi 5 156/119 Mbps
Wifi 4 119/107 Mbps

I expected there to be a bigger difference between Wifi 4 and Wifi 5, but that is a smaller problem.

What kernel are you using? The command 'uname -r' will give that info.

mkocagoel, 2023-02-22:
6.1.13-1-lts

lwfinger, 2023-02-22:
It appears that we have a regression before kernel 6.1.X. I will try to bisect it.

Thanks for the info.

mkocagoel, 2023-02-22:
Already thanks in advance.

lwfinger, 2023-02-23:
I found the problem. It was a merge error in my copying of the kernel code to this repo. You should be able to do a new pull and get code that works properly.

closed 2023-02-23 by mkocagoel