lwfinger/rtw88 #120: rtw88_8723de prevents system boot on 6.1.* kernels

cyberdynesoftware, 2023-02-20:
Booting a computer with a RTL8723DE wifi adapter with a 6.1.* kernel doesn't succeed. The boot process freezes at some point and then the computer simply reboots. Blacklisting rtw88_8723de prevents this; without rtw88_8723de, the boot succeeds normally, but no wifi. Manually loading rtw88_8723de with modprobe after the boot brings back wifi; this isn't very stable though. Sometimes the system freezes when loading the module.

This problem did not occur with 6.0.* kernels. Everything used to work normal. The upgrade to 6.1.* kernels introduced the problem.

lwfinger, 2023-02-21:
I do not have an RTL8723DE chip, thus I have no way to debug the proglem. You will need to help me.

Is there a difference between rtw88_8723de (the kernel version), and rtw_8723de (the version from this repo), which is newer code, and probably fixed a lot of things? This repo has code that will be in kernel 6.3 in about 2 months.

Keep rtw88_8723de blacklisted, build and install the drivers from this repo. If it does not help, then you need to run the command 'sudo dmesg > dmesg.txt' and attach dmesg.txt to this issue. Once I see that, it may suggest other things to do/try.

cyberdynesoftware, 2023-02-21:
I will do my very best to help you figure this out. Unfortunately I'm not big on the kernel and stuff, just a regular linux user. I haven't tried the version from this repo yet. My concern is, how can I make sure that it's the version from this repo that is loaded and not the one shipped with the kernel?

Since the driver shipped with the kernel works most of the time when I load it manually after boot, the real test would be to make the kernel load the driver from this repo while booting. How can I do that?

lwfinger, 2023-02-21:
My previous messaage told you how to tell the difference between the drivers - they have DIFFEREBT names. Note: I hate to repeat myself!!!

cyberdynesoftware, 2023-02-25:
Apologies. I did not notice the missing 88 in the name of the driver from this repo. Even after your last message I had to stare at the names for quite a moment to find the difference.

When the driver from this repo gets into the kernel, will the name then be without the 88? Or will the driver in the kernel retain the 88 and the missing 88 in this repo kind of identify the developement version?

Anyway, there is no difference with the driver version from this repo. The boot does not succeed. The dmesg that follows is from a boot with kernel 6.1.12 with both rtw88_8723de and rtw_8723de blacklistet, after loading rtw_8723de manually:

[dmesg2.txt](https://github.com/lwfinger/rtw88/files/10829829/dmesg2.txt)

I don't mind repeating myself and I want to emphasize again, that there was no problem with kernel versions 6.0.* Maybe this could be an important clue in order to find the problem.

lwfinger, 2023-02-25:
These warnings are scary, but apparently not the problem:

[  174.038638] ACPI Warning: Time parameter 255 us > 100 us violating ACPI spec, please fix the firmware. (20220331/exsystem-141)
[  185.095612] ACPI Warning: Time parameter 255 us > 100 us violating ACPI spec, please fix the firmware. (20220331/exsystem-141)
[  196.152709] ACPI Warning: Time parameter 255 us > 100 us violating ACPI spec, please fix the firmware. (20220331/exsystem-141)
[  207.201579] ACPI Warning: Time parameter 255 us > 100 us violating ACPI spec, please fix the firmware. (20220331/exsystem-141)

As root, create a file /usr/lib/modprobe.d/70-rtw88.conf and add the following lines:
options rtw88_core disable_msi=y disable_aspm=y
options rtw_core disable_msi=y disable_aspm=y

These two options will compensate for a faulty BIOS that is regularly seen in newer laptops.

You misunderstand the relationship between this repo and the kernel. This repo copies the latest code from the wireless-next repo, which currently has kernel 6.3 code, and modifies it so that it will build on older kernels. This code is not destined to ever be part of the kernel - it is already there. The difference in the module names is deliv=berate so that I can look at a log file, and determine which version a user is running.

cyberdynesoftware, 2023-02-25:
I created the file as instructed and booted the system with rtw88_8723de blacklisted and rtw_8723de not blacklisted. No difference. The boot did not succeed.

Thank you for the clarification of the purpose of this repo. Does this mean that the rtw88_8723de driver is not separate form the kernel? There is no dedicated repo from which the kernel pulls?

lwfinger, 2023-02-25:
There are a number of development projects from which the kernel source is pulled. For wireless, the wireless-next repo contains the material that will be in the next kernel. About 3 weeks ago, the material for kernel 6.3 was pushed to net-next, which was pushed to linux-next. That last repo is the one that Linus uses to get the new material.

The rtw88 drivers are built into the kernel, as are many other wireless drivers.

When the computer freezes on boot, and then reboots, is the caps lock light flashing?

Do you have another Linux system on your network?

cyberdynesoftware, 2023-02-25:
I'm trying to figure out, how the process of updating the kernel works and specifically how an update to the kernel could make a previously working driver prevent the system from booting.

No, caps lock does not flash when the computer freezes.

Yes, I do have other linux systems on my network. I did try to ping the laptop in question during boot but to no avail.

Interestingly in one of my tries the system did boot with rtw_8723de not blacklisted. According to lsmod the module was loaded, but no wifi. When I boot the system with rtw_8723de blacklisted and load the driver manually that does give me wifi.

su-id, 2023-03-17:
this issue should be solved by this commit:
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/drivers/net/wireless/realtek/rtw88?id=4a267bc5ea8f159b614d0549030216d0434eccca

has been backported to 6.1.x and 6.2.x

lwfinger, 2023-03-17:
That patch has been in this repo for as long it has been in the mainline kernel.

closed 2023-03-17 by lwfinger