lwfinger/rtw88 #121: RTW_8821ce no Signal after resume

mkocagoel, 2023-02-23:
After i Resume, my Wifi card is not finding any Connections anymore. I have to shutdown and boot again. Reboot doesnt work
[image]

My Blacklisted file:
options rtw_pci disable_aspm=1
options rtw_pci disable_msi=1

blacklist rtw88_8821ce
blacklist rtw88_8821c
blacklist rtw88_pci
blacklist rtw88_core

My sleep File:
"
cat /lib/systemd/system-sleep/rtw88.sleep 
"
#!/bin/sh
if [ "${1}" == "pre" ]; then
modprobe -r rtw_8822ce
echo "we are suspending at $(date)..." > /tmp/systemd_suspend_test
elif [ "${1}" == "post" ]; then
modprobe rtw_8822ce
fi

lwfinger, 2023-02-23:
Your options entries need to have rtw_pci after the "options" part. The syntax is "options <module_name> <details of option>. As you are not specifying a module, your lines are being ignored. As your photo shows, your BIOS is not changing power state. fixing that should help.

Normally, the "options" entries are not placed in a blacklist file, but given a separate file named something like "70_rtw8821ce.conf" in /he same directory as the blacklist file.

Look at your sleep file carefully. Why are you specifying a different device than you are using?

mkocagoel, 2023-02-23:
I am sry but i dont understand the point with options
Am i not using it correctly with
options (rtw_pci as module name)

lwfinger, 2023-02-23:
Sorry, I misread your lines. My comments about putting the options in a different file from the blacklists does stand.

mkocagoel, 2023-10-24:
so anything new?

mkocagoel, 2023-10-24:
I changed it and still get this:

`
Oct 18 08:44:16 muhiLinLap kernel: pcieport 0000:00:1c.4: Unable to change power state from D3hot to D0, device inac>
Oct 18 08:44:16 muhiLinLap kernel: pci 0000:02:00.0: Unable to change power state from unknown to D0, device inacces>
Oct 18 08:44:16 muhiLinLap kernel: nvidia 0000:01:00.0: Enabling HDA controller
Oct 18 08:44:16 muhiLinLap kernel: ACPI: EC: event unblocked

rtw_8821ce 0000:02:00.0: Unable to change power state from D3cold to D0, device i>
Oct 18 08:44:16 muhiLinLap kernel: rtw_8821ce 0000:02:00.0: Firmware version 24.11.0, H2C version 12
Oct 18 08:44:16 muhiLinLap kernel: rtw_8821ce 0000:02:00.0: mac power on failed
Oct 18 08:44:16 muhiLinLap kernel: rtw_8821ce 0000:02:00.0: failed to power on mac
Oct 18 08:44:16 muhiLinLap kernel: rtw_8821ce 0000:02:00.0: failed to setup chip efuse info
Oct 18 08:44:16 muhiLinLap kernel: rtw_8821ce 0000:02:00.0: failed to setup chip information
insmod /lib/modules/6.1.58-1-lts/updates/dkms/rtw_pci.ko.zst disable>
Oct 18 08:44:16 muhiLinLap systemd-sleep[4280]: insmod /lib/modules/6.1.58-1-lts/updates/dkms/rtw_8821c.ko.zst
Oct 18 08:44:16 muhiLinLap systemd-sleep[4280]: insmod /lib/modules/6.1.58-1-lts/updates/dkms/rtw_8821ce.ko.zst
Oct 18 08:44:16 muhiLinLap kernel: rtw_8821ce: probe of 0000:02:00.0 failed with error -114
`