lwfinger/rtw88 #125: Works perfectly on HP s15-eq1032ns!!

gbr000, 2023-03-08:
I'm just leaving this note I think it may be useful for others,

Notebook: [HP s15-eq1032ns](https://support.hp.com/us-en/document/c07043871)

What I have tried so far: all other options in internet with no success!
Issues: I had to restart 3x times because a black screen pos reboot, but after that the wifi seems to work perfectly!
Note: Also it could be that I was just a bit precipitated and didn't allowed the reboot.
After: The wifi it's working perfectly, the signal and it recognized all the networks, I still need to test the velocity.

Thanks to the developer!

jothi-prasath, 2023-05-28:
i am using the same wifi on [HP 14s-fq1030AU](https://support.hp.com/in-en/document/c07119618) 
the wifi works fine out of the box without any errors on both  archlinux/fedora 38 
i have some random disconnect issues in 2.4G but works perfectly in 5G which i mostly use
i have noticed that the wifi using more power #129 and that can be fixed by disabling the `SUPPORTS_DYNAMIC_PS`

lwfinger, 2023-05-28:
The power issue should now be fixed in this repo, but not yet in the kernel version.

closed 2023-05-28 by lwfinger