lwfinger/rtw88 #126: Failed on Linux 6.2.2-arch1-1

IsaacJReay, 2023-03-08:
Hello, this is my first time report an issue, so let me know if i haven't provided anything that you need. My system is Archlinux with latest everything. It has been failing since linux 6.2. 

Output of make
`Sign command: /usr/lib/modules/6.2.2-arch1-1/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Creating symlink /var/lib/dkms/rtw88/r191.4bede29/source -> /usr/src/rtw88-r191.4bede29

Building module:
Cleaning build area...(bad exit status: 2)
make -j8 KERNELRELEASE=6.2.2-arch1-1 -C /usr/lib/modules/6.2.2-arch1-1/build M=/var/lib/dkms/rtw88/r191.4bede29/build...(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.2.2-arch1-1 (x86_64)
Consult /var/lib/dkms/rtw88/r191.4bede29/build/make.log for more information.
Sign command: /usr/lib/modules/6.2.2-arch1-1/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...(bad exit status: 2)
make -j8 KERNELRELEASE=6.2.2-arch1-1 -C /usr/lib/modules/6.2.2-arch1-1/build M=/var/lib/dkms/rtw88/r191.4bede29/build...(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.2.2-arch1-1 (x86_64)
Consult /var/lib/dkms/rtw88/r191.4bede29/build/make.log for more information.`

Ouput of cat /var/lib/dkms/rtw88/r191.4bede29/build/make.log
```bash
DKMS make.log for rtw88-r191.4bede29 for kernel 6.2.2-arch1-1 (x86_64)
ថ្ងៃ​ពុធ ថ្ងៃ  8 ខែ មីនា ឆ្នាំ 2023, 10 ម៉ោ 47 នាទី 45 វិនាទី​
  CC [M]  /var/lib/dkms/rtw88/r191.4bede29/build/main.o
  CC [M]  /var/lib/dkms/rtw88/r191.4bede29/build/mac80211.o
  CC [M]  /var/lib/dkms/rtw88/r191.4bede29/build/util.o
  CC [M]  /var/lib/dkms/rtw88/r191.4bede29/build/debug.o
In file included from <command-line>:
././include/linux/kconfig.h:5:10: fatal error: generated/autoconf.h: No such file or directory
    5 | #include <generated/autoconf.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [scripts/Makefile.build:252: /var/lib/dkms/rtw88/r191.4bede29/build/main.o] Error 1
make[1]: *** Waiting for unfinished jobs....
  CC [M]  /var/lib/dkms/rtw88/r191.4bede29/build/tx.o
In file included from <command-line>:
././include/linux/kconfig.h:5:10: fatal error: generated/autoconf.h: No such file or directory
    5 | #include <generated/autoconf.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [scripts/Makefile.build:252: /var/lib/dkms/rtw88/r191.4bede29/build/mac80211.o] Error 1
  CC [M]  /var/lib/dkms/rtw88/r191.4bede29/build/rx.o
In file included from <command-line>:
././include/linux/kconfig.h:5:10: fatal error: generated/autoconf.h: No such file or directory
    5 | #include <generated/autoconf.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [scripts/Makefile.build:252: /var/lib/dkms/rtw88/r191.4bede29/build/debug.o] Error 1
In file included from <command-line>:
././include/linux/kconfig.h:5:10: fatal error: generated/autoconf.h: No such file or directory
    5 | #include <generated/autoconf.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~
In file included from <command-line>:
././include/linux/kconfig.h:5:10: fatal error: generated/autoconf.h: No such file or directory
    5 | #include <generated/autoconf.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
compilation terminated.
make[1]: *** [scripts/Makefile.build:252: /var/lib/dkms/rtw88/r191.4bede29/build/tx.o] Error 1
make[1]: *** [scripts/Makefile.build:252: /var/lib/dkms/rtw88/r191.4bede29/build/util.o] Error 1
In file included from <command-line>:
././include/linux/kconfig.h:5:10: fatal error: generated/autoconf.h: No such file or directory
    5 | #include <generated/autoconf.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [scripts/Makefile.build:252: /var/lib/dkms/rtw88/r191.4bede29/build/rx.o] Error 1
make: *** [Makefile:2021: /var/lib/dkms/rtw88/r191.4bede29/build] Error 2```

Kernel Information: `Linux 6.2.2-arch1-1 #1 SMP PREEMPT_DYNAMIC Fri, 03 Mar 2023 15:58:31 +0000 x86_64 GNU/Linux`
Kernel Headers: The same version as above
System: Archlinux

I think i have to wait, right? But i don't see anyone report an error in this kernel version, so here i am

lwfinger, 2023-04-14:
Note that I do not maintain the dkms stuff; however, these are the statements that the Makefile runs when 'make clean' is done:

clean:
        @rm -fr *.mod.c *.mod *.cmd *.o .*.cmd *.ko *~ .*.o.d .cache.mk
        @rm -fr .tmp_versions
        @rm -fr Modules.symvers
        @rm -fr Module.symvers
        @rm -fr Module.markers
        @rm -fr modules.order

In addition, I do not have a generated directory. This issue is clearly not with my code!

closed 2023-04-14 by lwfinger

Loader009, 2023-04-14:
Thanks - I already left a comment in the ArchLinux AUR package - https://aur.archlinux.org/packages/rtw88-dkms-git

Loader009, 2023-04-14:
@lwfinger wait a sec - you do the following:
`CLEAN="make -C $kernel_source_dir clean"`
https://github.com/lwfinger/rtw88/blob/master/dkms.conf#L4

The file I am talking about is: 
`/usr/lib/modules/6.2.10-zen1-1-zen/build/include/generated/autoconf.h`

You basically do a make clean on the kernel modules directory, which would be:
`KSRC := /lib/modules/$(KVER)/build`
https://github.com/lwfinger/rtw88/blob/master/Makefile#L3

Is that correct?

PRESFIL, 2023-04-16:
@Loader009, thank you for your attentiveness. I saw your message on AUR.
It seems you have already found the reason. I think I need to create a new issue.

lwfinger, 2023-04-16:
@Loader009: If the problem you see is in the DKMS code, please let me explain once again: I do not support DKMS. I have not written any of the DKMS code. I only apply patches to the DKMS code that are supplied to me. If you have a fix, supply such a patch!.

Loader009, 2023-04-16:
Understood - it should be fixed by removing `-C $kernel_source_dir` form `CLEAN="make -C $kernel_source_dir clean"`
But honestly - this is more like trial and error as I have never modified dkms stuff.