lwfinger/rtw88 #127: please check the makefile again

ohquait, 2023-03-14:
see the bold lines on middle !!!

obj-m	+= rtw_8821ce.o
rtw_8821ce-objs		:= rtw8821ce.o

**obj-m	   += rtw_8821cs.o
rtw_8821cs-objs		:= rtw8822bs.o**

obj-m	+= rtw_8821cu.o
rtw_8821cu-objs		:= rtw8821cu.o

i think the 8821cs.o should not including a 8822bs.o ???

lwfinger, 2023-03-14:
You're right. That was a copy-and-paste error. There was also a missing include statement in rtw8822cs.c. Now fixed.

ohquait, 2023-03-14:
always happy for you ... ;)

ohquait, 2023-03-14:
@lwfinger 

this was nothing :(

make.log

>   LD [M]  /var/lib/dkms/rtlwifi-rtw88/0.1.2-4pclos2023/build/rtw_8723du.o
  CC [M]  /var/lib/dkms/rtlwifi-rtw88/0.1.2-4pclos2023/build/rtw8821c.o
  CC [M]  /var/lib/dkms/rtlwifi-rtw88/0.1.2-4pclos2023/build/rtw8821c_table.o
  LD [M]  /var/lib/dkms/rtlwifi-rtw88/0.1.2-4pclos2023/build/rtw_8821c.o
  CC [M]  /var/lib/dkms/rtlwifi-rtw88/0.1.2-4pclos2023/build/rtw8821ce.o
  LD [M]  /var/lib/dkms/rtlwifi-rtw88/0.1.2-4pclos2023/build/rtw_8821ce.o
  CC [M]  /var/lib/dkms/rtlwifi-rtw88/0.1.2-4pclos2023/build/rtw8821cs.o
In file included from /var/lib/dkms/rtlwifi-rtw88/0.1.2-4pclos2023/build/rtw8821cs.c:5:
/var/lib/dkms/rtlwifi-rtw88/0.1.2-4pclos2023/build/rtw8821cs.c:13:29: error: ‘SDIO_VENDOR_ID_REALTEK’ undeclared here (not in a function); did you mean ‘SDIO_VENDOR_ID_MEDIATEK’?
   13 |                 SDIO_DEVICE(SDIO_VENDOR_ID_REALTEK,
      |                             ^~~~~~~~~~~~~~~~~~~~~~
./include/linux/mmc/sdio_func.h:95:20: note: in definition of macro ‘SDIO_DEVICE’
   95 |         .vendor = (vend), .device = (dev)
      |                    ^~~~
/var/lib/dkms/rtlwifi-rtw88/0.1.2-4pclos2023/build/rtw8821cs.c:14:29: error: ‘SDIO_DEVICE_ID_REALTEK_RTW8821CS’ undeclared here (not in a function); did you mean ‘SDIO_DEVICE_ID_MEDIATEK_MT7961’?
   14 |                             SDIO_DEVICE_ID_REALTEK_RTW8821CS),
      |                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
./include/linux/mmc/sdio_func.h:95:38: note: in definition of macro ‘SDIO_DEVICE’
   95 |         .vendor = (vend), .device = (dev)
      |                                      ^~~
make[2]: *** [scripts/Makefile.build:252: /var/lib/dkms/rtlwifi-rtw88/0.1.2-4pclos2023/build/rtw8821cs.o] Error 1
make[1]: *** [Makefile:2000: /var/lib/dkms/rtlwifi-rtw88/0.1.2-4pclos2023/build] Error 2
make[1]: Leaving directory '/usr/src/kernel-devel-6.2.6-pclos1'
make: *** [Makefile:114: all] Error 2

i think there are faulty links in or to headers or other buggy's ...
i check it and hope can do more for that
...

ohquait, 2023-03-14:
SOLVED, 

`--- a/rtw8821cs.c
+++ b/rtw8821cs.c
@@ -2,10 +2,10 @@
 /* Copyright(c) Martin Blumenstingl <martin.blumenstingl@googlemail.com>
  */
 
+#include "sdio.h"
 #include <linux/mmc/sdio_func.h>
-#include <linux/mmc/sdio_ids.h>
+#include "sdio_ids.h"
 #include <linux/module.h>
-#include "sdio.h"
 #include "rtw8821c.h"
 
 static const struct sdio_device_id rtw_8821cs_id_table[] =  {
`

lwfinger, 2023-03-14:
File sdio_ids.h gets updated along with the addition of files rtw88xxys.c. To be able to build on kernels without this fix, I created a standalone header sdio_ids.h. You have the right fix. I just pushed it.