lwfinger/rtw88 #128: Monitor Mode Stability Issues

tkuester, 2023-03-16:
Hey! I posted this to the linux wireless mailing list, but I didn't get much of a response. Hopefully it's not too forward of me to cross post this here, as well.

I am experimenting with an 8822bu dongle, and doing a packet capture on monitor mode similar to kismet. (I have a simple bash loop doing channel hopping in one window including HT40 and 80 MHz channels, and tcpdump in another.) After a varying window of time (between 1 to 10 hours) the monitor interface simply stops receiving packets. There are no output messages in the kernel otherwise indicating there is a problem.

Cycling the interface doesn't resolve the situation, I have to rmmod the driver, and then re-modprobe it. This allows me to get packets again, but the packets stop in another few hours.

I am running this on a BeagleBone (Debian 10, Linux 4.19) and have just setup a second copy on an STM32MP1 (Debian 11, Linux 5.15). I am running the driver from lwfinger's GitHub repository at 4bede29. I have cross compiled the drivers with cpp arm-linux-gnueabihf 4:8.3.0-1 (Debian). As of today, the driver has been able to run for 72 hours on Linux 5.15 without issue, so maybe I just need to upgrade the kernel.

Any ideas on what could be causing this, or steps to help troubleshoot / provide more information?

lwfinger, 2023-03-16:
I saw your posting on the linux wireless mailing list. The author of the USB code would also have seen it, but clearly neither of us have anything to say about your problem. I think it quite likely that there are upgrades to the USB subsystem between 4.19 and 5.15. As the development has been done on a 6.1 or newer kernel, I think you should upgrade, perhaps to even newer than 5.15.

tkuester, 2023-03-16:
Sorry to trouble you, and thank you for the response! It was my first time submitting to the mailing list, I wasn't sure if no replies meant my mail bounced off inboxes for some reason, or because no one had input. I'll close the issue here.

I still have a question about the "No IR" designation that's being reported by `iw phy`, should I try the mailing list again, or is here preferred?

closed 2023-03-16 by tkuester

lwfinger, 2023-03-16:
The "No IR" is a regulatory issue, not one with the driver.