lwfinger/rtw88 #129: rtw882ce more power usage

jothi-prasath, 2023-03-24:
when i was using kernel 5.19.13 the wifi power usage was avg 272 mW
now in kernel 6.2.8 the wifi power usage was 579 mW

lwfinger, 2023-03-24:
A lot of features that give higher throuput have been enabled between 5.19 and 6.2. The code from this repo was unable to use them as the underlying kernel had no support.

Are you getting higher throuput?

jothi-prasath, 2023-03-24:
This was 5.19.13 one
[image]
This was 6.2.8 one
[image]

lwfinger, 2023-03-24:
Run an iperf or speedest.net trial and tell me what the usage is under wifi load.

jothi-prasath, 2023-03-24:
At the middle of the speedtest
[image]

after speedtest at idle
[image]

jothi-prasath, 2023-05-25:
i found that this commit https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/net/mac80211/mlme.c?h=v6.3.3&id=28977e790b5d0e6d37db6e659cfbcde9d24ae718 is the culprit which introduces `SUPPORTS_DYNAMIC_PS` which disable powersave in rtw8821ce.

i revert this patch on kernel 6.3.4 and boom the power usage is low as desired.
[image]

lwfinger, 2023-05-25:
Thanks for finding the cause of this issue. I have sent E-mail to make certain that the author of the patch looks at it again and verifies that it is what he intended. If so, then rtw88 is failing to inplement  SUPPORTS_DYNAMIC_PS correctly.

Once I hear from the author, then I will report back.

lwfinger, 2023-05-25:
I just pushed a change that will disable SUPPORTS_DYNAMIC_PS while we find the bug in the drivers. With this change, you will not need to patch the kernel source unless you use the in-kernel driver.

lwfinger, 2023-05-26:
The Realtek team announced that they are preparing patches for rtw88 and rtw89 to fix this problem. I will let you know when I merge those patches.

lwfinger, 2023-05-27:
I just pushed Realtek's patch with SUPPORTS_DYNAMIC_PS turned on again. Please test with an unmodified kernel and let me know if it fixes your problem. Thanks.

jothi-prasath, 2023-05-28:
yeah i installed on 6.3.4-arch1-1 and it fixed

closed 2023-05-28 by lwfinger