lwfinger/rtw88 #130: Frequent Spontaneous Disconnects

kaganAhmetOkan, 2023-04-02:
I've had this issue since forever. I have `RTL8821CE` in my ASUS Laptop. The WIFI only works with rtw88 package, which I have installed through PARU in Arch Linux.

The WIFI likes to disconnect suddenly and has trouble reconnecting. Restarting the WIFI or NetworkManager sometimes works. And I see this line getting spammed in **journalctl**:
`Vivobook15-Arch wpa_supplicant[495]: wlan0: CTRL-EVENT-BEACON-LOSS`

I also noticed this line getting spammed sometimes:
`Vivobook15-Arch kernel: [UFW BLOCK] IN=wlan0 OUT= MAC=01:00:5e:xx:xx:xx:xx:xx:xx:xx:xx:xx:xx:xx SRC=192.168.1.1 DST=224.0.0.1 LEN=36 TOS=0x00 PREC=0x00 TTL=1 ID=0 DF PROTO=2`

I tried disabling the firewall to see what would happen and the WIFI seemed able to reconnect, though I don't know if it's a coincidence. What more should I provide to identify the problem?

lwfinger, 2023-04-02:
I recently did a long-term test with the RTW8821CE running the code from this repo. I hade no disconnects over a 4-day period, thus I cannot reproduce your problem.

You are getting some Unconplicated FirewWall (UFW) blocks. As I have never used that firewall, I can offer no support for that issue. Until you figure out how to configure it properly, I would leave it turned off.

jolesh, 2023-04-23:
Hej,
I have a similar (or the same) problem. First with Linux Mint 20, which I have been using for years, so I though that I might messed sth up during that time and created a new distro with a fresh Linux Mint 21.1. I think the issue only appears after my PC being suspended and woken up again, so I though it might be [this issue](https://github.com/lwfinger/rtw88/issues/56), but the solution which [kelebek333](https://github.com/kelebek333) posted didn't work.

### System Information
#### Linux Mint
Release Linux Mint 21.1 Vera 64-bit
Kernel Linux 5.15.0-56-generic x86_64
MATE 1.26.0 

#### Hardware
Memory: 22,9 GiB
Processor: AMD Ryzen 5 4600H with Radeon Graphics × 12 
Graphics: RENOIR (renoir, LLVM 15.0.6, DRM 3.42, 5.15.0-70-generic)

### Error messages

`sudo dmesg | grep rtw` after start up after being suspended
```
[ 9324.746597] rtw_8822ce 0000:03:00.0: stop vif 70:66:55:6c:54:4d on port 0
[ 9326.746711] rtw_8822ce 0000:03:00.0: failed to poll offset=0x5 mask=0x2 value=0x0
[ 9333.559106] rtw_8822ce 0000:03:00.0: failed to poll offset=0x5 mask=0x2 value=0x0
[ 9333.559120] rtw_8822ce 0000:03:00.0: mac power on failed
[ 9333.559123] rtw_8822ce 0000:03:00.0: failed to power on mac
[ 9335.561857] rtw_8822ce 0000:03:00.0: failed to poll offset=0x5 mask=0x2 value=0x0
[ 9335.561867] rtw_8822ce 0000:03:00.0: mac power on failed
[ 9335.561869] rtw_8822ce 0000:03:00.0: failed to power on mac
[ 9337.563399] rtw_8822ce 0000:03:00.0: failed to poll offset=0x5 mask=0x2 value=0x0
[ 9337.563408] rtw_8822ce 0000:03:00.0: mac power on failed
[ 9337.563410] rtw_8822ce 0000:03:00.0: failed to power on mac
[ 9350.002553] rtw_8822ce 0000:03:00.0: failed to poll offset=0x5 mask=0x2 value=0x0
[ 9350.002568] rtw_8822ce 0000:03:00.0: mac power on failed
[ 9350.002573] rtw_8822ce 0000:03:00.0: failed to power on mac
[ 9352.003297] rtw_8822ce 0000:03:00.0: failed to poll offset=0x5 mask=0x2 value=0x0
[ 9352.003312] rtw_8822ce 0000:03:00.0: mac power on failed
[ 9352.003316] rtw_8822ce 0000:03:00.0: failed to power on mac
[ 9364.004020] rtw_8822ce 0000:03:00.0: failed to poll offset=0x5 mask=0x2 value=0x0
[ 9364.004036] rtw_8822ce 0000:03:00.0: mac power on failed
[ 9364.004041] rtw_8822ce 0000:03:00.0: failed to power on mac
[ 9366.004874] rtw_8822ce 0000:03:00.0: failed to poll offset=0x5 mask=0x2 value=0x0
[ 9366.004889] rtw_8822ce 0000:03:00.0: mac power on failed
[ 9366.004892] rtw_8822ce 0000:03:00.0: failed to power on mac
[ 9378.003881] rtw_8822ce 0000:03:00.0: failed to poll offset=0x5 mask=0x2 value=0x0
[ 9378.003897] rtw_8822ce 0000:03:00.0: mac power on failed
[ 9378.003901] rtw_8822ce 0000:03:00.0: failed to power on mac
[ 9380.004569] rtw_8822ce 0000:03:00.0: failed to poll offset=0x5 mask=0x2 value=0x0
[ 9380.004584] rtw_8822ce 0000:03:00.0: mac power on failed
[ 9380.004587] rtw_8822ce 0000:03:00.0: failed to power on mac
```

jolesh, 2023-04-23:
Might also be related to #115

lwfinger, 2023-04-23:
@jolesh - I think your problem is different. I just pushed some updates. Do a 'git pull' and remake. Does this help?

jolesh, 2023-04-23:
@lwfinger Nice, I really hope it will work. I haven't cloned the repro so I can't simply pull and remake it. I just run a normal new Linux Mint 21.1 System.

lwfinger, 2023-04-23:
If you do that, it is unlikely that you will get the new one. It will take some time for Mint to get the new code. Do the following:

git clone https://github.com/lsfinger.git
cd rtw88
make
sudo make install

That will get you the latest code.

jolesh, 2023-04-23:
Nice, thanks. Where will it clone it to? Because when I do `cd rtw88` I get the error message: `bash: cd: rtw88: No such file or directory`

lwfinger, 2023-04-23:
A typo in the git wommand:
git clone https://lwfinger/rtw88.git

You should have seen an error message!

jolesh, 2023-04-24:
Now I see an error message: 
```
Cloning into 'rtw88'...
fatal: unable to access 'https://lwfinger/rtw88.git/': Could not resolve host: lwfinger
```

jolesh, 2023-04-24:
`make` gives me this error message
```
make -C /lib/modules/5.15.0-70-generic/build M=/home/jolesh/rtw88 modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-70-generic'
  CC [M]  /home/jolesh/rtw88/main.o
/home/jolesh/rtw88/main.c: In function ‘rtw_check_sta_active_iter’:
/home/jolesh/rtw88/main.c:2487:16: error: ‘struct ieee80211_vif’ has no member named ‘cfg’
 2487 |         if (vif->cfg.assoc || !is_zero_ether_addr(rtwvif->bssid))
      |                ^~
make[2]: *** [scripts/Makefile.build:297: /home/jolesh/rtw88/main.o] Error 1
make[1]: *** [Makefile:1907: /home/jolesh/rtw88] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-70-generic'
make: *** [Makefile:114: all] Error 2
```

lwfinger, 2023-04-24:
I thought I had built against kernel 5.15, but apparently not. I just pushed a fix for the error you found, and one that would have appeared later.

jolesh, 2023-04-25:
Thanks a lot. It seems like the install worked now. I will let you know about how well it works. And if I forget, then it is probably because it works without problems. 
Thanks a lot for your time, help and patience.

jolesh, 2023-04-29:
Okay, so I tried it for a few days now. For more or less one day it was working fine but now it is the same as before. However I am not sure if sth went wrong with the install or if it didn't fix (the source of) the problem or only so temporarily. I also have problems with Bluetooth (again).

lwfinger, 2023-04-29:
Did you update your kernel, and have dkms install an older version. If you reinstall from the rtw88 directory, does that help? If not, that is about as far as I can go. Read the last paragraph of README.md on where to send more inquiries, but be aware of two things: (1) That mailing list will reject html mail. It needs to be plain text. (2) Include all related messages from the dmesg log. An incomplete description of the problem will be ignored.

jolesh, 2023-05-03:
Hm, nevermind. I just tried to install it again with the same commands, made a kernal upgrade and rebooted and for the last 2 or 3 days it has been working perfectly fine.