lwfinger/rtw88 #132: will this driver support rtl8723bs and rtl8189es ?

ChalesYu, 2023-04-08:
Hi,
I know `rtl8xxxu` driver has been support  `rtl8723bu` and `rtl8188eu/etv`.
but some hardware changed interface from  `USB interface`  to  `SDIO interface` ,  name is `rtl8723bs` and `rtl8189es` .

So, I want to ask the `rtl8189es` and `rtl8723bs` will get support on which driver , on this `rtw88` driver?

lwfinger, 2023-04-08:
First of all, rtw88 is for Wifi 5 (802.11ac) devices. As far as I know, rtl8723bs is a Wifi 4 device. I have no idea what family to which rtl8189es belongs. Neither of these devices are driven by rtw88. The repo at https://github.com/hadess/rtl8723bs will handle that device using the vendor driver, which is not as clean as something like rtw88, but it likely works. There is an old repo at https://github.com/openwetek/rtl8189es for that device. I have no SDIO devices nor any knowledge of either driver.

closed 2023-04-08 by lwfinger

ChalesYu, 2023-04-15:
but it seems a little wired ,  rtw88 driver code looks supported  rtl8723de and rtl8723du ,it is a 80211n(WiFi 4) devices , but don't know if the real hardware works good.

Here are new version modified vendor driver for rtl8723bs , rtl8189es in my repo:
https://github.com/ChalesYu/rtl8822bs-aml/tree/rtl8188e_usb_sdio_5.13.3-17-test
https://github.com/ChalesYu/rtl8822bs-aml/tree/rtl8723b_usb_sdio_5.8.4-test

I don't have rtl8723b devices, so can not test.

These SDIO interface devices are always come with a host platform board, like tablet , TV-Box and SBC . This interface is not as universal as USB interface.

I guess support will appear at `rtl8xxxu` driver ,  someone has been write some half-complete code.
https://github.com/gamelaster/rtl8xxx

lwfinger, 2023-04-16:
The 8723d devices are wifi 5 - 8723b devices are wifi 4. RTW88 ONLY handles wifi 5!