lwfinger/rtw88 #141: make install does not work

GrobDebian, 2023-05-11:
Hello I followed the tutorial, but encountered an error when using: make install or make sign-install

```
make -C /lib/modules/5.10.0-22-amd64/build M=/home/robin/rtw88 modules
make[1] : entering directory « /usr/src/linux-headers-5.10.0-22-amd64 »
make[1] : leaving directory « /usr/src/linux-headers-5.10.0-22-amd64 »
Generating a RSA private key
..+++++
....+++++
writing new private key to 'MOK.priv'
-----
input password: 
input password again: 
make: depmod: No such file or directory
make: *** [Makefile:124 : install] Error 127
```
Could you help me?

lwfinger, 2023-05-11:
Your system does not have the utility depmod installed. On every distro I have used, it is part of the mandatory packages, but apparently not on yours. Is there a forum where you can ask how to add it?

GiyoMoon, 2023-05-11:
`depmod` is usually in the `kmod` package. Try installing it.
https://man7.org/linux/man-pages/man8/kmod.8.html

GrobDebian, 2023-07-16:
Hello,

Sorry for the delay of my response.
I tried to install kmod, but it is already installed on my computer :open_mouth:

closed 2023-07-16 by GrobDebian