lwfinger/rtw88 #145: openSUSE MicroOS support

knamroud, 2023-06-02:
Basically the title says everything.
Has anyone managed to install and run this on MicroOS?
What is the cleanest/safest way to do so?

lwfinger, 2023-06-02:
Enter the command 'sudo transactional-update shell'. Then while in that shell, do the following:
zypper install git make gcc kernel-devel kernel-default-devel
git clone https://github.com/lwfinger/rtw88.git
cd rtw88
make
make install

With your favorite editor, open /usr/lib/modules.d/70-rtw88.conf and add a line that says
blacklist rtw88_xxxxyy

Where xxxx is the device number, and yy is the suffux such as be, etc. Thus the result would be rtw88_8822be, for example. This is to keep the code built into the openSUSE kernel from interfering.

When this is done, 'exit' from the privileged shell and reboot. The drivers will be there when the machine comes up. Those read-only root file systes are a pita. Are you sure it is worth it?

knamroud, 2023-06-04:
Is there any way to make this persistent and not having to do this at every kernel update?

lwfinger, 2023-06-04:
When the kernel changes, all modules must be rebuilt - both internal and external; however, the process is somewhat simpler to rebuild after the initial installation.

sudo transactional-update shell
cd rtw88
git pull
make
make install
exit

Then reboot.

knamroud, 2023-06-18:
Hey, sorry to answer after closing the issue. 
Do you know by chance if the same is somehow possible on Silverblue?

lwfinger, 2023-06-19:
I did a bit of research and found the following:

"If the drivers are not in the upstream kernel or are not available as an RPM package, it is currently almost if not impossible to install any 3rd party drivers due to the read-only file system of SIlverblue." It seems that they do not provide anything like the transactional-update shell.

It seems that you would need to make an RPM of the sources to get anywhere. This step does not look complicated, but I have no interest is setting that up.