lwfinger/rtw88 #150: Failed to download firmware on RPI2 ArchLinux Arm

henkv1, 2023-06-19:
Building and installing the latest Git version works fine on my Raspberry Pi 2 running Arch Linux Arm. But when I load the module I get the following errors in my Kernel log:

[  696.449879] rtw_core: loading out-of-tree module taints kernel.
[  696.471367] rtw_8821cu 1-1.4:1.0: Firmware version 24.11.0, H2C version 12
[  696.653675] rtw_8821cu 1-1.4:1.0: error beacon valid
[  696.659289] rtw_8821cu 1-1.4:1.0: failed to download rsvd page
[  696.665864] rtw_8821cu 1-1.4:1.0: failed to download firmware
[  696.677470] rtw_8821cu 1-1.4:1.0: failed to setup chip efuse info
[  696.683622] rtw_8821cu 1-1.4:1.0: failed to setup chip information
[  696.690414] rtw_8821cu: probe of 1-1.4:1.0 failed with error -16

My system runs kernel 6.1.34 armv7:
[root@alarmpi rtw88]# uname -a
Linux alarmpi 6.1.34-1-rpi-ARCH #1 SMP Fri Jun 16 12:46:24 MDT 2023 armv7l GNU/Linux

Can you please help?

henkv1, 2023-06-19:
I have the same issue on Arch Linux x86_64 6.3.8-arch1-1:
[  251.030576] usb 1-3: new high-speed USB device number 6 using xhci_hcd
[  251.171302] usb 1-3: New USB device found, idVendor=0bda, idProduct=b812, bcdDevice= 2.10
[  251.171315] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  251.171321] usb 1-3: Product: USB3.0 802.11ac 1200M Adapter
[  251.171324] usb 1-3: Manufacturer: Realtek
[  251.171327] usb 1-3: SerialNumber: 123456

[  488.797488] rtw_8822bu 1-3:1.0: Firmware version 27.2.0, H2C version 13
[  488.850706] rtw_8822bu 1-3:1.0: error beacon valid
[  488.850829] rtw_8822bu 1-3:1.0: failed to download rsvd page
[  488.850982] rtw_8822bu 1-3:1.0: failed to download firmware
[  488.852575] rtw_8822bu 1-3:1.0: failed to setup chip efuse info
[  488.852576] rtw_8822bu 1-3:1.0: failed to setup chip information
[  488.852646] rtw_8822bu: probe of 1-3:1.0 failed with error -16
[  488.852672] usbcore: registered new interface driver rtw_8822bu

The adapter works fine with the rtl88x2bu driver.

henkv1, 2023-06-20:
I did some bisecting and it looks like 4b34fac2bce94f2f02f3dd4af066226c919b62c8 (wifi: rtw88: use struct instead of macros to set TX desc) is the first 'bad' commit.

lwfinger, 2023-06-20:
That is useful information. I will check that my implementation was correct, and then report it to Realtek.

lwfinger, 2023-06-20:
Realtek has issued a V2 of that patch, and I did not include the changes here. Please pull again and retry. Thanks for your help.

henkv1, 2023-06-21:
Unfortunately, this did not fix the issue.  791a7db90f752e75855faca3fe5bd82494dbdda6 is still the last commit I managed to get working.

henkv1, 2023-06-21:
Reverting 4b34fac2bce94f2f02f3dd4af066226c919b62c8 using the attached patch makes the adapters work with the latest commit. So there is an issue with 4b34fac2bce94f2f02f3dd4af066226c919b62c8 on my machines. Does anyone else experience the same?
[revert-4b34fac.patch.txt](https://github.com/lwfinger/rtw88/files/11815160/revert-4b34fac.patch.txt)

Samt43, 2023-06-21:
@henkv1 : i'm experiencing exactly the same on a Pi 2 with a rtl8822bu.
On master, 
rtw_8822bu 1-3:1.0: error beacon valid
rtw_8822bu 1-3:1.0: failed to download rsvd page
rtw_8822bu 1-3:1.0: failed to download firmware

It was working before.
Also, after applying your patch on master, i can confirm my dongle is now working again as it was some weeks ago, with the bug https://github.com/lwfinger/rtw88/issues/112 still not solved.

Thanks !

lwfinger, 2023-06-21:
I just pushed a version that works on my laptop with an RTW8822BU. I hope it works for you.. Sorry for any inconvenience.

henkv1, 2023-06-22:
I can confirm that the issue is fixed. Thank you.

closed 2023-06-22 by henkv1

giuliopaci, 2023-06-27:
Hi, I have a similar issue, but with different hardware (Realtek 8822ce).

In particular dmesg reports:
```
rtw_8822ce 0000:03:00.0: firmware: direct-loading firmware rtw88/rtw8822c_fw.bin
rtw_8822ce 0000:03:00.0: Firmware version 9.9.13, H2C version 15
rtw_8822ce 0000:03:00.0: firmware: direct-loading firmware rtw88/rtw8822c_wow_fw.bin
rtw_8822ce 0000:03:00.0: WOW Firmware version 9.9.4, H2C version 15
rtw_8822ce 0000:03:00.0: failed to download firmware
rtw_8822ce 0000:03:00.0: failed to setup chip efuse info
rtw_8822ce 0000:03:00.0: failed to setup chip information
rtw_8822ce: probe of 0000:03:00.0 failed with error -22
```

@lwfinger do you think it could be the same or similar issue?

lwfinger, 2023-06-27:
I'm not sure. What does 'md5sum /lib/firmware/rtw88/*' report? Copy and paste that output here.

lwfinger, 2023-06-27:
You have the latest firmware. Let me dig out my rtw8822ce chip and see what happens here.

lwfinger, 2023-06-27:
The rtw8822ce driver from this repo works on my computer. I noticed your errors in the Debian Bugzilla:

[   22.086224] rtw_8822ce 0000:03:00.0: PCIe Bus Error: severity=Uncorrected (Non-Fatal), type=Transaction Layer, (Requester ID)
[   22.088074] rtw_8822ce 0000:03:00.0:   device [10ec:c822] error status/mask=00004000/00400000
[   22.089135] rtw_8822ce 0000:03:00.0:    [14] CmpltTO                (First)
[   22.090186] rtw_8822ce 0000:03:00.0: AER: can't recover (no error_detected callback)
[   22.091271] pcieport 0000:00:1d.1: AER: device recovery failed

I found one E-mail thread discussing the CmpltTO error. They were able to eliminate it by doing the following:

Set iommu in bypass mode via bootargs iommu.passthrough=1

Please try that to see if it helps.

giuliopaci, 2023-06-27:
I will check that option. At the moment I got rid of those errors by using pcie_aspm=off, but the high instability persists.

lwfinger, 2023-06-28:
I just updated the repo to match the latest wireless-next and pushed the results.  I did not find anything that should have interfered with loading the firmware. I suspect that whatever is interfering with PCIe operations is also causing a problem with loading the firmware.

giuliopaci, 2023-06-28:
I tried the iommu.passthrough=1 option, but it did not improve the situation, so it seems that only pcie_aspm=off allows me to get rid.of AER errors in dmesg.

On the other hand with your last push I was able to load the module and the firmware seems to correctly load now.
Unfortunately the connection is still unstable and when it hangs I see
```
rtw_8822ce 0000:03:00.0: firmware failed to leave lps state
```

sometimes followed by

```
rtw_8822ce 0000:03:00.0: failed to send h2c command
```

However I think that now we can exclude that I am affected by this issue.