lwfinger/rtw88 #153: Make fails on Ubuntu 22.04.2 with Linux 5.19.0-46 generic

serpastorg, 2023-07-11:
Make fails this way in my fresh installed Ubuntu:

`make -C /lib/modules/5.19.0-46-generic/build M=/home/sergio/Descargas/rtw88 modules
make[1]: se entra en el directorio '/usr/src/linux-headers-5.19.0-46-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc (Ubuntu 11.3.0-1ubuntu1~22.04.1) 11.3.0
  You are using:           gcc (Ubuntu 11.3.0-1ubuntu1~22.04.1) 11.3.0
  CC [M]  /home/sergio/Descargas/rtw88/main.o
/home/sergio/Descargas/rtw88/main.c: In function ‘rtw_core_deinit’:
/home/sergio/Descargas/rtw88/main.c:2302:9: error: implicit declaration of function ‘timer_delete_sync’ [-Werror=implicit-function-declaration]
 2302 |         timer_delete_sync(&rtwdev->tx_report.purge_timer);
      |         ^~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:257: /home/sergio/Descargas/rtw88/main.o] Error 1
make[1]: *** [Makefile:1857: /home/sergio/Descargas/rtw88] Error 2
make[1]: se sale del directorio '/usr/src/linux-headers-5.19.0-46-generic'
make: *** [Makefile:117: all] Error 2
`

Please help me, I need to use my RTL8821 dongle. Thanks a lot

lwfinger, 2023-07-12:
I found the problem and pushed a fix. Do a 'git pull', and it should build again.

serpastorg, 2023-07-12:
Solved. Thank you very much. I was surprised with your answer. It was soooo soon :)

lwfinger, 2023-07-18:
Your gcc with version 5.5.0 is too old. You need version 11.3.0 from Ubuntu as that is the compiler used to build the kernel.