lwfinger/rtw88 #154: WIFI still not working after make install and reboot Huawei MateBook D14

GrobDebian, 2023-07-16:
Hello,

I have a Huawei MateBook D14 2020.

I followed the instructions to install the rtw88 (without module signing for SecureBoot). But wifi is still not working after reboot.
Do you know why?

lwfinger, 2023-07-16:
No, but your logs might. Try 'sudo dmesg | grep rtw' to see what it says. Also check 'lsmod | grep rtw' to verify that your modules are loaded.

Are you using Secure Boot? If you are, the modules must be signed.

GrobDebian, 2023-07-20:
I am not able to use sudo on Debian, but with "su" then ’dmesg | grep rtw’ nothing appears.
’lsmod | grep rtw’ gives no response too.

lwfinger, 2023-07-24:
Obviously, you are not getting the driver loaded. First of all, run the command 'lspci -nn' and post the result here. That will let us know what device you have, and whether it is in the tables.

closed 2025-02-06 by dubhater