lwfinger/rtw88 #157: WLAN0 had shown but network is down

briansune, 2023-07-25:
wlan0: flags=4098<BROADCAST,MULTICAST>  mtu 1500
        ether ff:ff:ff:ff:ff:ff  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0     Interface doesn't support scanning : Network is down

sit0      Interface doesn't support scanning.

lo        Interface doesn't support scanning.

Had tried nmcli radio wifi on and restart service.

Module                  Size  Used by
rtw_8723ds             16384  0
rtw_8723d              45056  1 rtw_8723ds
rtw_sdio               20480  1 rtw_8723ds
rtw_core              151552  2 rtw_sdio,rtw_8723d

lwfinger, 2023-07-25:
Bringing your interface up is not the driver's responsibility. That is done by your user-space network control software. What are you using? I recommend NetworkManager. It Just Works.

briansune, 2023-07-25:
Let me attach the dmesg:

[   99.221393] rtw_core: loading out-of-tree module taints kernel.
[  110.501543] rtw_8723ds mmc1:0001:1: Firmware version 48.0.0, H2C version 0

RadxaYuntian, 2023-11-02:
I'm also having this issue. The wired thing is I did have this working before on 8723DS :/ I also have `ff:ff:ff:ff:ff:ff` as MAC address, and I tried both `iwd` and `wpa_supplicant` as NetworkManager backend on Debian 12. None of them helps.

<details>
<summary>iwd</summary>
radxa@rock-pi-s:~$ sudo journalctl -u iwd | cat
Nov 02 17:37:25 rock-pi-s systemd[1]: Starting iwd.service - Wireless service...
Nov 02 17:37:26 rock-pi-s iwd[538]: Wireless daemon version 2.3
Nov 02 17:37:26 rock-pi-s iwd[538]: Loaded configuration from /etc/iwd/main.conf
Nov 02 17:37:26 rock-pi-s systemd[1]: Started iwd.service - Wireless service.
Nov 02 17:37:26 rock-pi-s iwd[538]: station: Network configuration is disabled.
Nov 02 17:37:26 rock-pi-s iwd[538]: Wiphy: 0, Name: phy0
Nov 02 17:37:26 rock-pi-s iwd[538]:         Permanent Address: ff:ff:ff:ff:ff:ff
Nov 02 17:37:26 rock-pi-s iwd[538]:         2.4Ghz Band:
Nov 02 17:37:26 rock-pi-s iwd[538]:                 Bitrates (non-HT):
Nov 02 17:37:26 rock-pi-s iwd[538]:                          1.0 Mbps
Nov 02 17:37:26 rock-pi-s iwd[538]:                          2.0 Mbps
Nov 02 17:37:26 rock-pi-s iwd[538]:                          5.5 Mbps
Nov 02 17:37:26 rock-pi-s iwd[538]:                         11.0 Mbps
Nov 02 17:37:26 rock-pi-s iwd[538]:                          6.0 Mbps
Nov 02 17:37:26 rock-pi-s iwd[538]:                          9.0 Mbps
Nov 02 17:37:26 rock-pi-s iwd[538]:                         12.0 Mbps
Nov 02 17:37:26 rock-pi-s iwd[538]:                         18.0 Mbps
Nov 02 17:37:26 rock-pi-s iwd[538]:                         24.0 Mbps
Nov 02 17:37:26 rock-pi-s iwd[538]:                         36.0 Mbps
Nov 02 17:37:26 rock-pi-s iwd[538]:                         48.0 Mbps
Nov 02 17:37:26 rock-pi-s iwd[538]:                         54.0 Mbps
Nov 02 17:37:26 rock-pi-s iwd[538]:                 HT Capabilities:
Nov 02 17:37:26 rock-pi-s iwd[538]:                         HT40
Nov 02 17:37:26 rock-pi-s iwd[538]:                         Short GI for 20Mhz
Nov 02 17:37:26 rock-pi-s iwd[538]:                         Short GI for 40Mhz
Nov 02 17:37:26 rock-pi-s iwd[538]:                 HT RX MCS indexes:
Nov 02 17:37:26 rock-pi-s iwd[538]:                         0-7
Nov 02 17:37:26 rock-pi-s iwd[538]:                         32
Nov 02 17:37:26 rock-pi-s iwd[538]:         Ciphers: BIP-CMAC-256 BIP-GMAC-256 BIP-GMAC-128 CCMP-256
Nov 02 17:37:26 rock-pi-s iwd[538]:                  GCMP-256 GCMP-128 BIP-CMAC-128 CCMP-128
Nov 02 17:37:26 rock-pi-s iwd[538]:                  TKIP
Nov 02 17:37:26 rock-pi-s iwd[538]:         Supported iftypes: ad-hoc station ap
Nov 02 17:37:26 rock-pi-s iwd[538]: Wiphy phy0 will only use the default interface
Nov 02 17:37:26 rock-pi-s iwd[538]: Error bringing interface 3 up: Cannot assign requested address
</details>

<details>
<summary>dmesg with debug_mask == -1</summary>
[  991.996901] rtw_8723ds mmc2:0001:1: rtw88 SDIO probe: vendor=0x024c device=d723 class=ff
[  991.997188] rtw_8723ds mmc2:0001:1: Firmware version 48.0.0, H2C version 0
[  992.315527] rtw_8723ds mmc2:0001:1: hw cap: hci=0x06, bw=0x03, ptcl=0x02, ant_num=2, nss=1
[  992.315570] rtw_8723ds mmc2:0001:1: use rfe_def[0]
[  992.317161] rtw_8723ds mmc2:0001:1: use rfe_def[0]
[  992.317176] rtw_8723ds mmc2:0001:1: phy cond=0x0304f400
[  992.317231] rtw_8723ds mmc2:0001:1: txpwr regd 3 does not be configured
[  992.317243] rtw_8723ds mmc2:0001:1: cfg txpwr regd 3 by regd 0 as alternative
[  992.317262] rtw_8723ds mmc2:0001:1: txpwr regd 4 does not be configured
[  992.317274] rtw_8723ds mmc2:0001:1: cfg txpwr regd 4 by regd 2 as alternative
[  992.317294] rtw_8723ds mmc2:0001:1: txpwr regd 5 does not be configured
[  992.317305] rtw_8723ds mmc2:0001:1: cfg txpwr regd 5 by regd 2 as alternative
[  992.317325] rtw_8723ds mmc2:0001:1: txpwr regd 6 does not be configured
[  992.317337] rtw_8723ds mmc2:0001:1: cfg txpwr regd 6 by regd 0 as alternative
[  992.317357] rtw_8723ds mmc2:0001:1: txpwr regd 7 does not be configured
[  992.317368] rtw_8723ds mmc2:0001:1: cfg txpwr regd 7 by regd 2 as alternative
[  992.317388] rtw_8723ds mmc2:0001:1: txpwr regd 8 does not be configured
[  992.317400] rtw_8723ds mmc2:0001:1: cfg txpwr regd 8 by regd 0 as alternative
[  992.317420] rtw_8723ds mmc2:0001:1: txpwr regd 9 does not be configured
[  992.317430] rtw_8723ds mmc2:0001:1: cfg txpwr regd 9 by regd 2 as alternative
[  992.317631] rtw_8723ds mmc2:0001:1: regd init state 0: apply alpha2 00, regd {10, 10}, dfs_region 0
[  992.318698] rtw_8723ds mmc2:0001:1: regd state: 0 -> 0
[  992.318741] rtw_8723ds mmc2:0001:1: get alpha2 00 from initiator 0: apply alpha2 00, regd {10, 10}, dfs_region 0
</details>

briansune, 2023-11-02:
@RadxaYuntian 

I am not sure the root cause but what I had discussed with @lwfinger is that the RTW88 have not default firmware loading when IC is not fused with a firmware and this is my case.

The board hardware is fully custom and IC is not from OEM or broker, which causes the IC is not fused with any default firmware.
@lwfinger  I am not sure you had update the driver with a default firmware like rtl8723ds old repository or not?

BR
Brian

briansune, 2023-11-02:
@RadxaYuntian 

Opps typo, it should be IC settings aka EFUSE not firmware.
The EFUSE is controlling the TX LUT power and DCDC, XTAL settings.

You are right eFUSE is what I am discussing.

BR

briansune, 2023-11-02:
@RadxaYuntian 

Why I will discuss @lwfinger with the EFUSE setting is because the XTAL is mostly the case along side with the DCDC.
If XTAL is not aligned with the EFUSE then the WIFI 100% won't work.
If DCDC is not aligned with EFUSE, the voltage you can measure is off the normal range.
Either way will cause the WIFI chip not working, so as long as RTW88 is not update there are not way it could happen.

BR

lwfinger, 2023-11-02:
I looked into providing a default EFUSE when it is not programmed, but I got diverted. If the MAC address is all ones, the EFUSE is not programmed.

briansune, 2023-11-02:
@lwfinger 

Yes, if all ones are shown then EFUSE is not preloaded for the IC. Then the driver need to take place to preload a default one which the old RTL8723DS had provided. But RTW88 does not, which you need to update or extract those from the old repository.

BR

briansune, 2023-11-02:
@lwfinger 

I am afraid I don't have that ability to transfer the old driver to the new old. I am mostly hardware engineer so this is really out of my scope and I would like to.

=S

RadxaYuntian, 2023-11-03:
I see there is `Efuse_Read1ByteFromFakeContent` in rtl8723ds, but I did not find where fakeEfuseContent got initialized. It appears it has a default value of all 0 and then memset to all 0xFF, and neither looks like some real eFuse data.

briansune, 2023-11-03:
@RadxaYuntian 

My method to see the EFUSE is using old RTL8723DS repository and make the debug flag to 4 and during driver load there is a series of message that shown the EFUSE are all ZERO and then it will say loading a default one.

RadxaYuntian, 2023-11-06:
Thanks for the hint. I have located `Hal_InitPGData` for detecting invalid eFuse and then a bunch of parse functions. An example of falling back to driver defaults can be found in `hal_load_pg_txpwr_info` for loading TX LUT. 

I'll see what will happen if I inject them in `rtw88`'s `rtw8723d_read_efuse`.

RadxaYuntian, 2023-11-08:
Spent last 2 days on this hack: https://github.com/radxa-pkg/rtw88/commit/9d2ffb52718d33f6ecba909e5715d28e5197c16d

Current status and insight learnt:
* Must assign a valid MAC address in `efuse->addr`. All zero or all 0xFF address will be accepted by the system as-is, and NetworkManager won't generate random address for those, so the device cannot be brought up. Wasted a lot of time on this.
* Device specific and generic `pg_txpwr_def_info` extracted from `rtl8723ds` both allows the chip to start, even when combined.
* The device currently appears to allow regulatory region changed (via `iw reg set`) and can scan and list Wi-Fi networks (via `nmcli device wifi rescan/list`). The discovered networks are much fewer than what's available, and I haven't been able to get it connected:

```
[  947.839160] wlan0: authenticate with c8:3a:35:9e:43:61
[  947.839258] wlan0: 80 MHz not supported, disabling VHT
[  948.570820] wlan0: send auth to c8:3a:35:9e:43:61 (try 1/3)
[  949.844179] wlan0: send auth to c8:3a:35:9e:43:61 (try 2/3)
[  950.836181] wlan0: send auth to c8:3a:35:9e:43:61 (try 3/3)
[  951.828260] wlan0: authentication with c8:3a:35:9e:43:61 timed out
```

I think to fix it is currently beyond my scope, so we will have to use rtl8723ds in the mean time.

briansune, 2023-11-08:
@RadxaYuntian 

Do a manual MAC assign make the LO lock?
And able to scan WIFI?
If a scan is able to get WIFI device then this is much easy to fix otherwise it is just back to beginning.

briansune, 2023-11-08:
@RadxaYuntian 

So the updated driver can work?
Could you forward and let me give it a try and see the different?
Thank you

RadxaYuntian, 2023-11-08:
In case I wasn't clear the driver **DOES NOT** work for me. I was not able to connect to any network.

If you still want to try it here is a Debian dkms package in zip (since GitHub doesn't allow deb uploading).

[rtw88.zip](https://github.com/lwfinger/rtw88/files/13295171/rtw88.zip)

I don't have the BOM but I think it is a 24M 12pF crystal. If you are curious here is the schematic for our product:

https://dl.radxa.com/rockpis/docs/hw/ROCK-PI-S_v13_sch_200910.pdf

Search 8723 to see the related page.

briansune, 2023-11-08:
@RadxaYuntian 

I need a source to cross compile as my platform is not Debian but ubuntu on different ARM system.

briansune, 2023-11-08:
@RadxaYuntian 

As for schematic short review show you are using DCDC setting. The custom board we had here do tired both LDO and DCDC on old RTL8723DS driver.

RadxaYuntian, 2023-11-08:
If you want source code it is available here: https://github.com/radxa-pkg/rtw88/tree/issue_157

RadxaYuntian, 2023-11-08:
Looking back I forgot to run with `debug_mask` and see what was wrong.

briansune, 2023-11-08:
@RadxaYuntian 

Let's compare and see if the issue is aligned.

briansune, 2023-11-08:
@RadxaYuntian 

I got the current message:

[   27.285840] rtw_core: loading out-of-tree module taints kernel.
[   37.657580] rtw_8723ds mmc1:0001:1: rtw88 SDIO probe: vendor=0x024c device=d723 class=ff
[   37.667085] rtw_8723ds mmc1:0001:1: Firmware version 48.0.0, H2C version 0
[   38.148561] rtw_8723ds mmc1:0001:1: hw cap: hci=0x06, bw=0x03, ptcl=0x02, ant_num=2, nss=1
[   38.148576] rtw_8723ds mmc1:0001:1: use rfe_def[17]
[   38.148586] rtw_8723ds mmc1:0001:1: rfe 17 isn't supported
[   38.154981] rtw_8723ds mmc1:0001:1: failed to setup chip efuse info
[   38.159943] rtw_8723ds mmc1:0001:1: failed to setup chip information

RadxaYuntian, 2023-11-08:
Your eFuse must have been partially populated, as I changed `rfe` to come from eFuse first: https://github.com/radxa-pkg/rtw88/commit/9d2ffb52718d33f6ecba909e5715d28e5197c16d#diff-bda2ebb54a6c298b057cfa4da3534fd505b263989ec51d3663e3f40d9f3a0ebaL233-R261

You can change line 261 back to `efuse->rfe_option = 0;`

briansune, 2023-11-08:
@RadxaYuntian 

Network can scan and WIFI devices are showing up.
Compared to previous this RTW88 driver it cannot list WIFI devices.
Looks like you had made a different.

And WIFI can ping and connect as well but not sure why nmcli works but not nmtui =S
I don't know much kernel stuff.

[image]

> [  651.687345] rtw_8723ds mmc1:0001:1: [IQK] path S0 RXIQK Step1!!
[  651.687417] rtw_8723ds mmc1:0001:1: [IQK] 0x67 @S0 RXIQK1 = 0xac
[  651.690200] rtw_8723ds mmc1:0001:1: [IQK] RF0x1@ path S0 RXIQK1 = 0xe6d
[  651.690735] rtw_8723ds mmc1:0001:1: [IQK] RF0x2@ path S0 RXIQK1 = 0x66d
[  651.692222] rtw_8723ds mmc1:0001:1: [IQK] GNT_BT @S0 RXIQK1 = 0xc
[  651.692301] rtw_8723ds mmc1:0001:1: [IQK] 0x948 @S0 RXIQK1 = 0x99000280
[  651.696716] rtw_8723ds mmc1:0001:1: [IQK] 0xeac = 0x4002000
[  651.696844] rtw_8723ds mmc1:0001:1: [IQK] 0xe94 = 0x1000000, 0xe9c = 0x30000
[  651.696970] rtw_8723ds mmc1:0001:1: [IQK] 0xe90(before IQK)= 0x6d0, 0xe98(afer IQK) = 0x44
[  651.697365] rtw_8723ds mmc1:0001:1: [IQK] 0xe40 = 0x81007c03 u4tmp = 0x81007c03
[  651.697377] rtw_8723ds mmc1:0001:1: [IQK] path S0 RXIQK STEP2!!
[  651.697445] rtw_8723ds mmc1:0001:1: [IQK] 0x67 @S0 RXIQK2 = 0xac
[  651.700377] rtw_8723ds mmc1:0001:1: [IQK] RF0x1 @S0 RXIQK2 = 0xe6d
[  651.700903] rtw_8723ds mmc1:0001:1: [IQK] RF0x2 @S0 RXIQK2 = 0x66d
[  651.702390] rtw_8723ds mmc1:0001:1: [IQK] GNT_BT @S0 RXIQK1 = 0xc
[  651.702471] rtw_8723ds mmc1:0001:1: [IQK] 0x948 @S0 RXIQK1 = 0x99000280
[  651.706357] rtw_8723ds mmc1:0001:1: [IQK] 0xea4 = 0xfb0000, 0xeac = 0x4072000
[  651.706487] rtw_8723ds mmc1:0001:1: [IQK] 0xea0(before IQK)= 0x3e500c, 0xea8(afer IQK) = 0x81a0
[  651.709733] rtw_8723ds mmc1:0001:1: [IQK] path S0 Rx IQK Success!!
[  651.710965] rtw_8723ds mmc1:0001:1: [IQK] back to BB mode, load original value!
[  651.713405] rtw_8723ds mmc1:0001:1: [IQK] restore 0x67 = 0xac
[  651.713420] rtw_8723ds mmc1:0001:1: [IQK] cmp 0:1 final_candidate is 0
[  651.714033] rtw_8723ds mmc1:0001:1: [IQK] X = 0xfd, TX1_A = 0xe1, oldval_1 0xe4
[  651.714046] rtw_8723ds mmc1:0001:1: [IQK] Y = 0x2, TX1_C = 0x1
[  651.715137] rtw_8723ds mmc1:0001:1: [IQK] final_candidate is 0
[  651.715157] rtw_8723ds mmc1:0001:1: [IQK] Result 0: rege94_s1=fd rege9c_s1=2 regea4_s1=100 regeac_s1=7 rege94_s0=fd rege9c_s0=4 regea4_s0=fa regeac_s0=8 (final candidate)
[  651.715174] rtw_8723ds mmc1:0001:1: [IQK] Result 1: rege94_s1=fd rege9c_s1=2 regea4_s1=100 regeac_s1=7 rege94_s0=fd rege9c_s0=4 regea4_s0=fb regeac_s0=7
[  651.715190] rtw_8723ds mmc1:0001:1: [IQK] Result 2: rege94_s1=0 rege9c_s1=0 regea4_s1=0 regeac_s1=0 rege94_s0=0 rege9c_s0=0 regea4_s0=0 regeac_s0=0
[  651.715205] rtw_8723ds mmc1:0001:1: [IQK] Result 3: rege94_s1=0 rege9c_s1=0 regea4_s1=0 regeac_s1=0 rege94_s0=0 rege9c_s0=0 regea4_s0=0 regeac_s0=0
[  651.715450] rtw_8723ds mmc1:0001:1: [IQK]0xc80 = 0x390100e1 0xc94 = 0x0 0xc14 = 0x40001d00 0xca0 = 0x12000
[  651.715629] rtw_8723ds mmc1:0001:1: [IQK]0xcd0 = 0x1c2 0xcd4 = 0x1c8007 0xcd8 = 0x80fa
[  651.715639] rtw_8723ds mmc1:0001:1: [IQK] finished
[  651.715650] wlan0: send auth to 04:d4:c4:45:9c:98 (try 1/3)
[  651.721490] wlan0: authenticated
[  651.761467] wlan0: associate with 04:d4:c4:45:9c:98 (try 1/3)
[  651.761520] rtw_8723ds mmc1:0001:1: [BTCoex], rtw_coex_connect_notify(): 2G start
[  651.761532] rtw_8723ds mmc1:0001:1: [BTCoex], (Before Ant Setup) Delay by IQK
[  651.761682] rtw_8723ds mmc1:0001:1: [BTCoex],  coex_stat->bt_disabled = 0x1
[  651.761695] rtw_8723ds mmc1:0001:1: [BTCoex], rtw_coex_set_ant_path() - PHASE_2G_RUNTIME
[  651.763266] rtw_8723ds mmc1:0001:1: [BTCoex], rtw_coex_run_coex(): reason = 5
[  651.763281] rtw_8723ds mmc1:0001:1: [BTCoex], wl_noisy_level = 0
[  651.763369] rtw_8723ds mmc1:0001:1: [BTCoex], WiFi is single-port 2G!!
[  651.763382] rtw_8723ds mmc1:0001:1: [BTCoex], rtw_coex_action_wl_not_connected()
[  651.763394] rtw_8723ds mmc1:0001:1: [BTCoex], Coex_Table - 1
[  651.763547] rtw_8723ds mmc1:0001:1: [BTCoex], Skip TDMA because no change TDMA(off, 0)
[  651.768130] wlan0: RX AssocResp from 04:d4:c4:45:9c:98 (capab=0x1011 status=0 aid=4)
[  651.768180] rtw_8723ds mmc1:0001:1: send H2C content 40a30040 000ff015
[  651.768391] rtw_8723ds mmc1:0001:1: send H2C content 00000101 00000000
[  651.768575] rtw_8723ds mmc1:0001:1: sta 04:d4:c4:45:9c:98 joined with macid 0
[  651.771026] rtw_8723ds mmc1:0001:1: [BTCoex], rtw_coex_connect_notify(): 2G finish
[  651.771042] rtw_8723ds mmc1:0001:1: [BTCoex], rtw_coex_run_coex(): reason = 7
[  651.771053] rtw_8723ds mmc1:0001:1: [BTCoex], wl_noisy_level = 1
[  651.772056] rtw_8723ds mmc1:0001:1: [BTCoex], WiFi is single-port 2G!!
[  651.772069] rtw_8723ds mmc1:0001:1: [BTCoex], rtw_coex_action_wl_only()
[  651.772083] rtw_8723ds mmc1:0001:1: [BTCoex], Coex_Table - 2
[  651.772351] rtw_8723ds mmc1:0001:1: [BTCoex], rtw_coex_set_table(): 0x6c0 = 66555555, 0x6c4 = 66555555
[  651.772366] rtw_8723ds mmc1:0001:1: [BTCoex], Skip TDMA because no change TDMA(off, 0)
[  651.773736] rtw_8723ds mmc1:0001:1: RSVD_PROBE_RESP loc: 0
[  651.773750] rtw_8723ds mmc1:0001:1: RSVD_PS_POLL loc: 1
[... 55 lines cut ...]
[  660.001704] rtw_8723ds mmc1:0001:1: [BTCoex], Hi-Pri Rx/Tx: 0/0, Lo-Pri Rx/Tx: 0/0
[  660.001727] rtw_8723ds mmc1:0001:1: send H2C content 17000042 00000000
[  660.004111] rtw_8723ds mmc1:0001:1: IGI=0x25, rssi_min=23, cck_fa=357
[  660.004124] rtw_8723ds mmc1:0001:1: cck_fa_avg=230, cck_pd_default=16
[  660.004135] rtw_8723ds mmc1:0001:1: lv: (2) -> (2)
[... cut ...]
[  664.004385] rtw_8723ds mmc1:0001:1: lv: (0) -> (2)
[  664.004439] rtw_8723ds mmc1:0001:1: is_linked=1, lv=2, n_rx=1, cs_ratio=0x14, pd_th=0xd, cck_fa_avg=499
[  664.004666] rtw_8723ds mmc1:0001:1: send H2C content 00000058 00000010
[  664.004827] rtw_8723ds mmc1:0001:1: send H2C content 40830040 000ff010
[  664.173301] rtw_8723ds mmc1:0001:1: recv C2H, id=0x0c, seq=0x06, len=7
[  664.889400] rtw_8723ds mmc1:0001:1: recv C2H, id=0x0c, seq=0x07, len=7
[  665.810090] rtw_8723ds mmc1:0001:1: recv C2H, id=0x0c, seq=0x08, len=7

briansune, 2023-11-08:
@RadxaYuntian 

But I don't understand what is the major reason to use RTW88 rather than RTL8723DS driver?

@lwfinger 
What are the different compare between new RTW88-based RTL8723DS and old RTL8723DS driver?

briansune, 2023-11-08:
@RadxaYuntian 

And there is a catch that looks like it is not too stable compared to old RTL8723DS:

[image]

RadxaYuntian, 2023-11-08:
Yeah I know the very beginning is not 0xFF as I also saw the eFuse dump. But for my chip at least `rfe` is also 0xFF which is why I changed the code to test it, instead of simply passing 0 in the orignial.

briansune, 2023-11-08:
@RadxaYuntian 

> mmc1: new high speed SDIO card at address 0001
mmc0: new high speed SDHC card at address b36b
RTW: sdio_init: sdio clk rate: 25000000
RTW: CHIP TYPE: RTL8723D
RTW: rtw_hal_config_rftype RF_Type is 3 TotalTxPath is 1
RTW: Chip Version Info: CHIP_8723D_Normal_Chip_TSMC_D_CUT_1T1R_RomVer(0)
NET: Registered protocol family 10
RTW: SetHwReg: bMacPwrCtrlOn=1
 mmcblk0: p1 p2
random: fast init done
RTW: sdio_power_on_check: val_mix:0x0000063f, res:0x0000063f
Segment Routing with IPv6
RTW: sdio_power_on_check: 0x100 the result of cmd52 and cmd53 is the same.
sit: IPv6, IPv4 and MPLS over IPv4 tunneling driver
RTW: sdio_power_on_check: 0x1B8 test Pass.
NET: Registered protocol family 17
RTW: ReadAdapterInfo8723DS, 0x4e=0xc2
can: controller area network core (rev 20170425 abi 9)
RTW: EEPROM type is E-FUSE
NET: Registered protocol family 29
RTW: hal_EfuseSwitchToBank: Efuse switch bank to 0
can: raw protocol (rev 20170425)
can: broadcast manager protocol (rev 20170425 t)
can: netlink gateway (rev 20190810) max_hops=1
Bluetooth: RFCOMM TTY layer initialized
Bluetooth: RFCOMM socket layer initialized
Bluetooth: RFCOMM ver 1.11
Bluetooth: BNEP (Ethernet Emulation) ver 1.3
Bluetooth: BNEP filters: protocol multicast
RTW: hal_ReadEFuse_WiFi: data end at address=0xae
Bluetooth: BNEP socket layer initialized
RTW: HW EFUSE
Bluetooth: HIDP (Human Interface Emulation) ver 1.2
Bluetooth: HIDP socket layer initialized
RTW: 0x000:
29
Registering SWP/SWPB emulation handler
81
00
Loading compiled-in X.509 certificates
7C
E1
of-fpga-region fpga-full: FPGA Region probed
88
07
hctosys: unable to open rtc device (rtc0)
00
A0
04
EC
35
12
C0
A2
of_cfs_init
D8
of_cfs_init: OK
RTW: 0x010:
2B
cfg80211: Loading compiled-in X.509 certificates for regulatory database
2A
2A
2B
cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
2C
2C
platform regulatory.0: Direct firmware load for regulatory.db failed with error -2
2F
[... 552 lines cut ...]
RTW: init_mlme_default_rate_set: support CCK
RTW: init_mlme_default_rate_set: support OFDM
RTW: init_channel_set((null)) ChannelPlan ID:0x20, ch num:13
RTW: rtw_alloc_macid((null)) if2, hwaddr:ff:ff:ff:ff:ff:ff macid:1
RTW: rtw_drv_add_vir_if if2 mac_addr : da:2f:e6:09:c4:4b
RTW: rtw_wiphy_alloc(phy0)
RTW: rtw_wdev_alloc(padapter=(ptrval))
RTW: rtw_wiphy_alloc(phy1)
RTW: rtw_wdev_alloc(padapter=(ptrval))
RTW: rtw_wiphy_register(phy0)
RTW: Register RTW cfg80211 vendor cmd(0x67) interface
RTW: _rtw_reg_notifier
RTW: _rtw_reg_notifier_apply: NL80211_REGDOM_SET_BY_CORE to DRV
RTW: rtw_ndev_init(wlan0) if1 mac_addr=d8:2f:e6:09:c4:4b
RTW: rtw_ndev_notifier_call(wlan0) state:17
RTW: cfg80211_rtw_get_txpower
RTW: rtw_ndev_notifier_call(wlan0) state:5
RTW: rtw_wiphy_register(phy1)
RTW: Register RTW cfg80211 vendor cmd(0x67) interface
RTW: _rtw_reg_notifier
RTW: _rtw_reg_notifier_apply: NL80211_REGDOM_SET_BY_CORE to DRV
RTW: rtw_ndev_init(wlan1) if2 mac_addr=da:2f:e6:09:c4:4b
RTW: rtw_ndev_notifier_call(wlan1) state:17
RTW: cfg80211_rtw_get_txpower
RTW: rtw_ndev_notifier_call(wlan1) state:5

lwfinger, 2023-11-08:
@briansune - The main difference between the 2 drivers is the one in rtw88 is maintained. The old one from Realtek is not.

The difference in sensitivity is that you are not setting the RX and TX gains. You need to read the old driver to see what it shows, and do the same in your simulated EFUSE. That is where this information should be.

briansune, 2023-11-08:
@lwfinger 

I didn't mention any sensitivity nor trying to simulated the EFUSE that is what @RadxaYuntian trying to move the old repository RTL8723DS to the new RTW88. Maybe the sensitivity you trying to refer is about the stability of the ping test.

BTW, it is maintained then the EFUSE preloaded feature is also included? As grabbing the old EFUSE is just a temporary fix,

lwfinger, 2023-11-08:
The stability of the ping test, the sensitivity of the scan, and the ease of making a connection will all depend on the TX/RX gains. If the device has the EFUSE programmed, then the gains will come from the EFUSE. If not, it will need to be set the way the MAC address is.

RadxaYuntian, 2023-11-09:
Do you mean `Hal_ReadRFGainOffset`? `PPG_THERMAL_OFFSET_8723D` and `PPG_BB_GAIN_2G_TX_OFFSET_8723D` are defined as 0xFF in Brian's eFuse, so I guess I'm not looking at the right place. The related `power_trim` function also seems to be only available for `rtw8822.c` in `rtw88` driver. It appears to be a lot of missing functionality for rtl8723ds.

briansune, 2023-11-09:
@RadxaYuntian 

I am afraid my expertise's are out of the scope of these coding stuff. I can help on hardware testing and simple cross-compile stuff =S.
As for the EFUSE are my extracted same as yours?

lwfinger, 2023-11-09:
@RadxaYuntian - There could be missing functionality in the 8723ds code, but it works in the PCI and USB drivers, and for the SDIO drivers that have properly coded EFUSE.

In file rtw8723ds.h, struct rtw8723d_efuse shows the layout of the EFUSE that the program expects. In array txpwr_idx_table should be the parameters that the program expects. I have no idea what the vendor driver does.

briansune, 2023-11-09:
@lwfinger 

It is possible to add a printout function or debug to the exist RTW88 and allow @RadxaYuntian to compare with a pre-FUSED IC?

lwfinger, 2023-11-09:
Of course it is possible, but I won't do it. @RadxaYuntian should tell you what variables he wants to see or send you a patch file. The main repository has to be kept clean.

RadxaYuntian, 2023-11-10:
> As for the EFUSE are my extracted same as yours?

<details>
<summary>My eFuse dump</summary>

```
[   12.084581] RTW: HW EFUSE
[   12.084595] RTW: 0x000: 
[   12.084604] 29 
[   12.084611] 81 
[   12.084617] 00 
[   12.084622] 7C 
[   12.084628] E1 
[   12.084633] 88 
[   12.084638] 07 
[   12.084644] 00    
[   12.084650] A0 
[   12.084656] 04 
[   12.084661] EC 
[   12.084667] 35 
[   12.084672] 12 
[   12.084678] C0 
[   12.084683] A2 
[   12.084689] D8
[   12.084696] RTW: 0x010: 
[   12.084701] FF 
[   12.084706] FF 
[   12.084712] FF 
[   12.084718] FF 
[   12.084723] FF 
[   12.084730] FF 
[   12.084736] FF 
[   12.084741] FF    
[   12.084747] FF 
[   12.084753] FF 
[   12.084758] FF 
[   12.084764] FF 
[   12.084769] FF 
[   12.084775] FF 
[   12.084780] FF 
[   12.084786] FF
[   12.084793] RTW: 0x020: 
[   12.084798] FF 
[   12.084804] FF 
[   12.084809] FF 
[   12.084822] FF 
[   12.084846] FF 
[   12.084854] FF 
[   12.084859] FF 
[   12.084865] FF    
[   12.084870] FF 
[   12.084876] FF 
[   12.084881] FF 
[   12.084887] FF 
[   12.084892] FF 
[   12.084898] FF 
[   12.084904] FF 
[   12.084910] FF
[   12.084917] RTW: 0x030: 
[   12.084922] FF 
[   12.084927] FF 
[   12.084933] FF 
[   12.084938] FF 
[   12.084944] FF 
[   12.084949] FF 
[   12.084955] FF 
[   12.084960] FF    
[   12.084966] FF 
[   12.084971] FF 
[   12.084977] FF 
[... 466 lines cut ...]
[   12.095125] FF 
[   12.095132] FF 
[   12.095137] FF 
[   12.095143] FF 
[   12.095148] FF 
[   12.095153] FF 
[   12.095159] FF    
[   12.095165] FF 
[   12.095170] FF 
[   12.095177] FF 
[   12.095182] FF 
[   12.095187] FF 
[   12.095193] FF 
[   12.095199] FF 
[   12.095204] FF
```
</details>

TL;DR: it is really all 0xFF past first 16 bytes.

> In array txpwr_idx_table should be the parameters that the program expects. I have no idea what the vendor driver does.

Vendor driver supplies a device-specific fallback `rtl8723d_pg_txpwr_def_info` and a generic fallback `pg_txpwr_def_info` when the value from eFuse or the more specific fallback failed to meet the `hal_chk_pg_txpwr_info_2g` and `hal_chk_pg_txpwr_info_5g` checks in `hal_load_pg_txpwr_info`. I have tested both data on my device but they did not provide good performance.

Although even if I could make it to work by adjusting my included defaults to match Brian's eFuse dump, we still need a way to notify the network stack that the Phy does not have a MAC address, so user space configuration daemon could handle this task. Hardcoding a MAC in the driver is not acceptable for production use.

briansune, 2023-11-10:
@RadxaYuntian 

Did you do a read back sanity check after loading those value or the dump is after the write?
If you can guide me where should I add the code and do a 2nd read back after write on the old repository driver.

RadxaYuntian, 2023-11-10:
That is just what the vendor `rtl8723ds` prints when debug is enabled and log level is set to 4. I believe this is the raw data on the eFuse.

If you want to override the entire eFuse, you need to first update [`struct rtw8723ds_efuse`](https://github.com/lwfinger/rtw88/blob/master/rtw8723d.h#L52-L55) so the final `struct rtw8723d_efuse` has the same size of the physical eFuse (0x200):

```suggestion
struct rtw8723ds_efuse {
	u8 res4[0x1f];				/* 0xd0 */
	u8 ppg_thermal;				/* 0xef */
	u8 res5[0x2a];				/* 0xf0 */
	u8 mac_addr[ETH_ALEN];		/* 0x11a */
	u8 res6[0xce];				/* 0x120 */
	u8 ppg_bb_gain;				/* 0x1ee */
	u8 res7[0x10];				/* 0x1ef */
};
```

Then in the `rtw8723d_read_efuse` function you can overwrite the content of `log_map` to be that of your emulated eFuse.

briansune, 2023-11-10:
@RadxaYuntian 

What I am trying to say is that:
At the old RTL8723DS do you also print the eFUSE content after the driver have loaded a default eFUSE setting after checking it is incorrect? If so I can also cross-check and provide my dump to compare.

RadxaYuntian, 2023-11-10:
The eFuse dumped by vendor driver is the raw eFuse data: https://github.com/lwfinger/rtl8723ds/blob/master/core/efuse/rtw_efuse.c#L2312

You can see the driver check the validity of the eFuse AFTER it was dumped: https://github.com/lwfinger/rtl8723ds/blob/master/hal/rtl8723d/rtl8723d_hal_init.c#L3634-L3649

The vendor driver also did not update the in-memory eFuse dump. The default value is plugged in at run-time, after it checks the eFuse provided value. This is also how I implemented the default value in rtw88 HACK: https://github.com/lwfinger/rtl8723ds/blob/master/hal/rtl8723d/rtl8723d_hal_init.c#L3973-L3975

briansune, 2023-11-10:
@RadxaYuntian 

First two explains are clear.
So the dump is before default settings are loaded in. And printout is also before load.

As for the last then I would suggest why not simply reuse the before loaded print and print out the after load settings and let us do a cross-check if it is the same? (use result to debug should be more easy I guess)

Then after we had confirm both eFUSE after default settings loading the hack can be more sure to start with (I hope)?

lwfinger, 2023-11-10:
@RadxaYuntian - Yes, there are features missing for rtw8723ds, but the underllying code for rtw8723d is maintained. If you/we come up with code changes, they can be introduced into wireless-next, and they will ultimately be applied to the kernel. That is what "maintained" means. It does not mean that it is complete at the moment, but the mechanism exists to make it better. BTW, the guy that wrote the kernel's rtw8723ds could not envision that there would be models in the wild with empty EFUSEs. I still cannot believe that vendors would do this!!

As to assigning a MAC address. The first 3 digits should be the Realtek manufacturer values, and the last 3 assigned from a random number. To give reproducibility, we could also set up a module parameter so that the MAC address is read at boot time. Module parameters are frowned upon, but the case could be made for a new one here.

briansune, 2023-11-10:
@lwfinger 

According to my previous RF company experience the EFUSE is what defines the conformance test and standard.
If CE standard then the max power of the TX has to be fixed to a level as well as EMC to fit the location of sell. This do make sense as 3rd party distributor can only got blank IC.

The EFUSE is preloaded between supplier broker and the company agreement. While user can adjust the TX LUT and RX filter but from first place it much be preprogrammed to fit the standard that country going to distribute or sell. Realtek won't even response to small company and this is  why WIFI module exist (USB SDIO or PCIE) as they provide 100% verified by the local conformance test and standard.

briansune, 2023-11-10:
@lwfinger 

> BTW, the guy that wrote the kernel's rtw8723ds could not envision that there would be models in the wild with empty EFUSEs. I still cannot believe that vendors would do this!

I want to correct such idea. This is incorrect, even the EFUSE is preprogrammed from first place EFUSE could be corrupted and once this corruption exist the EFUSE must be self correct or reloaded with a basic fit on most country or TX RX requirements.
This is why the original RTL8723DS have such feature. EFUSE correction and check both need to be exist otherwise the driver itself is considered as incomplete from first place.

Correct me if it is not the case.

briansune, 2023-11-10:
@RadxaYuntian 

I had an info got to update is that the board I dump my EFUSE is an new one and different that this post original open.
I do tried to cross-compile the RTW88 original driver. Without your modification this board and load the driver but the TX RX is so unstable and very easy to lost connection.

RTW 88 
[image]
vs OLD RTL8723DS
[image]

@lwfinger 
Do you have encountered such situation? This chip do have preloaded EFUSE.

lwfinger, 2023-11-10:
I have no personal exterience with any SDIO chips, but the autfor of rtw8723ds.c in the rtw88 repo must have. I will contact him to see if I can get his dump.

I forgot to mention this before, but one of the things that gets added to the EFUSE by the factory are the optimum power settings for THAT chip. Without that information, it will be difficult to get proper performance.

briansune, 2023-11-11:
@lwfinger 

Optimum TX LUT is what you are referring to. RX have no power LUT but only filter table or IF settings. And the EFUSE situation still unable to explain why a fused chip work poorly on RTW88. If this is blank IC original RTW88 should result same as the very beginning of the post but one of the board chip is not (could be reused or new old stock). There are no point to fuse a chip but using wrong settings when the chip is new.

RadxaYuntian, 2023-11-13:
> BTW, the guy that wrote the kernel's rtw8723ds could not envision that there would be models in the wild with empty EFUSEs. I still cannot believe that vendors would do this!!

For us it is a little bit different. We are only a single board manufacturer, and there are customers who uses our products to integrate into their own solution. So we generally don't touch eFuse since the customer might want to make adjustment to it.

The vendor rtl8723ds driver could generate random MAC when eFuse one is invalid, and that is preferred compared to a manual module parameter.

> Without your modification this board can load the driver but the TX RX is so unstable and very easy to lost connection.

This could well due to some eFuse values are not defined. In the vendor driver as I mentioned above, it will perform runtime check for most eFuse config value, and dynamically adjust them if some is detected to be invalid. Such functionality is missing in RTW88, which requires all parameters stored in the eFuse must be valid.

You could dump your new eFuse content, and compared with the working one you posted above.

briansune, 2023-11-13:
@RadxaYuntian 

I am very confused:
Q) What you trying to do here? Update the RTW88 driver missing feature?
Q) So do the only board that your side manufacture have issue or not on RTW88 driver instead of the MAC? 
E) As I had already mention custom board design without WIFI module would highly requires a EFUSE preload feature. This had explained above clearly. And RTW88 driver runs with a preload EFUSE IC (not your modified RTW88) and the EFUSE is not programmed by our side only preform badly on the same DUT we trying to compare with.

Q) So do RTW88 involve runtime check?
Q) Even it is undergo a runtime check, do RTW88 involve a read and write back runtime process? (this also had mentioned as ask many time on above) 
E) The same board as RTL8723DS IC and with preloaded EFUSE that are not EFUSE programmed by our side on two driver shows a different behavior. If vendor driver adjust it at runtime then no matter the default EFUSE is RTL8732DS old driver will recover a better EFUSE for operation.
Q) So original vendor driver already have enough info to enables the fix of RTW88.
Q) How can I dump the EFUSE content when the IC itself is already contains non-empty EFUSE and runtime fixed by RTL8732DS vendor driver? (No guidelines are shown or given)

E) the above dump is undergo on the latest board and only the same board. The only info is unaligned is the chip EFUSE is not purely blank.

RadxaYuntian, 2023-11-13:
> Q) What you trying to do here? Update the RTW88 driver missing feature?

Make RTW88 working on our 8723DS based product, that doesn't have eFuse programmed, and was working on RTL8723DS (a.k.a vendor driver).

> Q) So do the only board that your side manufacture have issue or not on RTW88 driver instead of the MAC?

MAC address missing is due to un-programmed eFuse, which RTW88 currently cannot handle correctly.

> Q) So do RTW88 involve runtime check?

It currently does not, which is why I keep comparing back to vendor driver which does have runtime check of eFuse contents.

> Q) Even it is undergo a runtime check, do RTW88 involve a read and write back runtime process? (this also had mentioned as ask many time on above)

There is no write back on RTW88 or vendor driver, since eFuse is program once only. You do not want drive to write it without your permission.

> E) The same board as RTL8723DS IC and with preloaded EFUSE that are not EFUSE programmed by our side on two driver shows a different behavior. If vendor driver adjust it at runtime then no matter the default EFUSE is RTL8732DS old driver will recover a better EFUSE for operation.

This is basically correct.

> Q) So original vendor driver already have enough info to enables the fix of RTW88.

In theory yes, as shown by device working with vendor driver under the same condition. However, the current attempt to add eFuse validation and fall back failed to achieve the same result on my board. This means either my code is wrong, or there are some other parts where the driver handles the hardware differently. Prefilling eFuse value is a low hanging fruit that I can handle, comparing the behavior is not.

> Q) How can I dump the EFUSE content when the IC itself is already contains non-empty EFUSE and runtime fixed by RTL8732DS vendor driver? (No guidelines are shown or given)

You dumped it [here](https://github.com/lwfinger/rtw88/issues/157#issuecomment-1801866371). As explained [here](https://github.com/lwfinger/rtw88/issues/157#issuecomment-1805212363) this is the raw eFuse.

briansune, 2023-11-13:
@RadxaYuntian 

Q) So what do you need?
E=Explain) In this post very beginning, the board is equipped with a non-fused EFUSE IC, which should result in the same situation as yours.
Q) You had pointed out the function but could you elaborate a bit more how to add it or use it to make the compile prints out the dump? (I have limited software background on driver and kernel etc)

RadxaYuntian, 2023-11-13:
> Q) So what do you need?
> E=Explain) In this post very beginning, the board is equipped with a non-fused EFUSE IC, which should result in the same situation as yours.

I'd like to see the dump of [this device](https://github.com/lwfinger/rtw88/issues/157#issuecomment-1805947004). It should match [my eFuse](https://github.com/lwfinger/rtw88/issues/157#issuecomment-1805026898), as my hypothesis back then was:

> So I think the issue is in the other fields where Brian's eFuse has a different value than the defaults I included from 

So if we both have basically the same eFuse, then I think it will be worth to pick some defaults from your first eFuse dump that was pre-programmed with some working value.

> Q) You had pointed out the function but could you elaborate a bit more how to add it or use it to make the compile prints out the dump? (I have limited software background on driver and kernel etc)

It's the same method you described [here](https://github.com/lwfinger/rtw88/issues/157#issuecomment-1792341090).

briansune, 2023-11-13:
@RadxaYuntian 

>I'd like to see the dump of https://github.com/lwfinger/rtw88/issues/157#issuecomment-1805947004. It should match https://github.com/lwfinger/rtw88/issues/157#issuecomment-1805026898, as my hypothesis back then was:

This device is the new board I am mentioning, which works without your modification of RTW88.
The dump is https://github.com/lwfinger/rtw88/issues/157#issuecomment-1801866371 via the RTL8723DS vendor driver.

Q) Or you would like the old board dump that is not working with the default RTW88 driver?

dubhater, 2023-11-13:
For the ping problem, did y'all try to disable the power saving? ``iw wlan0 set power_save off``

RadxaYuntian, 2023-11-13:
> Q) Or you would like the old board dump that is not working with the default RTW88 driver?

Yes I want the dump for this one. Thanks.

briansune, 2023-11-13:
@dubhater 

> iw wlan0 set power_save off

[image]

lwfinger, 2023-11-13:
You need the sudo version.

briansune, 2023-11-13:
> You need the sudo version.

I do used sudo see the image carefully. the fail is not intended but just pressing up to command line ping.

[image]

briansune, 2023-11-13:
@RadxaYuntian 

I had find the old board and do some test with the default RTW88 driver.
This is the board that used to test and issue of this post is based on this board at very beginning.
Results are as follow:

<details>
  <summary>Erros</summary>

  ```
briansune@zynq_brian:~/drv$ ip link
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: sit0@NONE: <NOARP> mtu 1480 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/sit 0.0.0.0 brd 0.0.0.0
3: wlan0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/ether ff:ff:ff:ff:ff:ff brd ff:ff:ff:ff:ff:ff
briansune@zynq_brian:~/drv$ sudo ip link set dev wlan0 address 12:e1:7d:f8:a3:e5
briansune@zynq_brian:~/drv$ sudo iw dev wlan0 scan | grep SSID
briansune@zynq_brian:~/drv$ rtw_8723ds mmc1:0001:1: failed to poll offset=0x5f8 mask=0xff value=0x0

briansune@zynq_brian:~/drv$ sudo iw dev wlan0 scan | grep SSID
rtw_8723ds mmc1:0001:1: failed to poll offset=0x5f8 mask=0xff value=0x0
rtw_8723ds mmc1:0001:1: mac power on failed
rtw_8723ds mmc1:0001:1: failed to power on mac
rtw_8723ds mmc1:0001:1: leave idle state failed
rtw_8723ds mmc1:0001:1: failed to leave ips state
rtw_8723ds mmc1:0001:1: failed to leave idle state
briansune@zynq_brian:~/drv$ sudo iw dev wlan0 scan | grep SSID
rtw_8723ds mmc1:0001:1: failed to poll offset=0x5f8 mask=0xff value=0x0
rtw_8723ds mmc1:0001:1: mac power on failed
rtw_8723ds mmc1:0001:1: failed to power on mac
rtw_8723ds mmc1:0001:1: leave idle state failed
rtw_8723ds mmc1:0001:1: failed to leave ips state
rtw_8723ds mmc1:0001:1: failed to leave idle state
briansune@zynq_brian:~/drv$

  ```

</details>

<details>
  <summary>Old Board Dump Log</summary>

  ```
RTW: HW EFUSE
Bluetooth: RFCOMM ver 1.11
RTW: 0x000:
29
81
00
7C
E1
88
07
00
A0
04
EC
35
12
C0
A2
D8
RTW: 0x010:
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
RTW: 0x020:
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
RTW: 0x030:
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
RTW: 0x040:
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
RTW: 0x050:
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
RTW: 0x060:
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
RTW: 0x070:
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
RTW: 0x080:
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
RTW: 0x090:
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
RTW: 0x0a0:
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
RTW: 0x0b0:
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
RTW: 0x0c0:
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
RTW: 0x0d0:
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
RTW: 0x0e0:
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
RTW: 0x0f0:
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
RTW: 0x100:
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
RTW: 0x110:
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
RTW: 0x120:
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
RTW: 0x130:
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
RTW: 0x140:
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
RTW: 0x150:
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
RTW: 0x160:
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
RTW: 0x170:
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
RTW: 0x180:
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
RTW: 0x190:
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
RTW: 0x1a0:
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
RTW: 0x1b0:
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
RTW: 0x1c0:
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
RTW: 0x1d0:
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
RTW: 0x1e0:
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
RTW: 0x1f0:
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF
FF

  ```

</details>

[40 more replies]

closed 2023-11-21 by lwfinger