lwfinger/rtw88 #158: Bluetooth questions

briansune, 2023-08-02:
@lwfinger 

Apart from the SDIO issue.
I had tried the bluetooth and not sure this driver is involved bluetooth or not.

sudo hciconfig -a
hci0:   Type: Primary  Bus: USB
        BD Address: 48:8F:4C:39:49:40  ACL MTU: 1021:8  SCO MTU: 255:12
        UP RUNNING
        RX bytes:5701 acl:0 sco:0 events:201 errors:0
        TX bytes:34247 acl:0 sco:0 commands:178 errors:0
        Features: 0xff 0xff 0xff 0xfe 0xdb 0xfd 0x7b 0x87
        Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3
        Link policy: RSWITCH HOLD SNIFF PARK
        Link mode: SLAVE ACCEPT
        Name: 'RTK_BT_4.1'
        Class: 0x000000
        Service Classes: Unspecified
        Device Class: Miscellaneous,
        HCI Version: 4.2 (0x8)  Revision: 0x828a
        LMP Version: 4.2 (0x8)  Subversion: 0x96f1
        Manufacturer: Realtek Semiconductor Corporation (93)

hci0:   Type: Primary  Bus: USB
        BD Address: 48:8F:4C:39:49:40  ACL MTU: 1021:8  SCO MTU: 255:12
        UP RUNNING
        RX bytes:5978 acl:0 sco:0 events:204 errors:0
        TX bytes:34256 acl:0 sco:0 commands:181 errors:0

bluetoothctl
Agent registered
[bluetooth]# show
No default controller available
[bluetooth]# scan on
No default controller available
[bluetooth]#

So I am not sure why it can scan both BLE and General BT.
Once it uses bluetoothctl it cannot run any bluetooth parring or scan.

Thank you

briansune, 2023-08-02:
@lwfinger 

Okay problem solved my BAD:
usb-autosuspend problem HAHA 😛

briansune, 2023-08-02:
@lwfinger 

But I observe a very puzzling issue.
If I do a cold power on that insert the driver the:

bluetoothctl
Agent registered
[bluetooth]# show
No default controller available
[bluetooth]# scan on
No default controller available
[bluetooth]#

Will happen.
However, what if I unplug the usb and replug the bluetoothctl no longer happen.
Do there any driver startup reset bug or unstable?

Thank you

lwfinger, 2023-08-02:
I do not know. Note that the Bluetooth driver is separate from rtw88. You should be asking the BT experts about this problem.

lwfinger, 2023-08-02:
If your BT device is not in the tables, then it is possible that the driver will be loaded, but it will not get the right firmware. What does 'lsusb' say about your device. Are you using the driver that is built into your kernel, or a newer driver such as my GitHub repo rtw89-BT?

briansune, 2023-08-02:
@lwfinger 

I am using built in kernel driver.
It is repeatable on every cold boot and replug usb will resolve the issue and reboot after the replug will also keep working properly.

briansune, 2023-08-02:
@lwfinger 

lsusb reply:
lsusb
Bus 001 Device 002: ID 0bda:d723 Realtek Semiconductor Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

lwfinger, 2023-08-03:
Your problem with cold boot sounds as if the USB chain is being scanned before the power is ready.

A 5.4.0 kernel is going to be using the "fallback" driver recognition. It will not load the firmware correctly the way a newer driver would do.

briansune, 2023-08-03:
@lwfinger 

I am not really understand all the knowledge behind sorry for that.
But I guess it will always need a software remount and mount of the RTL8723DU for cold boot?
Any suggestion could resolve this issue to make more stable?

Current status is that Bluetooth instead of that issue is working very well and stable.
Connect rfcomm and WIFI coexist is working very well as well.

Thank you

lwfinger, 2023-08-03:
From a cold boot, how long does it take before the +5 line to stabilize at the USB ports with the rtl8723du plugged in?

If you plug a POWERED USB hub into the main board, and plug the rtl8723du into that, does it make a difference?

briansune, 2023-08-03:
@lwfinger 

My board / hardware is setup as follow.

[CPU + USB ULPI + USB HUB + RTL8723DU]
The above hardware are all fixed on board so once it is powered on it will power up all together.
Meanwhile USB ULPI & HUB will under go hardware reset via CPU.

Such issue could resolve when power down and power up cycle is short.
So I highly suspect it is a reset issue.

Any software reset could be added?

briansune, 2023-08-07:
@lwfinger 

Any update on this issue

Thank you

closed 2023-11-22 by briansune