lwfinger/rtw88 #159: Make sign-install - Debian 12 - linux-headers-6.1.0-11-amd64

RENANZG, 2023-08-11:
**Noob here. Why can't I make the module work with secure boot enabled?**
```
user@debian:~$ sudo modprobe rtw_8723du
[sudo] senha para user: 
modprobe: ERROR: could not insert 'rtw_8723du': Key was rejected by service
```
```
user@debian:~$ sudo modinfo rtw_8723du 
filename:       /lib/modules/6.1.0-11-amd64/kernel/drivers/net/wireless/realtek/rtw88/rtw_8723du.ko
license:        Dual BSD/GPL
description:    Realtek 802.11n wireless 8723du driver
author:         Hans Ulli Kroll <linux@ulli-kroll.de>
alias:          usb:v7392pD611d*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v0BDApD723d*dc*dsc*dp*icFFiscFFipFFin*
depends:        rtw_usb,usbcore,rtw_8723d
retpoline:      Y
name:           rtw_8723du
vermagic:       6.1.0-11-amd64 SMP preempt mod_unload modversions 
```

**I thought it was spoiling, it looks like other drivers loaded.**
```
user@debian:~ $ sudo modprobe rtw_8723de
```

**This is the entire installation procedure that has been adopted:**
```
user@debian:~ $ sudo apt-get install make gcc linux-headers-$(uname -r) build-essential git
Lendo listas de pacotes... Pronto
Construindo árvore de dependências... Pronto
Lendo informação de estado... Pronto        
make is already the newest version (4.3-4.1).
gcc is already the newest version (4:12.2.0-3).
linux-headers-6.1.0-11-amd64 is already the newest version (6.1.38-4).
build-essential is already the newest version (12.9).
git is already the newest version (1:2.39.2-1.1).
0 pacotes atualizados, 0 pacotes novos instalados, 0 a serem removidos e 0 não atualizados.
user@debian:~ $ git clone https://github.com/lwfinger/rtw88.git
Cloning into 'rtw88'...
remote: Enumerating objects: 2132, done.
remote: Counting objects: 100% (243/243), done.
remote: Compressing objects: 100% (94/94), done.
remote: Total 2132 (delta 156), reused 201 (delta 149), pack-reused 1889
Receiving objects: 100% (2132/2132), 5.98 MiB | 3.54 MiB/s, done.
Resolving deltas: 100% (1251/1251), done.
user@debian:~ $ ^[[200~cd rtw88~^C
user@debian:~ $ cd rtw88
user@debian:~ /rtw88$ make
make -C /lib/modules/6.1.0-11-amd64/build M=/home/user/rtw88 modules
make[1]: a entrar na pasta "/usr/src/linux-headers-6.1.0-11-amd64"
  CC [M]  /home/user/rtw88/main.o
  CC [M]  /home/user/rtw88/mac80211.o
  CC [M]  /home/user/rtw88/util.o
  CC [M]  /home/user/rtw88/debug.o
  CC [M]  /home/user/rtw88/tx.o
  CC [M]  /home/user/rtw88/rx.o
  CC [M]  /home/user/rtw88/mac.o
  CC [M]  /home/user/rtw88/phy.o
  CC [M]  /home/user/rtw88/coex.o
  CC [M]  /home/user/rtw88/efuse.o
  CC [M]  /home/user/rtw88/fw.o
  CC [M]  /home/user/rtw88/ps.o
  CC [M]  /home/user/rtw88/sec.o
  CC [M]  /home/user/rtw88/wow.o
  CC [M]  /home/user/rtw88/bf.o
  CC [M]  /home/user/rtw88/regd.o
  CC [M]  /home/user/rtw88/sar.o
  LD [M]  /home/user/rtw88/rtw_core.o
  CC [M]  /home/user/rtw88/rtw8822b.o
  CC [M]  /home/user/rtw88/rtw8822b_table.o
[... 110 lines cut ...]
  LD [M]  /home/user/rtw88/rtw_pci.ko
  BTF [M] /home/user/rtw88/rtw_pci.ko
Skipping BTF generation for /home/user/rtw88/rtw_pci.ko due to unavailability of vmlinux
  CC [M]  /home/user/rtw88/rtw_sdio.mod.o
  LD [M]  /home/user/rtw88/rtw_sdio.ko
  BTF [M] /home/user/rtw88/rtw_sdio.ko
Skipping BTF generation for /home/user/rtw88/rtw_sdio.ko due to unavailability of vmlinux
  CC [M]  /home/user/rtw88/rtw_usb.mod.o
  LD [M]  /home/user/rtw88/rtw_usb.ko
  BTF [M] /home/user/rtw88/rtw_usb.ko
Skipping BTF generation for /home/user/rtw88/rtw_usb.ko due to unavailability of vmlinux
make[1]: a sair da pasta "/usr/src/linux-headers-6.1.0-11-amd64"
user@debian:~/rtw88$ sudo make sign-install
make -C /lib/modules/6.1.0-11-amd64/build M=/home/user/rtw88 modules
make[1]: a entrar na pasta "/usr/src/linux-headers-6.1.0-11-amd64"
make[1]: a sair da pasta "/usr/src/linux-headers-6.1.0-11-amd64"
...+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++*..+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++*....+..+...............+.......+......+...............+..+...+...............+....+......+..+.+............+........+.+.........+..+...+....+...+........+.+.....+....+.....+.......+..+.+.........+..................+.....+...+.+........+............+.........+.+...+..+......+.......+...+...+...........+.+.....+....+........+...............+......+.+...+...........+......+.........+......+.........+..........+...........+.+..+.+...............+.....+....+.....+.+............+..+...............+...+.......+...+.....+.........+.+........+....+...+...+.........+.....+..........+......+......+.....+....+......+.....+...+...+....+......+..+............+.....................+....+...+..+...+......+.+......+..+......+.+......+..................+..+...+.+.................................+........+....+...........+.......+...........+.+.........+.....+.......+..+..........+..+...+.............+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
.....+..+...+......+...+.......+......+..+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++*..........+.....+...+.+...+......+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++*..+....+.....+...+.+............+..+...+...+....+.....+.+.....+.+...........+...................+...+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
-----
input password: 
input password again: 
Install rtw88 SUCCESS
user@debian:~ /rtw88$ 
```
**Thanks**

closed 2023-08-11 by lwfinger

RENANZG, 2023-08-11:
Thanks. I think I'll uninstall everything and reset the BIOS
```
cd ~/rtw88
sudo make uninstall
sudo depmod 
sudo update-initramfs -u -k all
```

lwfinger, 2023-08-11:
With a 55.15 kernel, you do not need to blacklist anything.

I am not sure resetting the BIOS will help. One of mine has the option to uninstall all the keys, and then reinstall them. That one does work.

RENANZG, 2023-08-14:
Confusion seems to be here:

Debian Bug report logs - #989463 please align shim-signed dkms behaviour with Ubuntu
[https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=989463](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=989463)

lwfinger, 2023-08-14:
BTF generation has nothing to do with signing - it has to do with debug information. The messages you are seeing are harmless.