lwfinger/rtw88 #160: Network card - Advice

RENANZG, 2023-08-19:
Hello. Do you think it wouldn't be more practical for me to change the network card?

The notebook I'm using has the rtw_8323du card. It's not being able to enable secure boot. (I seem to be having I/O problems with the usb ports on my external hdd case.)

`[67929.839681] Buffer I/O error on dev sdc1, logical block 30434158, async page read
[67929.840298] Buffer I/O error on dev sdc1, logical block 30434174, async page read
[67929.840358] Buffer I/O error on dev sdc1, logical block 30434174, async page read
[67929.840691] Buffer I/O error on dev sdc1, logical block 30434178, async page read
[67932.916177] rtw_8723du 1-4:1.2: failed to get tx report from firmware
[67936.916204] rtw_8723du 1-4:1.2: failed to get tx report from firmware
`

Could I buy this module to replace it?

Thank you very much

lwfinger, 2023-08-19:
Secure boot requires that you sign the modules with 'sudo make sign-install', AND that you enroll the MOK key on the next reboot. If you are signing them, then you failed to enroll the key when you saw the MOK blue screen. Recovering from that is difficult.

I am not sure that the external hard drive problems are USB related. Have you done a forced fsck on that driver?

Is the RTW8723DU internal? I have never seen that format - only an external USB format.

If you have a mini-PCIe slot available and the internal wifi antenna leads, then that Intel card would work. I think that they solved their recent round of problems with the AX200.

RENANZG, 2023-08-19:
1. I've done several things on this external hard drive usb 3.0, in addition to the "checkdisk /F" I added the parameter "usbcore.autosuspend=-1".

2.  RTW8723DU is internal 😲 

3. I can't open the notebook to check without losing the warranty. (I couldn't get information by other means.)
```

[ 4702.275138] rtw_8723du 1-4:1.2: failed to get tx report from firmware
[ 4710.138950] rtw_8723du 1-4:1.2: timed out to flush queue 1
[ 4712.414876] rtw_8723du 1-4:1.2: failed to get tx report from firmware
[ 4717.928761] [UFW BLOCK] IN=nordlynx OUT= MAC= SRC=180.248.97.45 DST=10.5.0.2 LEN=171 TOS=0x04 PREC=0xC0 TTL=54 ID=8589 PROTO=ICMP TYPE=3 CODE=3 [SRC=10.5.7.161 DST=180.248.97.45 LEN=143 TOS=0x04 PREC=0x00 TTL=54 ID=61572 DF PROTO=UDP SPT=59023 DPT=50159 LEN=123 ] 
[ 4718.962736] rtw_8723du 1-4:1.2: timed out to flush queue 1
[ 4719.426966] rtw_8723du 1-4:1.2: timed out to flush queue 1
[ 4722.750613] rtw_8723du 1-4:1.2: failed to get tx report from firmware
[ 4722.921862] [UFW BLOCK] IN=nordlynx OUT= MAC= SRC=10.5.0.2 DST=239.192.152.143 LEN=165 TOS=0x00 PREC=0x00 TTL=32 ID=13886 DF PROTO=UDP SPT=6771 DPT=6771 LEN=145 
[ 4729.250531] usb 2-1: USB disconnect, device number 2
[ 4729.250708] xhci_hcd 0000:00:15.0: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.
[ 4729.258457] device offline error, dev sdc, sector 87469744 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 2                       
[ 4729.258517] device offline error, dev sdc, sector 6664 op 0x1:(WRITE) flags 0x0 phys_seg 1 prio class 2                          
[ 4729.258542] Buffer I/O error on dev sdc1, logical block 577, lost async page write                                               
[ 4729.258562] device offline error, dev sdc, sector 10472 op 0x1:(WRITE) flags 0x0 phys_seg 1 prio class 2                         
[ 4729.258566] Buffer I/O error on dev sdc1, logical block 1053, lost async page write                                              
[ 4729.258580] device offline error, dev sdc, sector 10568 op 0x1:(WRITE) flags 0x0 phys_seg 1 prio class 2                         
[ 4729.258584] Buffer I/O error on dev sdc1, logical block 1065, lost async page write                                              
[ 4729.258603] device offline error, dev sdc, sector 10584 op 0x1:(WRITE) flags 0x0 phys_seg 2 prio class 2                         
[ 4729.258607] Buffer I/O error on dev sdc1, logical block 1067, lost async page write                                              
[ 4729.258611] Buffer I/O error on dev sdc1, logical block 1068, lost async page write                                              
[ 4729.258623] device offline error, dev sdc, sector 10848 op 0x1:(WRITE) flags 0x0 phys_seg 1 prio class 2                         
[ 4729.258627] Buffer I/O error on dev sdc1, logical block 1100, lost async page write                                              
[ 4729.258643] device offline error, dev sdc, sector 10920 op 0x1:(WRITE) flags 0x0 phys_seg 1 prio class 2                         
[ 4729.258647] Buffer I/O error on dev sdc1, logical block 1109, lost async page write                                              
[ 4729.258661] device offline error, dev sdc, sector 10936 op 0x1:(WRITE) flags 0x0 phys_seg 1 prio class 2                         
[ 4729.258665] Buffer I/O error on dev sdc1, logical block 1111, lost async page write                                              
[ 4729.258678] device offline error, dev sdc, sector 11136 op 0x1:(WRITE) flags 0x0 phys_seg 1 prio class 2                         
[ 4729.258682] Buffer I/O error on dev sdc1, logical block 1136, lost async page write                                              
[ 4729.258695] device offline error, dev sdc, sector 11400 op 0x1:(WRITE) flags 0x0 phys_seg 1 prio class 2                         
[ 4729.258699] Buffer I/O error on dev sdc1, logical block 1169, lost async page write                                              
[ 4730.858006] [UFW BLOCK] IN=nordlynx OUT= MAC= SRC=100.127.248.46 DST=10.5.0.2 LEN=56 TOS=0x00 PREC=0x00 TTL=250 ID=57233 DF PROTO=ICMP TYPE=3 CODE=13 [SRC=10.5.7.161 DST=187.103.62.88 LEN=60 TOS=0x00 PREC=0x00 TTL=59 ID=50749 DF PROTO=TCP INCOMPLETE [8 bytes] ] 
[ 4730.937606] [UFW BLOCK] IN=nordlynx OUT= MAC= SRC=100.127.248.46 DST=10.5.0.2 LEN=56 TOS=0x00 PREC=0x00 TTL=250 ID=276 DF PROTO=ICMP TYPE=3 CODE=13 [SRC=10.5.7.161 DST=187.103.62.88 LEN=132 TOS=0x04 PREC=0x00 TTL=59 ID=44477 DF PROTO=UDP SPT=59023 DPT=42930 LEN=112 ] 
[ 4732.190616] rtw_8723du 1-4:1.2: failed to get tx report from firmware
[ 4734.262923] buffer_io_error: 1846 callbacks suppressed
[ 4734.262938] Buffer I/O error on dev sdc1, logical block 30349920, async page read                                                
[ 4734.265317] Buffer I/O error on dev sdc1, logical block 30349920, async page read                                                
[ 4734.266457] Buffer I/O error on dev sdc1, logical block 30349920, async page read                                                
[ 4734.266592] Buffer I/O error on dev sdc1, logical block 3034992

```

lwfinger, 2023-08-20:
It is unlikely that your motherboard has a PCIe slot, thus that Intel card will do you no good.

I am worried about those disk errors. What file system do you have on /dev/sdc1?

RENANZG, 2023-08-20:
NTFS 😬

lwfinger, 2023-08-20:
You will need to fix those errors while running Windows. NTFS is only partially supported in Linux.

RENANZG, 2023-08-22:
Yes. I did it. I used WinPE ISO, then at the command line I typed chdsk /F. I think the problem is due to the fact that my HD does not support the UASP protocol. Thank you very much for your attention lwfinger.

RENANZG, 2023-09-09:
Iwfinger, I don't know if you could take that doubt. Do you think this setting below can cause any damage to the connection? My system is transplanted from another notebook I think it was the name of the old network card. It has snuggled random disconnection from the internet, sometimes from the external hds (I saw that this is not uncommon, even in windowns).

```
user@debian:~$ ip link
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: wlx4607b84d436b: <NO-CARRIER,BROADCAST,MULTICAST,DYNAMIC,UP> mtu 1500 qdisc noqueue state DOWN mode DORMANT group default qlen 1000
    link/ether 46:07:b8:4d:43:6b brd ff:ff:ff:ff:ff:ff

user@debian:~$ sudo cat /etc/network/interfaces
# This file describes the network interfaces available on your system
# and how to activate them. For more information, see interfaces(5).
source /etc/network/interfaces.d/*
# The loopback network interface
auto lo
iface lo inet loopback
# The primary network interface
allow-hotplug wlp0s20f3
iface wlp0s20f3 inet dhcp
        wpa-ssid INTERNET_O_SUL_CASA
        wpa-psk  2a7b1c1d
user@debian:~$ sudo cat /etc/network/interfaces.d/
cat: /etc/network/interfaces.d/: É pasta
user@debian:~$ sudo ls /etc/network/interfaces.d/
user@debian:~$ 
```

<details>
<summary>I put the journalctl, if you are patient and in order to take a look.</summary>  
<pre>
<code>

set 07 14:53:09 debian wpa_supplicant[1463]: wlx4607b84d436b: CTRL-EVENT-BEACON-LOSS
set 07 14:53:09 debian connmand[1489]: wlx4607b84d436b {RX} 661579 packets 761943676 bytes
set 07 14:53:09 debian kernel: rtw_8723du 1-4:1.2: failed to get tx report from firmware
set 07 14:53:09 debian connmand[1489]: wlx4607b84d436b {TX} 411021 packets 347041858 bytes
set 07 14:53:09 debian connmand[1489]: wlx4607b84d436b {update} flags 36867 <UP>
set 07 14:53:09 debian connmand[1489]: wlx4607b84d436b {newlink} index 2 address 46:07:B8:4D:43:6B mtu 1500
set 07 14:53:09 debian connmand[1489]: wlx4607b84d436b {newlink} index 2 operstate 2 <DOWN>
set 07 14:53:09 debian wpa_supplicant[1463]: wlx4607b84d436b: CTRL-EVENT-DISCONNECTED bssid=00:19:46:6e:c8:c9 reason=4 locally_generat>
set 07 14:53:09 debian wpa_supplicant[1463]: wlx4607b84d436b: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
set 07 14:53:10 debian avahi-daemon[1390]: Withdrawing address record for 192.168.10.6 on wlx4607b84d436b.
set 07 14:53:10 debian avahi-daemon[1390]: Leaving mDNS multicast group on interface wlx4607b84d436b.IPv4 with address 192.168.10.6.
set 07 14:53:10 debian avahi-daemon[1390]: Interface wlx4607b84d436b.IPv4 no longer relevant for mDNS.
set 07 14:53:10 debian connmand[1489]: wlx4607b84d436b {del} address 192.168.10.6/24 label wlx4607b84d436b
set 07 14:53:10 debian connmand[1489]: wlx4607b84d436b {del} route 192.168.10.0 gw 0.0.0.0 scope 253 <LINK>
set 07 14:53:11 debian wpa_supplicant[1463]: wlx4607b84d436b: SME: Trying to authenticate with 00:19:46:6e:c8:c9 (SSID='INTERNET_O_SUL>
set 07 14:53:11 debian kernel: wlx4607b84d436b: authenticate with 00:19:46:6e:c8:c9
set 07 14:53:11 debian kernel: wlx4607b84d436b: 80 MHz not supported, disabling VHT
set 07 14:53:11 debian connmand[1489]: ipconfig state 7 ipconfig method 1
set 07 14:53:12 debian kernel: wlx4607b84d436b: send auth to 00:19:46:6e:c8:c9 (try 1/3)
set 07 14:53:12 debian connmand[1489]: Probably roaming right now! Staying connected...
set 07 14:53:13 debian kernel: rtw_8723du 1-4:1.2: failed to get tx report from firmware
set 07 14:53:13 debian kernel: wlx4607b84d436b: send auth to 00:19:46:6e:c8:c9 (try 2/3)
set 07 14:53:13 debian kernel: rtw_8723du 1-4:1.2: failed to get tx report from firmware
set 07 14:53:14 debian kernel: wlx4607b84d436b: send auth to 00:19:46:6e:c8:c9 (try 3/3)
set 07 14:53:14 debian kernel: rtw_8723du 1-4:1.2: failed to get tx report from firmware
set 07 14:53:15 debian kernel: wlx4607b84d436b: authentication with 00:19:46:6e:c8:c9 timed out

set 07 19:15:17 debian kernel: rtw_8723du 1-4:1.2: failed to send h2c command
set 07 19:15:17 debian kernel: rtw_8723du 1-4:1.2: failed to send h2c command
set 07 19:16:52 debian kernel: rtw_8723du 1-4:1.2: failed to send h2c command
set 07 19:18:44 debian kernel: rtw_8723du 1-4:1.2: failed to send h2c command
set 07 19:18:44 debian kernel: rtw_8723du 1-4:1.2: failed to send h2c command
set 07 19:22:00 debian kernel: rtw_8723du 1-4:1.2: failed to send h2c command
set 07 19:22:00 debian kernel: rtw_8723du 1-4:1.2: failed to send h2c command
set 07 19:22:12 debian kernel: rtw_8723du 1-4:1.2: failed to send h2c command
set 07 19:22:12 debian kernel: rtw_8723du 1-4:1.2: failed to send h2c command
set 07 19:26:52 debian kernel: rtw_8723du 1-4:1.2: failed to send h2c command
set 07 19:26:52 debian kernel: rtw_8723du 1-4:1.2: failed to send h2c command
[... 377 lines cut ...]
set 08 10:38:13 debian kernel: rtw_8723du 1-4:1.2: failed to send h2c command
set 08 10:38:13 debian kernel: rtw_8723du 1-4:1.2: failed to send h2c command
set 08 10:38:17 debian ntfs-3g[27240]: Unmounting /dev/sdc1 (BACKUP)
set 08 10:38:17 debian kernel: buffer_io_error: 59 callbacks suppressed
set 08 10:38:17 debian kernel: Buffer I/O error on dev sdc1, logical block 280957, lost async page write
set 08 10:38:17 debian kernel: Buffer I/O error on dev sdc1, logical block 631900, lost async page write
set 08 10:38:17 debian kernel: Buffer I/O error on dev sdc1, logical block 970696, lost async page write
[... cut ...]
set 08 10:44:20 debian nordvpnd[3244]: 2023/09/08 10:44:20 Get "https://api.nordvpn.com/v1/helpers/ips/insights": lookup api.nordvpn.com on 103.86.99.100:53: dial udp 103.86.99.100:53: connect: network is unreachable
set 08 10:44:20 debian nordvpnd[3244]: 2023/09/08 10:44:20 rotated api transport to: std-http-transport
set 08 10:44:20 debian nordvpnd[3244]: 2023/09/08 10:44:20 Get "https://api.nordvpn.com/v1/helpers/ips/insights": looking address up: looking addr ip up: lookup api.nordvpn.com on 103.86.99.100:53: dial udp 103.86.99.100:53: connect: network is unreachable
(END)
/code>
</pre>
</details>

Would you recommend me a study material to understand this subject?
Thanks.

brianwitte, 2023-09-10:
Hi, @RENANZG. Thank you for following up on this issue with us, Larry and I are very limited in our ability to help with driver specific issues after it has built successfully or any of the disk errors. The scope of this repository and what we can troubleshoot is limited to build errors when you build and install the driver.

I will try to help you get moving in the right direction, but will be closing this issue with this comment.

From the look of your journalctl output, I see four main things:

1. Messages like CTRL-EVENT-BEACON-LOSS and CTRL-EVENT-DISCONNECTED from wpa_supplicant imply the connection is unstable or lost.
2. The "rtw_8723du 1-4:1.2: failed to send h2c command" error indeed suggests an issue with the rtw_8723du driver itself.
3. The Inconsistent IP pool management (start not found) suggests an issue in the IP address management, possibly related to DHCP.
4. The mounted usb device you have still has those I/O and syncing errors which needs to be addressed separately.

None of these seem to be related to how you are building the driver. I would try to clear yourself of the disk errors and then go from there.

If you are still seeing unstable connection issues related to `rtw_8723du` , reach out to Realtek engineers via E-mail at [linux-wireless@vger.kernel.org](mailto:linux-wireless@vger.kernel.org), but you may not get much help unless your issue is reproducible and thoroughly explained/documented.

closed 2023-09-10 by brianwitte