lwfinger/rtw88 #163: Is rtw_8723du.ko signature missing in make file ?

RENANZG, 2023-09-21:
Is rtw_8723du.ko signature missing in make file ?

```
_sign:
ifeq ($(NO_SKIP_SIGN), y)
	@openssl req -new -x509 -newkey rsa:2048 -keyout MOK.priv -outform DER -out MOK.der -nodes -days 36500 -subj "/CN=Custom MOK/"
	@mokutil --import MOK.der
else
	echo "Skipping key creation"
endif
	@$(KSRC)/scripts/sign-file sha256 MOK.priv MOK.der rtw_pci.ko
	@$(KSRC)/scripts/sign-file sha256 MOK.priv MOK.der rtw_core.ko
	@$(KSRC)/scripts/sign-file sha256 MOK.priv MOK.der rtw_8723d.ko
	@$(KSRC)/scripts/sign-file sha256 MOK.priv MOK.der rtw_8723de.ko
	@$(KSRC)/scripts/sign-file sha256 MOK.priv MOK.der rtw_8822b.ko
	@$(KSRC)/scripts/sign-file sha256 MOK.priv MOK.der rtw_8822be.ko
	@$(KSRC)/scripts/sign-file sha256 MOK.priv MOK.der rtw_8822bu.ko
	@$(KSRC)/scripts/sign-file sha256 MOK.priv MOK.der rtw_8822bs.ko
	@$(KSRC)/scripts/sign-file sha256 MOK.priv MOK.der rtw_8821c.ko
	@$(KSRC)/scripts/sign-file sha256 MOK.priv MOK.der rtw_8821ce.ko
	@$(KSRC)/scripts/sign-file sha256 MOK.priv MOK.der rtw_8821cu.ko
	@$(KSRC)/scripts/sign-file sha256 MOK.priv MOK.der rtw_8821cs.ko
	@$(KSRC)/scripts/sign-file sha256 MOK.priv MOK.der rtw_8822c.ko
	@$(KSRC)/scripts/sign-file sha256 MOK.priv MOK.der rtw_8822ce.ko
	@$(KSRC)/scripts/sign-file sha256 MOK.priv MOK.der rtw_8822cu.ko
	@$(KSRC)/scripts/sign-file sha256 MOK.priv MOK.der rtw_8822cs.ko

sign-install: all sign install_
```

Thanks

lwfinger, 2023-09-21:
The signing of rtw_8723du and rtw_8723ds were missing. I just added them. Do a 'git pull' and you should be able to sign the .ko.

closed 2023-09-21 by lwfinger

RENANZG, 2023-09-21:
[image]

lwfinger, 2023-09-21:
Cute. I like that.  Another happy customer.

RENANZG, 2023-10-03:
Hello. I did a tutorial to manually enable secure boot in Debian.  Another detail, the "unsigned" makefile script seems to be still signing the modules. It might be interesting to put this in the readme for those who prefer to do it manually.

----------------

How to get WiFi Module signed for Secure Boot

Mandatory packages if Secure Boot is active: openssl, sign-file and mokutil

1. You can create a personal public/private RSA key pair to sign the kernel modules. You can chose to store the key/pair, for example, in the <ins>/var/lib/shim-signed/modules/</ins> directory. Then create a new pair of private key (module.priv) and public key (module.der).

```console
$ sudo mkdir -p var/lib/shim-signed/modules
$ sudo openssl req -config /usr/lib/ssl/openssl.cnf -new -x509 -newkey rsa:2048 -nodes -days 36500 -outform DER -keyout "/var/lib/shim-signed/modules/module.priv" -out "/var/lib/shim-signed/modules/module.der" -subj "/CN=Modules/"
$ ls -l /var/lib/shim-signed/modules/
total 8
-rw-r--r-- 1 root root  779 module.der
-rw------- 1 root root 1704 module.priv
$ sudo chmod 600 /var/lib/shim-signed/modules/*
```

2. Enroll the public key (VirtualBox.der) to MOK (Machine Owner Key) by entering the command:
```
$ sudo mokutil --import /var/lib/shim-signed/modules/module.der
input password:
input password again:
```
Recheck if your key will be prompted on next boot:
```
$ sudo mokutil --list-new
```

3. Reboot and check

The password in this step is a temporary use password you'll only need to remember for a few minutes. Reboot the machine. When the bootloader starts, you should see a screen asking you to press a button to enter the MOK manager EFI utility. Note that any external external keyboards won't work in this step. Select Enroll MOK in the first menu, then continue, and then select Yes to enroll the keys, and re-enter the password established in previous step. Then select OK to continue the system boot.

There are several ways to verify if your key "Modules" is loaded and signed

```
$ sudo mokutil --test-key /var/lib/shim-signed/modules/module.der
$ sudo lsmod | grep rtw_8723d
$ sudo modinfo -n rtw_8723d
$ sudo dmesg | grep cert
$ sudo dmesg | grep Modules
```

4. Sign the module 

Where was the module installed?
```
$ sudo modinfo -n rtw_8723d
  /lib/modules/6.1.0-12-amd64/kernel/drivers/net/wireless/realtek/rtw88/rtw_8723d.ko
```
For sing the module, depending on your platform, the exact location of `sign-file` might vary. In Debian 12 (Bookworm) it was in <ins>/usr/src/linux-headers-$(uname -r)/scripts/sign-file</ins> .

```
$ uname -r
  6.1.0-12-amd64
$ /usr/src/linux-headers-6.1.0-12-amd64/scripts/sign-file 
Usage: scripts/sign-file [-dp] <hash algo> <key> <x509> <module> [<dest>]
       scripts/sign-file -s <raw sig> <hash algo> <x509> <module> [<dest>]
```
Sign the module:
```
$ sudo /usr/src/linux-headers-6.1.0-12-amd64/scripts/sign-file sha256 /var/lib/shim-signed/modules/module.priv /var/lib/shim-signed/modules/module.der /lib/modules/6.1.0-12-amd64/kernel/drivers/net/wireless/realtek/rtw88/rtw_8723d.ko
```
Verify it:
```
$ sudo modinfo rtw88_8723d
(...)
signer:         Modules
sig_key:        XX:XX:XX:XX:XX:XX:XX:XX...
sig_hashalgo:   sha256
signature:      XX:XX:XX:XX:XX:XX:XX:XX...
(...)
```
If you have your own keys:
```
$ sudo /usr/src/linux-headers-$(uname -r)/scripts/sign-file sha256 /var/lib/shim-signed/mok/MOK.priv /var/lib/shim-signed/mok/MOK.der /lib/modules/6.1.0-12-amd64/misc/vboxdrv.ko
```

Other form
```
$ sudo /usr/src/linux-headers-$(uname -r)/scripts/sign-file sha256 ./MOK.priv ./MOK.der "$(modinfo -n vboxdrv)"

```
If the modules are needed to boot your machine, make sure to update the initramfs, e.g. using
```
sudo update-initramfs -k all -u
```

----------------------

**THANKS**

RENANZG, 2023-10-07:
For information.  There is a serious problem with module signing in Debian.

```diff
- Debian Bug report logs - [#1012741](mailto:1012741@bugs.debian.org) modprobe: ERROR: could not insert 'crc_itu_t': Key was rejected by service
- https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1012741
```
Brief here:

https://lowendtalk.com/discussion/182547/fixing-debian-error-modprobe-error-could-not-insert-key-was-rejected-by-service

"Looks like there has been a bug in the package signer that outputs signatures that are 2 bytes short if the upper 2 bytes of something are 0 (so it thinks it can fit in a 16-bit rather than 32-bit integer). So, out of the millions of packages in Debian, there are around 75 that have incorrect signatures.

Anyway, the good news is that it's now fixed in release, but seemingly not fixed in every provider's templates. But the solution is 
pretty easy now the fix is released:

``` bash
apt-get update
apt-get dist-upgrade
```
"

lwfinger, 2023-10-07:
Thanks for the info.

RENANZG, 2023-10-07:
Larry, there is a typo in the end of opeenssl.conf:
```diff
- nsComment = "OpenSSL Generated Certificate"0`
```
```diff
+ nsComment = "OpenSSL Generated Certificate"
```