lwfinger/rtw88 #168: rtl8822bu hotspot not working for all devices

sskaje, 2023-10-21:
RTL Device: Asus AC53 Nano, 0b05:184c
Driver: rtw88, rtw_8822bu

Platform/OS for hotspot:
* NanoPi R6s + FriendlyWrt 
  * kernel 5.10
  * kernel 6.1 (arm64)
* Debian bookworm (amd64) + kernel 6.4

brief settings:
* Country Code: US
* OpenWrt Mode: AC, channel 149, bandwidth 80Mhz
* Security:
  * WPA2 PSK CCMP
  * Open 

Client devices: 
* iPhone 8 Plus 
* iPhone 14 Pro
* MacBook Pro 2015 + macOS 13.4
* Dell Laptop + QC61x4A +
  * Windows 11  
  * Debian bookworm (amd64) + kernel 6.4

I tried on 3 Platforms, all with rtw_8822bu from rtw88, same behaviours. This makes me feel the problem may be caused by the hardware itself or driver.

The interesting thing is, only iPhone 14 Pro can successfully connect to the Hotspot, and works well. 

I try with other devices, timeout, asking for password again, repeats.

Once I have a not-working device, tried to connect to the AP, the AP will soon be dead.

dmesg from FriendlyWrt with comments
```
[   22.068944] br-lan: port 3(wlan0) entered forwarding state
[   24.264474] platform mtd_vendor_storage: deferred probe pending

# iPhone 14 Pro connected before below lines

# when I try to connect with an iPhone 8 Plus, cannot join, error below
# sometimes it asks to enter correct password

[  365.541003] rtw_8822bu 1-1:1.0: timed out to flush queue 3

# retry with iphone 8 plus, cannot join again

[  450.647958] rtw_8822bu 1-1:1.0: timed out to flush queue 3

# Tried to connect with Windows 11, first windows get connected, 
# soon web connection became extremely slow, and then, AP dies

[  490.527071] rtw_8822bu 1-1:1.0: timed out to flush queue 3
[  496.237008] rtw_8822bu 1-1:1.0: timed out to flush queue 3
[  496.696988] rtw_8822bu 1-1:1.0: timed out to flush queue 3
[  501.986885] rtw_8822bu 1-1:1.0: timed out to flush queue 3
[  507.716908] rtw_8822bu 1-1:1.0: timed out to flush queue 3
[  513.466701] rtw_8822bu 1-1:1.0: timed out to flush queue 3
[  516.818020] rtw_8822bu 1-1:1.0: error beacon valid
[  516.818498] rtw_8822bu 1-1:1.0: failed to download drv rsvd page
[  517.118912] rtw_8822bu 1-1:1.0: error beacon valid
[  517.119457] rtw_8822bu 1-1:1.0: failed to download drv rsvd page
[  517.325996] rtw_8822bu 1-1:1.0: error beacon valid
[  517.326521] rtw_8822bu 1-1:1.0: failed to download drv rsvd page
[  519.724626] rtw_8822bu 1-1:1.0: error beacon valid
[  519.725261] rtw_8822bu 1-1:1.0: failed to download drv rsvd page
```

I also tried other security settings, same behaviors.

And an old rtl8822bu driver (https://github.com/friendlyarm/rtl8822bu) can get this device working as AP smoothly, all devices above can get connected, only problem is that driver only have the dongle running at 802.11a mode and channel bandwidth 20Mhz. 

Any clue where I can collect more logs for debugging?

```
root@FriendlyWrt:~# iw list
Wiphy phy0
		wiphy index: 0
	max # scan SSIDs: 4
	max scan IEs length: 2243 bytes
	max # sched scan SSIDs: 0
	max # match sets: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Device supports T-DLS.
	Available Antennas: TX 0x3 RX 0x3
	Configured Antennas: TX 0x3 RX 0x3
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
		 * mesh point
	Band 1:
		Capabilities: 0x196f
			RX LDPC
			HT20/HT40
			SM Power Save disabled
			RX HT20 SGI
			RX HT40 SGI
			RX STBC 1-stream
			Max AMSDU length: 7935 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 2 usec (0x04)
		HT Max RX data rate: 300 Mbps
		HT TX/RX MCS rate indexes supported: 0-15, 32
		Frequencies:
			* 2412 MHz [1] (30.0 dBm)
			* 2417 MHz [2] (30.0 dBm)
			* 2422 MHz [3] (30.0 dBm)
			* 2427 MHz [4] (30.0 dBm)
			* 2432 MHz [5] (30.0 dBm)
			* 2437 MHz [6] (30.0 dBm)
			* 2442 MHz [7] (30.0 dBm)
			* 2447 MHz [8] (30.0 dBm)
			* 2452 MHz [9] (30.0 dBm)
			* 2457 MHz [10] (30.0 dBm)
			* 2462 MHz [11] (30.0 dBm)
			* 2467 MHz [12] (disabled)
			* 2472 MHz [13] (disabled)
			* 2484 MHz [14] (disabled)
	Band 2:
		Capabilities: 0x196f
			RX LDPC
			HT20/HT40
			SM Power Save disabled
			RX HT20 SGI
			RX HT40 SGI
			RX STBC 1-stream
			Max AMSDU length: 7935 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 2 usec (0x04)
		HT Max RX data rate: 300 Mbps
		HT TX/RX MCS rate indexes supported: 0-15, 32
		VHT Capabilities (0x03d071b2):
			Max MPDU length: 11454
			Supported Channel Width: neither 160 nor 80+80
			RX LDPC
			short GI (80 MHz)
			TX STBC
			SU Beamformee
			MU Beamformee
			+HTC-VHT
		VHT RX MCS set:
			1 streams: MCS 0-9
			2 streams: MCS 0-9
			3 streams: not supported
			4 streams: not supported
			5 streams: not supported
			6 streams: not supported
			7 streams: not supported
			8 streams: not supported
		VHT RX highest supported: 780 Mbps
		VHT TX MCS set:
			1 streams: MCS 0-9
			2 streams: MCS 0-9
			3 streams: not supported
			4 streams: not supported
			5 streams: not supported
			6 streams: not supported
			7 streams: not supported
			8 streams: not supported
		VHT TX highest supported: 780 Mbps
		Frequencies:
			* 5180 MHz [36] (23.0 dBm)
			* 5200 MHz [40] (23.0 dBm)
			* 5220 MHz [44] (23.0 dBm)
			* 5240 MHz [48] (23.0 dBm)
			* 5260 MHz [52] (23.0 dBm) (radar detection)
			* 5280 MHz [56] (23.0 dBm) (radar detection)
			* 5300 MHz [60] (23.0 dBm) (radar detection)
			* 5320 MHz [64] (23.0 dBm) (radar detection)
			* 5500 MHz [100] (23.0 dBm) (radar detection)
			* 5520 MHz [104] (23.0 dBm) (radar detection)
			* 5540 MHz [108] (23.0 dBm) (radar detection)
			* 5560 MHz [112] (23.0 dBm) (radar detection)
			* 5580 MHz [116] (23.0 dBm) (radar detection)
			* 5600 MHz [120] (23.0 dBm) (radar detection)
			* 5620 MHz [124] (23.0 dBm) (radar detection)
			* 5640 MHz [128] (23.0 dBm) (radar detection)
			* 5660 MHz [132] (23.0 dBm) (radar detection)
			* 5680 MHz [136] (23.0 dBm) (radar detection)
			* 5700 MHz [140] (23.0 dBm) (radar detection)
			* 5720 MHz [144] (23.0 dBm) (radar detection)
			* 5745 MHz [149] (30.0 dBm)
			* 5765 MHz [153] (30.0 dBm)
			* 5785 MHz [157] (30.0 dBm)
			* 5805 MHz [161] (30.0 dBm)
			* 5825 MHz [165] (30.0 dBm)
	interface combinations are not supported
	HT Capability overrides:
		 * MCS: ff ff ff ff ff ff ff ff ff ff
		 * maximum A-MSDU length
		 * supported channel width
		 * short GI for 40 MHz
		 * max A-MPDU length exponent
		 * min MPDU start spacing
	max # scan plans: 1
	max scan plan interval: -1
	max scan plan iterations: 0
	Supported extended features:
		* [ RRM ]: RRM
		* [ SET_SCAN_DWELL ]: scan dwell setting
		* [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
		* [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
		* [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
		* [ SCAN_RANDOM_SN ]: use random sequence numbers in scans
		* [ CAN_REPLACE_PTK0 ]: can safely replace PTK 0 when rekeying
		* [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
		* [ DEL_IBSS_STA ]: deletion of IBSS station support
		* [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
		* [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support
		
```

Samt43, 2023-10-27:
Looks like the same problem as i had : https://github.com/lwfinger/rtw88/issues/112
Have you tried in 2.4GHz ?
For me i could never solve this issue for now and i'm still using the old driver due to this, it's directly related to the driver/firmware.
Good luck !

Samt43, 2023-10-27:
https://github.com/lwfinger/rtw88/issues/151 could also related

sskaje, 2023-10-27:
> Looks like the same problem as i had : #112 Have you tried in 2.4GHz ? For me i could never solve this issue for now and i'm still using the old driver due to this, it's directly related to the driver/firmware. Good luck !

I did try a very old version of driver, https://github.com/friendlyarm/rtl8822bu, only 2.4GHz can be recognized, and hotspot works fine.

The interesting part is, with such an old dongle (6+years), only the iPhone 14 pro works. 
This makes me want to find out what is different from my iPhone 14 pro and other devices during Wi-Fi connection establishing, and I'm trying to find out what I should do and how.

And yes, I read #151, error from https://github.com/lwfinger/rtw88/issues/151#issuecomment-1755895931 looks same as mine.

dubhater, 2024-07-25:
This problem may be [fixed](https://github.com/lwfinger/rtw88/commit/4a1ee64475672a9186e10d87ac5e251448f86914) now.

dubhater, 2025-02-06:
Probably fixed in 6.12.

closed 2025-02-06 by dubhater