lwfinger/rtw88 #169: rtw8822ce not working for wpa2 enterprise wifi

rajb957, 2023-10-24:
Hi!

I am using rtw8822ce driver in my hp pavilion 14. It worked perfectly for years but after a recent update onwards it disconnected from wpa2 enterprise wifi with no fix even on downgrading the kernel. It works for personal wifi though. the dmesg -wL just gives time out

```
[   35.002485] wlo1: deauthenticating from 1e:5c:f2:17:44:11 by local choice (Reason: 3=DEAUTH_LEAVING)
[   35.537232] wlo1: authenticate with 38:17:c3:b5:bb:d1
[   35.911908] wlo1: send auth to 38:17:c3:b5:bb:d1 (try 1/3)
[   35.917916] wlo1: send auth to 38:17:c3:b5:bb:d1 (try 2/3)
[   35.924576] wlo1: send auth to 38:17:c3:b5:bb:d1 (try 3/3)
[   35.930914] wlo1: authentication with 38:17:c3:b5:bb:d1 timed out
[   36.430724] wlo1: authenticate with 38:17:c3:b5:bd:31
[   36.809990] wlo1: send auth to 38:17:c3:b5:bd:31 (try 1/3)
[   36.815904] wlo1: send auth to 38:17:c3:b5:bd:31 (try 2/3)
[   36.822661] wlo1: send auth to 38:17:c3:b5:bd:31 (try 3/3)
[   36.829026] wlo1: authentication with 38:17:c3:b5:bd:31 timed out
[   37.674055] wlo1: authenticate with e8:26:89:79:2e:d0
[   38.099428] wlo1: send auth to e8:26:89:79:2e:d0 (try 1/3)
[   38.107108] wlo1: send auth to e8:26:89:79:2e:d0 (try 2/3)
[   38.113671] wlo1: send auth to e8:26:89:79:2e:d0 (try 3/3)
[   38.120269] wlo1: authentication with e8:26:89:79:2e:d0 timed out
[   39.442264] wlo1: authenticate with 38:17:c3:b5:bb:c1
[   39.442303] wlo1: 80 MHz not supported, disabling VHT
[   39.808059] wlo1: send auth to 38:17:c3:b5:bb:c1 (try 1/3)
[   39.842593] wlo1: send auth to 38:17:c3:b5:bb:c1 (try 2/3)
[   39.876337] wlo1: send auth to 38:17:c3:b5:bb:c1 (try 3/3)
[   39.914629] wlo1: authentication with 38:17:c3:b5:bb:c1 timed out
```

It is dual boot laptop with windows 11 installed where it works nicely with the same credentials and without any interference i checked other distros as well namely fedora, zorin and ubuntu all of which have the same error. most likely it has something to do with the recent update on any of the driver rtw8822ce

the driver info (lspci -nnk) is as follows

```
02:00.0 Network controller [0280]: Realtek Semiconductor Co., Ltd. RTL8822CE 802.11ac PCIe Wireless Network Adapter [10ec:c822]
	DeviceName: WLAN
	Subsystem: Hewlett-Packard Company Device [103c:85f7]
	Kernel driver in use: rtw_8822ce
	Kernel modules: rtw88_8822ce, rtw_8822ce
```

Even on changing the kernel driver in use using 
sudo modprobe -rfv rtw_8822ce
sudo modprobe -rfv rtw88_8822ce

it doesn't work even for the other drivers as well i.e.

```
02:00.0 Network controller [0280]: Realtek Semiconductor Co., Ltd. RTL8822CE 802.11ac PCIe Wireless Network Adapter [10ec:c822]
	DeviceName: WLAN
	Subsystem: Hewlett-Packard Company Device [103c:85f7]
	Kernel driver in use: rtw_8822ce
	Kernel modules: rtw88_8822ce, rtw_8822
	
```

This has the same issue as well.

Pls look into it and feel free to reach out to me if you think if it is a kernel problem

lwfinger, 2023-10-30:
I do not think it is a kernel or driver problem. I think your distro broke the user-space part of the connection to enterprise servers. The fact that personal wifi works shows that wpa_supplicant is OK, but the setup for enterprise is broken.

rajb957, 2023-10-30:
What should I do to fix this issue? I've tried booting into different distros like Ubuntu, Zorin OS, and Fedora, but it hasn't resolved the problem. I've also attempted common solutions I found on the internet, but most of them were specific to drivers or WiFi using legacy security, neither of which seem relevant to my situation. In terms of the enterprise WiFi setup, many students at my college use Linux for development purposes and don't encounter WiFi issues, so I don't believe the issue is with the WiFi itself. I also reached out to the IT department, and they've narrowed it down to the fact that the WiFi isn't receiving any connection requests from my laptop. I'm a bit confused about how to narrow down and identify the specific problem. Could you recommend a course of action that might help me understand what's going wrong?

dubhater, 2023-10-30:
Maybe it's a wifi firmware problem. Have you tried downgrading that?

https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/log/rtw88

You can see in dmesg which version you have. If you have journalctl or similar you can also see what version you had before the update which broke things.

lwfinger, 2023-10-30:
It is not a firmware problem. His distro updated the software then initiates an enterprise connection, and broke it. He needs to ask on their forums to see if anyone else has problems, or he needs to change his network control software to NetworkManager, or some other software if he is already using NM. I have no experience with enterprise network control. I tried setting up a server once, but never got it to work.

rajb957, 2023-10-31:
On a serious note, since I've encountered these issues across multiple distros, it's possible that the Linux kernel is directly responsible for the problem. I did attempt to downgrade the kernel, even going back to one as old as three years, but unfortunately, it didn't resolve the issue. Narrowing down the root cause has been a significant challenge. It seems to me that some other software, like Pipewire or potentially a driver related to the touchpad or another device, may be interfering with the WiFi setup or accessing something it shouldn't, leading to these errors. However, I have very little knowledge about how the Linux kernel works beneath certain protocols.

I'm looking for an ideal forum or platform to post such queries.

lwfinger, 2023-10-31:
You have not told us what distro you are using, nor what network control software. Without such critical info, how can we help?

Your IT department showed that you are not making any attempts to connect. That is the problem, not the kernel, or any other software than your network control package. Either it is faulty, or your configuration is incorrect. Run every part of the network configuration with maximum debugging.

rajb957, 2023-10-31:
I am currently using Fedora 38, but I have also tried booting into Zorin OS, Ubuntu, and a fresh Fedora installation. The network control software I am using is NetworkManager.

While trying to debug as you suggested the problem and following the commands for debugging in this https://access.redhat.com/documentation/en-us/red_hat_enterprise_linux/8/html/configuring_and_managing_networking/introduction-to-networkmanager-debugging_configuring-and-managing-networking

It somehow connected again and started working perfectly fine as before i don't know why setting RateLimitBurst=0
and adding this as told works but it miraculously works. Didn't understand what worked but thanks for for helping me out

closed 2023-10-31 by rajb957

linwei5d, 2023-11-18:
I meet similar problem,and I'm confused why there is a "wpa_supplicant[1888]: rmdir[ctrl_interface=/var/run/wpa_supplicant]: Permission denied".This may not be the problem of the driver,but I really need some help.Device: Tenda U9(RTL8811CU).I'm sure the passwd is correct.

`Sat Nov 18 07:01:06 2023 daemon.notice wpa_supplicant[1888]: phy1-sta0: SME: Trying to authenticate with ca:b9:1e:26:2f:49 (SSID='Keyanketang' freq=5180 MHz)
Sat Nov 18 07:01:06 2023 kern.info kernel: [  487.347406] phy1-sta0: authenticate with ca:b9:1e:26:2f:49
Sat Nov 18 07:01:06 2023 kern.info kernel: [  487.387320] phy1-sta0: send auth to ca:b9:1e:26:2f:49 (try 1/3)
Sat Nov 18 07:01:07 2023 kern.info kernel: [  488.830781] phy1-sta0: send auth to ca:b9:1e:26:2f:49 (try 2/3)
Sat Nov 18 07:01:08 2023 kern.info kernel: [  489.822769] phy1-sta0: send auth to ca:b9:1e:26:2f:49 (try 3/3)
Sat Nov 18 07:01:09 2023 kern.info kernel: [  490.818757] phy1-sta0: authentication with ca:b9:1e:26:2f:49 timed out
Sat Nov 18 07:01:09 2023 daemon.notice wpa_supplicant[1888]: phy1-sta0: CTRL-EVENT-SSID-TEMP-DISABLED id=1 ssid="Keyanketang" auth_failures=5 duration=60 reason=CONN_FAILED
Sat Nov 18 07:01:17 2023 daemon.notice netifd: radio1 (10950): ./mac80211.sh: eval: line 1007: ipconfig: not found
Sat Nov 18 07:01:17 2023 daemon.notice netifd: radio1 (10950): ./mac80211.sh: eval: line 1077: ipconfig: not found
Sat Nov 18 07:01:17 2023 daemon.notice netifd: radio1 (10950): ./mac80211.sh: eval: line 1078: ipconfig: not found
Sat Nov 18 07:01:18 2023 daemon.notice netifd: radio1 (10950): ./mac80211.sh: eval: line 1108: ipconfig: not found
Sat Nov 18 07:01:18 2023 daemon.notice netifd: radio1 (10950): sh: out of range
Sat Nov 18 07:01:18 2023 daemon.notice wpa_supplicant[1888]: Set new config for phy phy1
Sat Nov 18 07:01:18 2023 daemon.notice hostapd: Set new config for phy phy1:
Sat Nov 18 07:01:18 2023 daemon.notice wpa_supplicant[1888]: Set new config for phy phy1
Sat Nov 18 07:01:18 2023 daemon.notice netifd: Wireless device 'radio1' is now up
Sat Nov 18 07:01:25 2023 daemon.notice ksmbd: Stopping Ksmbd userspace service.
Sat Nov 18 07:01:25 2023 kern.warn kernel: [  506.174008] ksmbd: The ksmbd server is experimental
Sat Nov 18 07:01:25 2023 daemon.notice ksmbd: Starting Ksmbd userspace service.
Sat Nov 18 07:01:26 2023 daemon.warn odhcpd[2310]: No default route present, overriding ra_lifetime!
Sat Nov 18 07:02:03 2023 daemon.notice hostapd: Set new config for phy phy1:
Sat Nov 18 07:02:03 2023 daemon.notice wpa_supplicant[1888]: Set new config for phy phy1
Sat Nov 18 07:02:03 2023 daemon.notice wpa_supplicant[1888]: Remove interface phy1-sta0
Sat Nov 18 07:02:03 2023 daemon.notice wpa_supplicant[1888]: phy1-sta0: CTRL-EVENT-DSCP-POLICY clear_all
Sat Nov 18 07:02:03 2023 daemon.notice wpa_supplicant[1888]: phy1-sta0: CTRL-EVENT-DSCP-POLICY clear_all
Sat Nov 18 07:02:03 2023 daemon.notice wpa_supplicant[1888]: nl80211: deinit ifname=phy1-sta0 disabled_11b_rates=0
Sat Nov 18 07:02:03 2023 daemon.err wpa_supplicant[1888]: rmdir[ctrl_interface=/var/run/wpa_supplicant]: Permission denied
Sat Nov 18 07:02:03 2023 daemon.notice netifd: Interface 'wwan' is disabled
Sat Nov 18 07:02:03 2023 daemon.notice netifd: Wireless device 'radio1' is now down
Sat Nov 18 07:02:03 2023 daemon.notice netifd: radio1 (11707): WARNING: Variable 'data' does not exist or is not an array/object
Sat Nov 18 07:02:03 2023 daemon.notice netifd: radio1 (11707): ./mac80211.sh: eval: line 1007: ipconfig: not found
Sat Nov 18 07:02:03 2023 daemon.notice netifd: radio1 (11707): ./mac80211.sh: eval: line 1077: ipconfig: not found
Sat Nov 18 07:02:03 2023 daemon.notice netifd: radio1 (11707): ./mac80211.sh: eval: line 1078: ipconfig: not found
Sat Nov 18 07:02:04 2023 daemon.notice hostapd: Set new config for phy phy1:
Sat Nov 18 07:02:04 2023 daemon.notice wpa_supplicant[1888]: Set new config for phy phy1
Sat Nov 18 07:02:04 2023 daemon.notice netifd: radio1 (11707): ./mac80211.sh: eval: line 1108: ipconfig: not found
Sat Nov 18 07:02:04 2023 daemon.notice netifd: radio1 (11707): sh: out of range
Sat Nov 18 07:02:04 2023 daemon.notice wpa_supplicant[1888]: Set new config for phy phy1
Sat Nov 18 07:02:04 2023 daemon.notice hostapd: Set new config for phy phy1:
Sat Nov 18 07:02:04 2023 daemon.notice wpa_supplicant[1888]: Set new config for phy phy1
Sat Nov 18 07:02:05 2023 daemon.notice netifd: Wireless device 'radio1' is now up
Sat Nov 18 07:02:05 2023 daemon.notice netifd: Interface 'wwan' is enabled
Sat Nov 18 07:02:16 2023 daemon.notice wpa_supplicant[1888]: phy1-sta0: SME: Trying to authenticate with ca:b9:1e:26:2f:49 (SSID='Keyanketang' freq=5180 MHz)
Sat Nov 18 07:02:16 2023 kern.info kernel: [  556.993475] phy1-sta0: authenticate with ca:b9:1e:26:2f:49
Sat Nov 18 07:02:16 2023 kern.info kernel: [  557.031324] phy1-sta0: send auth to ca:b9:1e:26:2f:49 (try 1/3)
Sat Nov 18 07:02:16 2023 kern.info kernel: [  557.790768] phy1-sta0: send auth to ca:b9:1e:26:2f:49 (try 2/3)
Sat Nov 18 07:02:17 2023 kern.info kernel: [  558.846798] phy1-sta0: send auth to ca:b9:1e:26:2f:49 (try 3/3)
Sat Nov 18 07:02:18 2023 kern.info kernel: [  559.810764] phy1-sta0: authentication with ca:b9:1e:26:2f:49 timed out
Sat Nov 18 07:02:26 2023 daemon.notice wpa_supplicant[1888]: phy1-sta0: SME: Trying to authenticate with ca:b9:1e:26:2f:49 (SSID='Keyanketang' freq=5180 MHz)
Sat Nov 18 07:02:26 2023 kern.info kernel: [  567.921078] phy1-sta0: authenticate with ca:b9:1e:26:2f:49
[... cut ...]
Sat Nov 18 07:03:07 2023 kern.info kernel: [  608.830786] phy1-sta0: send auth to ca:b9:1e:26:2f:49 (try 2/3)
Sat Nov 18 07:03:08 2023 kern.info kernel: [  609.790776] phy1-sta0: send auth to ca:b9:1e:26:2f:49 (try 3/3)
Sat Nov 18 07:03:09 2023 kern.info kernel: [  610.782774] phy1-sta0: authentication with ca:b9:1e:26:2f:49 timed out
Sat Nov 18 07:03:09 2023 daemon.notice wpa_supplicant[1888]: phy1-sta0: CTRL-EVENT-SSID-TEMP-DISABLED id=2 ssid="Keyanketang" auth_failures=2 duration=20 reason=CONN_FAILED`

rajb957, 2023-11-18:
I started getting the same error after few hours of working. Although i was a busy and hence used mobile hotspot, It still doesn't work in my case. In past there have been many such problems like https://askubuntu.com/questions/285234/cannot-connect-to-wpa2-wpa-enterprise-peap-and-mschap but they eventually got fixed so i think it's wise enough to wait until it get's fixed.

Does it work for personal wifi in your case?

KareemMukbil, 2024-03-08:
Have you managed to fix this?

lwfinger, 2024-03-08:
No. I do not have access to an enterprise AP. You need to capture the "over the air" packets and report this issue with the file of captured packets on linux-wireless@vger.kernel.org as this problem is beyond the expertise of the GitHub issues.