lwfinger/rtw88 #170: Realtek 8822cu : AP enabled but no SSID appear on clients

elclaudio, 2023-11-12:
Hi guys

I've ordered 2 wifi dongles based on (supposedly) mediatek chipset MT7612 and MT7921 (as per online description) to test and eventually serve as an access point on my X86 router. However, it seem's that one of them is in fact a realtek 8822cu inside... plus a dual state one : it appear as a cdrom device, not a big deal, using usb modeswitch, it switch back to a nic interface.

So, I've compiled rtw88 driver ( https://github.com/lwfinger/rtw88 ) installed it. module is loaded, wifi interface is detected, hostapd run  and AP-ENABLED is showing on terminal. However, I don't see any SSID of it on differents clients, and can't connect to this AP; Changing settings from 2.4 Ghz or 5 Ghz doesn't change anything. So I wonder what could be wrong. I'm open to suggestions. 

tests was done on an odroid-c2 runnig armbian kernel 6.1 (Linux odroidc2 6.1.50-current-meson64)

`Bus 001 Device 006: ID 0bda:c812 Realtek Semiconductor Corp. 802.11ac NIC`

```
[ 1578.842834] rtw_core: loading out-of-tree module taints kernel.
[ 1578.850560] rtw_8822cu 1-1.1:1.0: Firmware version 9.9.13, H2C version 15
[ 1578.850894] rtw_8822cu 1-1.1:1.0: WOW Firmware version 9.9.4, H2C version 15
[ 1579.219416] usbcore: registered new interface driver rtw_8822cu
[ 1579.239648] rtw_8822cu 1-1.1:1.0 wlxXXXXXXXXX: renamed from wlan0
[ 7479.353310] rtw_8822cu 1-1.1:1.0: Firmware version 9.9.13, H2C version 15
[ 7479.353632] rtw_8822cu 1-1.1:1.0: WOW Firmware version 9.9.4, H2C version 15
[ 7479.643138] rtw_8822cu 1-1.1:1.0 wlxXXXXXXXXX: renamed from wlan0
```
```

wlxXXXXXXXXX: interface state UNINITIALIZED->COUNTRY_UPDATE
wlxXXXXXXXXX: interface state COUNTRY_UPDATE->HT_SCAN
wlxXXXXXXXXX: interface state HT_SCAN->ENABLED
wlxXXXXXXXXX: AP-ENABLED
```

[image]

elclaudio, 2023-11-12:
add up: tested also firmware 9.9.15 : no changes [https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/rtw88](https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/rtw88)
)
tested on a vm with fedora 38 x64, kernel Linux fedora 6.2.9-300.fc38.x86_64 : rtw88 is already installed, so not compiled . the dongle is recognized out of the box but same symptoms : tha access point is up but no ssid appear on clients, like if the network doesn't exists at all.

lwfinger, 2023-11-12:
This complaint should be posted on the mailing list linux-wireless@vger.kernel.org as this is a problem with the driver. In this GitHub issue set, I am involved with keeping the driver up to date, and making sure that it builds on older kernels.

elclaudio, 2023-11-12:
> This complaint should be posted on the mailing list [linux-wireless@vger.kernel.org](mailto:linux-wireless@vger.kernel.org) as this is a problem with the driver. In this GitHub issue set, I am involved with keeping the driver up to date, and making sure that it builds on older kernels.

Ok thanks, I'm trying to see if I have any response on IRC first

dubhater, 2024-07-25:
This problem may be [fixed](https://github.com/lwfinger/rtw88/commit/4a1ee64475672a9186e10d87ac5e251448f86914) now.

dubhater, 2025-02-06:
Probably fixed in 6.12.

closed 2025-02-06 by dubhater