lwfinger/rtw88 #173: GWTC116-2: Connectivity fails after ~500MB of Download activity.

blackgamma7, 2023-12-08:
I have been using this driver on a GWTC116-2 that has been flashed with Linux Mint. I discovered an issue wherein after around 500-550 MB of downloading, with notable consistency, the connection would be unusable, with apt-get failing to connect to any of the mirrors during an upgrade.

dubhater, 2023-12-08:
You forgot a few details, like which chip you have? What kernel version? What driver version? If I remember correctly, this problem was already fixed, so you probably don't have the latest version.

blackgamma7, 2023-12-08:
Card: rtl_8723du Kernel: 5.15.0.89
though running inxi -Fxz showed the network as "No Device Data found".
I'm not sure how to pull up the exact driver version - I know I installed it some time in august and tried to "sudo make install" with the latest version since then.

blackgamma7, 2023-12-08:
git log shows that it's the latest commit - Oct 24, 2023

dubhater, 2023-12-08:
I guess the bug wasn't fixed after all.

There is another driver which may work better: https://github.com/lwfinger/rtl8723du

blackgamma7, 2024-03-04:
Installed the "obselete" driver and so far, connection has been more reliable.
You mentioned the bug was supposedly fixed, could you tell me when and how that change was made?

lwfinger, 2024-03-04:
Look at 'git log'. That will describe the changes.

closed 2024-03-04 by lwfinger

dubhater, 2024-03-04:
I was probably remembering this: https://lore.kernel.org/linux-wireless/20230210181946.cqnvxiomjv77xnf2@fatal.se/

dubhater, 2024-05-19:
@blackgamma7  If you are still interested, try the latest commits from this repository. I think your problem is [fixed](https://github.com/lwfinger/rtw88/commit/4c77cee823053b789185f04e17547dfc0451d9a6).

blackgamma7, 2024-06-10:
Tried out the update, still had the issue where the netowork would disconnect after <1GB of download.

dubhater, 2024-06-10:
Can you double check what version of the driver is loaded with ``lsmod | grep rtw`` and ``git log``?

blackgamma7, 2024-06-11:
I used repo commit 530df1a431313796351feaf9fd8350eef9fbbc93 (dated june 8th)
```
lsmod | grep rtw
rtw_8723du             16384  0
rtw_usb                24576  1 rtw_8723du
rtw_8723d              61440  1 rtw_8723du
rtw_8723x              24576  1 rtw_8723d
rtw_core              303104  3 rtw_8723d,rtw_usb,rtw_8723x
mac80211             1249280  2 rtw_usb,rtw_core
cfg80211              974848  3 rtw_core,8723du,mac80211
```