lwfinger/rtw88 #175: RTW8821ce no wifi after suspend

mkocagoel, 2023-12-20:
My Blacklist file looks like following:

lwfinger, 2023-12-20:
Problems that happen after suspend, are better handled by the "suspend" scripts. Do the following:
sudo cp suspend_rtw8821ce /usr/lib/systemd/system-sleep/.

That will unload the driver before hibernate or sleep, and reload it on resume. This solution is in README.md.

mkocagoel, 2023-12-20:
sry, forgot to mention. I already have following conf:
/usr/lib/systemd/system-sleep/suspend_rtw8821ce
```
#!/bin/sh
if [ "${1}" == "pre" ]; then
  modprobe -rv rtw_8821ce
elif [ "${1}" == "post" ]; then
  modprobe -v rtw_8821ce
fi
```
(I wrote it without copying, but didnt forget to make it executable)

lwfinger, 2023-12-20:
Perhaps your machine is not settled before the reload happens. Add a line that says "sleep 2" at the beginning of the "post" section.

mkocagoel, 2023-12-20:
Thank you for the fast response, I will try following:
```
#!/bin/sh
if [ "${1}" == "pre" ]; then
  modprobe -rv rtw_8821ce
elif [ "${1}" == "post" ]; then
  sleep 2
  modprobe -v rtw_8821ce
fi
```
i will close it if it works 👍🏻

mkocagoel, 2023-12-20:
> Perhaps your machine is not settled before the reload happens. Add a line that says "sleep 2" at the beginning of the "post" section.
This worked for me, Thank you 👍🏻

mkocagoel, 2024-02-21:
Following error:
```
Feb 21 09:44:21 archlinux kernel: pcieport 0000:00:1c.4: Unable to change power state from D3hot to D0, device inaccessible
Feb 21 09:44:21 archlinux kernel: pci 0000:02:00.0: Unable to change power state from unknown to D0, device inaccessible

```

```
#!/bin/sh
if [ "${1}" == "pre" ]; then
  modprobe -rv rtw_8821ce
elif [ "${1}" == "post" ]; then
  sleep 2
  modprobe -v rtw_8821ce
fi
```
Doesnt work :(

lwfinger, 2024-02-21:
Where did you put that script? Are you running the driver from this repo, or are you using the ones built into the kernel? If the latter, then you need to change the driver names in the script to rtw88_8821ce.

mkocagoel, 2024-02-21:
lspci -v output:
```
02:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8821CE 802.11ac PCIe Wireless Network Adapter
	DeviceName: Realtek RTL8188EE 802.11bgn Wi-Fi Adapter
	Subsystem: Hewlett-Packard Company RTL8821CE 802.11ac PCIe Wireless Network Adapter
	Flags: bus master, fast devsel, latency 0, IRQ 16
	I/O ports at 4000 [size=256]
	Memory at a4100000 (64-bit, non-prefetchable) [size=64K]
	Capabilities: [40] Power Management version 3
	Capabilities: [50] MSI: Enable- Count=1/1 Maskable- 64bit+
	Capabilities: [70] Express Endpoint, MSI 00
	Capabilities: [100] Advanced Error Reporting
	Capabilities: [148] Device Serial Number 00-e0-4c-ff-fe-c8-21-01
	Capabilities: [158] Latency Tolerance Reporting
	Capabilities: [160] L1 PM Substates
	Capabilities: [170] Precision Time Measurement
	Capabilities: [17c] Vendor Specific Information: ID=0003 Rev=1 Len=054 <?>
	Kernel driver in use: rtw_8821ce
	Kernel modules: rtw88_8821ce, rtw_8821ce
```
The Script is in:
/usr/lib/systemd/system-sleep/suspend_rtw8821ce

lwfinger, 2024-02-21:
Is the script executable? Use 'ls -l /usr/lib/systemd/system-sleep/suspend_rtw8821ce' and look for "x" for the owner of the file. If it was not executable, use 'sudo chmod a+x /usr/lib/systemd/system-sleep/suspend_rtw8821ce'. If is was already executable, change the sleep statement to 'sleep 4'. Perhaps the system takes longer to settle down.

mkocagoel, 2024-02-21:
`-rwxr-xr-x 1 root root 135 Feb 18 21:22 /usr/lib/systemd/system-sleep/suspend_rtw8821c`
Gonna try it, thanks

mkocagoel, 2024-02-26:
seems like 'sleep 4' fixed it for now. Ty for fast response

closed 2024-02-26 by mkocagoel