lwfinger/rtw88 #176: Assistance Needed for Realtek 8812CU Chipset

breakds, 2023-12-28:
Hello,

I am reaching out for assistance regarding a Wi-Fi connectivity issue I'm facing with a Realtek 8812CU chipset. Here are the details:

1. Wi-Fi Chipset: I have a Wi-Fi adapter that uses the Realtek 8812CU chipset. This identification is based on the product ID obtained from the system (lsusb output), indicating a Realtek device.

2. Hardware Setup: The chipset is integrated into a custom board, which is based on the Raspberry Pi 4B model.

3. Raspbian Performance: On the original Raspbian OS, the adapter is recognized and managed by the 88x2cu module. It functions smoothly without any notable issues.

4. Custom Linux Environment: I am in the process of creating a custom Linux environment for this hardware. Under this setup, the Wi-Fi adapter is recognized by the rtw88 driver as an rtw8822cu device. However, the network connection is extremely unstable. I'm experiencing latencies exceeding 1000ms, and the dmesg log frequently shows the error rtw8822cu: time out to flush queue 3 every few seconds.

5.Potential Misidentification Concern: I am concerned that the issue may stem from the chipset being misidentified as 8822cu instead of 8812cu, but I am not certain if this is the root cause.

My questions are:

- Is it feasible to use the rtw88 driver to support the 8812CU chipset effectively?
- If not, could you please advise on how to locate and install the appropriate module or firmware for the 8812CU? I am not very familiar with this process and would greatly appreciate any guidance or suggestions.

I sincerely thank you in advance for any assistance or insights you can provide. Your expertise and support would be incredibly helpful in resolving this issue.

Best regards,
Break

lwfinger, 2023-12-28:
The 8812cu uses the same driver as the 8822cu, thus the driver identification is correct.

The USB drivers for rtw88 have been developed outside the Realtek facility where the PCIe drivers are done. They are very much a work in progress. If the rtw88x2cu driver works for you, use it. I suspect it is vendor driver. I have not used that particular driver.

I am about to receive samples of some rtw88xxyu dongles, but none of them are the *cu variety, thus I will not be likely to help your problem.

breakds, 2023-12-28:
> The 8812cu uses the same driver as the 8822cu, thus the driver identification is correct.
>
> The USB drivers for rtw88 have been developed outside the Realtek facility where the PCIe drivers are done. [...]

Thanks a lot for the reply! It is good to know that 8812cu and 8822cu share the same driver. I can try to dig into it more.

closed 2023-12-28 by breakds