lwfinger/rtw88 #177: ETA on a pull up to wireless-next

hjmallon, 2024-03-15:
We are trying to use this driver with RTL8821CU chipset, but it didn't work very well. I can see that some relevant bugfixes have gone into wireless-next. Is there a plan to rebase this driver soon?

lwfinger, 2024-03-15:
This repo has every commit that is in wireless-next. I update this driver whenever new patches are applied to wireless-next.

What kinds of problems are you having with the rtl8821cu? Mine works very well, but there seem to be a wide variation in those chips. One theory is that the EFUCE is not properly encoded on some devices. Please indicate your vendor and product.

hjmallon, 2024-03-15:
Does this repo have this commit? https://git.kernel.org/pub/scm/linux/kernel/git/wireless/wireless-next.git/commit/drivers/net/wireless/realtek/rtw88?id=41a7acb7dde8395f52a707bbba7712a898dfafb0

It doesn't seem to (but it was only added 10days ago)

One problem was that we were getting lots of firmware logging, which I am guessing this commit might fix

wireless-next commits: https://git.kernel.org/pub/scm/linux/kernel/git/wireless/wireless-next.git/log/drivers/net/wireless/realtek/rtw88

lwfinger, 2024-03-15:
That patch adds routine rtw_usb_reg_sec(), which a quick grep shows is present. Furthermore, the 'git log' shows the following:

commit d679f46afc2499705ecd6d21f5a62d2dd098aa88
Author: Bitterblue Smith <rtl8821cerfe2@gmail.com>
Date:   Fri Mar 1 00:32:45 2024 +0200

    wifi: rtw88: 8821cu: Fix firmware upload fail

lwfinger, 2024-03-15:
Have you done a 'git pull'?

hjmallon, 2024-03-15:
I really think I must have some wires crossed sorry. 

1. Go to this link https://github.com/lwfinger/rtw88
2. Most recent commit is listed as 5 months ago
3. Go to usb.c on that webpage
4. It doesn’t have that symbol

lwfinger, 2024-03-15:
My .git/config got messed up and I was pushing to the wrong place. After I fixed that, there was a pull thant I was missing that completely destroyed my recent work! Fortunately, I have another branch that had not yet been pushed.

Run the following: 'git checkout origin/rtw8821au -b rtw8821au'. That will get you the latest rtw88. Meanwhile, I will be working on repairing the master branch.

lwfinger, 2024-03-15:
The master branch should be repaired now.

hjmallon, 2024-03-18:
Thanks that looks much better! I'm just diffing wireless-next and this driver.

1. https://github.com/lwfinger/rtw88/blob/master/rx.c#L188C22-L188C33

Is this a typo? Should it be `rx_status->flag |= RX_FLAG_HT;`?

2. There are some additions of `NL80211_IFTYPE_P2P_CLIENT` / `NL80211_IFTYPE_P2P_GO` in main.c which do not seem to be upstream?

lwfinger, 2024-03-18:
1. Yes, it should be HT, not VHT, but it only applies to kernels older than 4.12 and we could not possibly build for kernels that old.

2. I had a request for P2P functionality, which is why those changes were made. At this point, I have ne idea if it works.

closed 2024-03-18 by hjmallon

hjmallon, 2024-03-18:
Closing as this is completed now, thanks again