lwfinger/rtw88 #179: Build fails with error

draemato, 2024-03-16:
After update build fails with this:

```
/home/user/rtw88/fw.c: In function ‘rtw_get_rsvd_page_probe_req_location’:
/home/user/rtw88/fw.c:1012:21: error: implicit declaration of function ‘cfg80211_ssid_eq’; did you mean ‘cfg80211_bss_iter’? [-Werror=implicit-function-declaration]
 1012 |                     cfg80211_ssid_eq(rsvd_pkt->ssid, ssid))
      |                     ^~~~~~~~~~~~~~~~
      |                     cfg80211_bss_iter
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:243: /home/user/rtw88/fw.o] Error 1
make[2]: *** [/usr/lib/modules/6.7.9-zen1-1-zen/build/Makefile:1909: /home/user/rtw88] Error 2
make[1]: *** [Makefile:234: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.7.9-zen1-1-zen/build'
make: *** [Makefile:126: all] Error 2
```
It never happened before.
arch linux, kernel 6.7.9-zen1-1-zen. headers and all other prerequsites are installed.

istudyatuni, 2024-03-16:
It seems that this function was added in the [last commit](https://github.com/lwfinger/rtw88/commit/6e082ea6f1ed5db7cba1ff301addea49d9c4eab7#diff-8cf158c2c4dc7809e0ab54fa4bca3d50a5915ef6d367a7aafbed7096a40a2358L938-R1012), but doesn't exists (searched for name via github search)

lwfinger, 2024-03-16:
I fixed this error and pushed the code. Thanks for reporting.

closed 2024-03-16 by lwfinger