lwfinger/rtw88 #187: What is this driver doing in different way?

vognev, 2024-04-10:
I have a strange issue with the rk3566 SBC, which is using 8821CS chip for WiFi and Bluetooth.

Mainline drivers combination (`rtw88_8821cs` and `btrtl`), when both loaded as modules, in 99% of boots are ending with probe error:

> rtw_8821cs: probe of mmc3:0001:1 failed with error -110

before which a bunch of sdio readXX errors are printed (exactly a case described here - https://lore.kernel.org/lkml/CA+fyA4SqT4jgpWeq_ipp7i3ZM7s=gQkSdtsPw6x+zeJdGcXvqw@mail.gmail.com/T/)

but `rtw_8821cs`, replacing `rtw88_8821cs` is just loading fine and showing network adapter in system.

I've tried to diff kernel's driver sources with this repo, and not found any significant differences. Made a patch out of that diff, and when it is applied, no difference to driver's behavior too.

The difference is basically in following message only:
> rtw_core: loading out-of-tree module taints kernel.

Is that adding some delay in module initialization so the device is initialized in proper order and sequence?

lwfinger, 2024-04-10:
No, the code is somewhat different. If you used a distribution kernel, then it was 6.8 or older. As the README.md states, the code in this repo now contains the code that will be in kernel 6.10, thus this repo is newer than the distributed kernel.

vognev, 2024-04-10:
> If you used a distribution kernel

self-built 6.8.1 vanilla (fails to probe 99% of boots), and with a patch to realtek/rtw88, made by diffing it and this repo (same results)

> No, the code is somewhat different

But not too much, and perhaps issue is not in code itself, since when diffing it for making aforementioned patch, no significant changes can be spotted (ignoring some preprocessor cases for older kernels), which causing me to think that issue in my case is in driver and somehow related to the timing, which perhaps out-of-tree driver loading is adding

vognev, 2024-04-10:
well, seems like simply adding `msleep(250)` to the beginning of `rtw_core_init()` totally fixing my issue, despite i still neither understand why that hack works, nor why driver from this repo works in it's turn too.

thank you for feedback and some clues and sorry for any wasted time

lwfinger, 2024-04-10:
I do not think you wasted my time. It looks as if you found an important timing isssue.

Does patch1.txt that only is applied to SDIO probe routine fix the problem, or must it be in `rtw_core_init()?`  If so, does patch2.txt fix the problem? I hate to lock up a CPU for 250 ms if it is not necessary. Did you try shorter times?

[patch2.txt](https://github.com/lwfinger/rtw88/files/14933516/patch2.txt)
[patch1.txt](https://github.com/lwfinger/rtw88/files/14933517/patch1.txt)

vognev, 2024-04-10:
1st patch seems working in a bit more than 50% of boots, adapted it (just to be sure i'm testing what i've changed), reduced delay and collected some statistic

```tested.patch
diff --git a/drivers/net/wireless/realtek/rtw88/sdio.c b/drivers/net/wireless/realtek/rtw88/sdio.c
index 0cae5746f..f5e1e9cd7 100644
--- a/drivers/net/wireless/realtek/rtw88/sdio.c
+++ b/drivers/net/wireless/realtek/rtw88/sdio.c
@@ -1325,6 +1325,8 @@ int rtw_sdio_probe(struct sdio_func *sdio_func,
        rtwdev->hci.ops = &rtw_sdio_ops;
        rtwdev->hci.type = RTW_HCI_TYPE_SDIO;
 
+       rtw_info(rtwdev, "Sleeping 150ms in rtw_sdio_probe");
+       msleep(150);
        ret = rtw_core_init(rtwdev);
        if (ret)
                goto err_release_hw;
```

attaching dmesg-es of 10 sample boots: [tested.tgz](https://github.com/lwfinger/rtw88/files/14935058/tested.tgz)

lwfinger, 2024-04-10:
Thanks for all of those dmesg items. It is some kind of interference/timing issue with the Bluetooth device. A UART device should not interfere with an SDIO unit, but they do share a package. Perhaps it is a power issue on that chip.

When the "dw-apb-uart fe650000.serial: failed to request DMA" line is logged AFTER the sleeping entry, wifi fails.

When that DMA line is logged BEFORE the sleeping entry, wifi works.

Failed: 0 1 3 7

[   10.230516] rtw_8821cs mmc3:0001:1: Sleeping 150ms in rtw_sdio_probe
[   10.306569] Bluetooth: HCI UART driver ver 2.3
[   10.306717] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   10.307167] of_dma_request_slave_channel: dma-names property of node '/serial@fe650000' missing or empty
[   10.307199] dw-apb-uart fe650000.serial: failed to request DMA
[   10.543474] rtw_8821cs mmc3:0001:1: Firmware version 24.8.0, H2C version 12
[   10.730744] rtw_8821cs mmc3:0001:1: sdio read32 failed (0x11080): -110

Works: 2 4 5 6 8 9

[   10.453621] dw-apb-uart fe650000.serial: failed to request DMA
[   10.455741] rtw_8821cs mmc3:0001:1: Sleeping 150ms in rtw_sdio_probe
[   10.639186] rtw_8821cs mmc3:0001:1: Firmware version 24.8.0, H2C version 12
[   11.183509] Bluetooth: hci0: RTL: examining hci_ver=08 hci_rev=000c lmp_ver=08 lmp_subver=8821
[   11.187912] Bluetooth: hci0: RTL: rom_version status=0 version=1
[   11.187947] Bluetooth: hci0: RTL: loading rtl_bt/rtl8821cs_fw.bin

Try changing the msleep(150) to mdelay(500). Does that improve the success rate? Any other side effects appear?

vognev, 2024-04-10:
with mdelay(500), got 20 successful boots in the row, and have not noticed anything really bad

lwfinger, 2024-04-10:
Great. I sent E-mail to the linux-wireless mailing list to see if there is a more sophisticated way to ensure that the BT interface is done initializing before the SDIO wifi part tries to get going. If I get no suggestions, I will send a patch that incorporates the mdelay().

lwfinger, 2024-04-11:
I sent in a patch to add the mdelay(500) statement [1] . The SDIO group at Realtek would like to verify that it is the BT causing the problem [2].
They would like you to blacklist the BT driver and remove any delay to see if the SDIO error goes away. I am not certain what the name is - probably hci_uart, but lsmod will tell you the correct name.
[1] https://marc.info/?l=linux-wireless&m=171279881119866&w=2
[2] https://marc.info/?l=linux-wireless&m=171280149320836&w=2

Thanks.

vognev, 2024-04-11:
rmmod/modprobe of rtw88_8821cs later after boot works just fine and wlan device appears
i believe i've already tried blacklisting btrtl/hci_uart earlier, but just to be sure will re-test now

vognev, 2024-04-11:
so, did three different tests: [nobtrtl.tgz](https://github.com/lwfinger/rtw88/files/14941024/nobtrtl.tgz)

boot0..boot2 - vanilla kernel without any patches and `btrtl` being blacklisted, all failed
boot3..boot5 - adding mdelay patch on top of that changing nothing
boot6..boot8 - all three are ok if `btrtl` is builtin module
```patch
diff --git a/device/testing/linux-powkiddy-x55/config-powkiddy-x55.aarch64 b/device/testing/linux-powkiddy-x55/config-powkiddy-x55.aarch64
index a339b68e7..6b29ea9d5 100644
--- a/device/testing/linux-powkiddy-x55/config-powkiddy-x55.aarch64
+++ b/device/testing/linux-powkiddy-x55/config-powkiddy-x55.aarch64
@@ -1,6 +1,6 @@
 #
 # Automatically generated file; DO NOT EDIT.
-# Linux/arm64 6.8.1 Kernel Configuration
+# Linux/arm64 6.8.4 Kernel Configuration
 #
 CONFIG_CC_VERSION_TEXT="aarch64-alpine-linux-musl-gcc (Alpine 13.2.1_git20240309) 13.2.1 20240309"
 CONFIG_CC_IS_GCC=y
@@ -1535,7 +1535,6 @@ CONFIG_BT_BNEP=y
 CONFIG_BT_BNEP_MC_FILTER=y
 CONFIG_BT_BNEP_PROTO_FILTER=y
 CONFIG_BT_HIDP=y
-CONFIG_BT_HS=y
 CONFIG_BT_LE=y
 CONFIG_BT_LE_L2CAP_ECRED=y
 CONFIG_BT_LEDS=y
@@ -1548,10 +1547,10 @@ CONFIG_BT_DEBUGFS=y
 #
 # Bluetooth device drivers
 #
-CONFIG_BT_RTL=m
+CONFIG_BT_RTL=y
 # CONFIG_BT_HCIBTUSB is not set
 # CONFIG_BT_HCIBTSDIO is not set
-CONFIG_BT_HCIUART=m
+CONFIG_BT_HCIUART=y
 CONFIG_BT_HCIUART_SERDEV=y
 # CONFIG_BT_HCIUART_H4 is not set
 # CONFIG_BT_HCIUART_NOKIA is not set
```

if i'm not missing (or not messed) something, looks like bluetooth part also is initializing wifi one

lwfinger, 2024-04-11:
It turns out that I was confused, and that should be an msleep(500), not mdelay(500). The sleep functions do not tie up a CPU as I thought.

One thing that is confusing. It the dmesg files from yesterday, the system found firmware rtl_bt/rtl8821cs_fw.bin, but in boot6 today, the firmware was not found. Having the BT driver builtin rather than as a module should not matter.

I am reporting the fact that wifi fails to initialize correctly if the BT driver is not available no matter the delay/sleep time.

Although having the BT driver builtin works, that is not a satisfactory solution for the user that just expects the distributed system to work. Could you try an msleep(500) with the BT driver as a module? Is that long enough?

lwfinger, 2024-04-11:
Here is the E-mail I sent to the linux-wireless list this morning:
On 4/11/24 2:08 AM, Kalle Valo wrote:
> (dropping stable list from cc)
> It's actually the opposite, msleep() allows other processes to run.
>
> "In general, use of mdelay is discouraged and code should be refactored
> to allow for the use of msleep."
>
> https://docs.kernel.org/timers/timers-howto.html

Kalle and Ping-ke,

Sorry that I got that wrong. I have become just another confused old man. 🙂

The OP at GitHub has tested kernels where he blacklisted the BT drivers. The wifi failed every time (3 tries with a delay and 3 times without). When he built a kernel with BT built-in rather than as a module, it worked (again 3 tries). There definitely is an interaction between the BT initialization and the wifi setup. If the BT part is not started and has not completed the UART initialization, SDIO wifi does not initialize correctly.

When it fails, unloading the wifi driver and reloading it restores wifi.

I originally wondered if there was a power problem when both were simultaneously started, but I see that is not the case. To reiterate, the BT must start first and complete UART initialization before the wifi starts initializing. It seems to me that a suitable msleep() at the start of the SDIO probe routine seems to be a viable workaround even though it is not aesthetically pleasing. A sleep of 150 ms is too short, but 500 seems to work reliably.

Perhaps the SDIO experts at Realtek may be able to offer a more satisfying explanation. The OP at GitHub and I will try to answer any questions they might have. Note that I do not have the hardware, but my contact at GitHub is knowledgeable, and seems eager to help.

Larry

vognev, 2024-04-11:
well, what I have to say, is that msleep/mdelay not making any difference - both are working stable after considerable amount of reboots

> but in boot6 today, the firmware was not found

for that I had to additionally include firmware in initramfs (specific to postmarketOS I believe, it's tooling I'm using to build/deploy kernel).

> the fact that wifi fails to initialize correctly if the BT driver is not available no matter the delay/sleep time

also surprised me now, earlier i thought that it is just some interference or timing issue. some background story behind that here - https://gitlab.com/postmarketOS/pmaports/-/merge_requests/4972/diffs#044064abb8936c8cb0a52de6c3fd92d41f2bb925_3

vognev, 2024-04-11:
> Here is the E-mail I sent to the linux-wireless list this morning

I'm tracking corresponding thread in mailing list

lwfinger, 2024-04-11:
Good. I hope we have given the Realtek engineers enough info that they can provide a suitable explanation. I have no personal usage of any SDIO equipment.

lwfinger, 2024-04-12:
I have not heard back from Realtek, and I am resending the patch. If you send me your real E-mail to Larry.Finger@lwfinger.net, I will add a "Tested-by:" annotation for you. That will expose that mail address to the Internet, thus you may wish to skip that part.

My revised commit message is as follows: Please correct any errors.

Subject: [RFC V2] rtw88: Fix startup problems for SDIO wifi plus UART Bluetooth
To: Kalle Valo <kvalo@kernel.org>
Cc: Johannes Berg <johannes@sipsolutions.net>,
    linux-wireless@vger.kernel.org

On the Powkiddy X55 (Links 1 and 2), the Anbernic RG353V (Link 3), and likely
other similar devices, the SDIO part of RTW8821CS fails to start correctly
with error -110 for every attempt to read or write device registers if the
UART Bluetooth portion of the chip has not reached completion of the routines
that initialize the serial interface. Note that unloading and reloading
rtw88_8821cs will enable wifi except where noted below.

Experimentation has revealed the following behavior:
 1. If the Bluetooth routines are blacklisted, wifi cannot start. In addition,
    unloading and reloading the wifi driver also fails to start wifi.
 2. If the Bluetooth routines are configured to be built into the kernel,
    wifi always starts.
 3. If an msleep(150) is added before the SDIO driver begins iinitializing
    wifi, a success rate of about 60% is obtained.
 4. Using an msleep(500) results in a success rate of 100%.

As it would not be appropriate to force distros to make the BT routines
built into the kernel, the msleep(500) seems appropriate.

Link: https://github.com/lwfinger/rtw88/issues/187
Link: https://gitlab.com/postmarketOS/pmaports/-/merge_requests/4972/diffs#044064abb8936c8cb0a52de6c3fd92d41f2bb925_36_32
Link: https://lore.kernel.org/lkml/CA+fyA4SqT4jgpWeq_ipp7i3ZM7s=gQkSdtsPw6x+zeJdGcXvqw@mail.gmail.com/T/
Fixes: 65371a3f14e7 ("wifi: rtw88: sdio: Add HCI implementation for SDIO based chipsets")

Cc: stable@vger.kernel.org # v6.4+
---

vognev, 2024-04-13:
You can take my contact from Maintainer line (Link 2)

want to correct your point 1

> If the Bluetooth routines are blacklisted, wifi cannot start.

precisely on boot

> In addition,
> unloading and reloading the wifi driver also fails to start wifi.

this part seems not quite true, even if BT driver is blacklisted, and WiFi fails to initialize on boot, later, rmmod/modprobe of 8821cs helps get wlan device working:

(`5000ms msleep` message should be irrelevant, just leftover from testing)

```
[  198.780779] mmc_host mmc1: Bus speed (slot 0) = 375000Hz (slot req 400000Hz, actual 375000HZ div = 0)
[  198.836769] mmc_host mmc1: Bus speed (slot 0) = 50000000Hz (slot req 50000000Hz, actual 50000000HZ div = 0)
[  211.630353] rtw_8821cs mmc3:0001:1: rtw_sdio_probe: 5000ms msleep before rtw_core_init
[  217.067662] rtw_8821cs mmc3:0001:1: Firmware version 24.8.0, H2C version 12
[  217.078311] rtw_8821cs mmc3:0001:1: sdio read32 failed (0x11080): -110
[  217.078474] rtw_8821cs mmc3:0001:1: sdio write32 failed (0x11080): -110
[  223.093188] wlan0: authenticate with f4:f2:6d:5c:97:07 (local address=f4:3c:3b:d8:8a:ec)
[  223.226080] wlan0: send auth to f4:f2:6d:5c:97:07 (try 1/3)
[  223.229058] wlan0: authenticated
[  223.229995] wlan0: associate with f4:f2:6d:5c:97:07 (try 1/3)
[  223.237085] wlan0: RX AssocResp from f4:f2:6d:5c:97:07 (capab=0x401 status=0 aid=3)
[  223.243631] wlan0: associated
```

[dmesg_no_hci_uart_reinsert.txt](https://github.com/lwfinger/rtw88/files/14965168/dmesg_no_hci_uart_reinsert.txt)

so I really not sure about real cause of this issue, it also can be related to bad/not optimal DTB maybe, as mmc host doing a lot of bus speed training? before wifi is stable (that is traced to some power management routines imo), and this patch is just lucky to defer wifi initialization to proper point in time.
Only wild guess, I'm totally noob in this area.

vognev, 2024-04-14:
seems figured it out, and that is only coincidence that rtw88 from this repo was loaded just in right time to initialize WiFi. Small DTB patch fixed issue for me: https://github.com/ROCKNIX/distribution/pull/63/files#diff-e2de6222b1794f89311bdc1597c1597c76f34503c575ea3f7e7ec9c5376218d6

sorry for wasted time spent on debugging it in wrong direction

closed 2024-04-14 by vognev

lwfinger, 2024-04-14:
As not all users of the Rockchip are likely to be aware of the DTB patches, I am going to submit the patch anyway. A 1/2 second sleep is not a big deal and it covers most of the cases.