lwfinger/rtw88 #188: make error

noticu, 2024-04-11:
/home/icu/rtw88/rtw8703b.c: In function ‘try_mac_from_devicetree’:
/home/icu/rtw88/rtw8703b.c:546:9: error: too many arguments to function ‘of_get_mac_address’
  546 |   ret = of_get_mac_address(node, efuse->addr);
      |         ^~~~~~~~~~~~~~~~~~
In file included from /home/icu/rtw88/rtw8703b.c:4:
/usr/src/linux-headers-5.10.0-27-common/include/linux/of_net.h:25:27: note: declared here
   25 | static inline const void *of_get_mac_address(struct device_node *np)
      |                           ^~~~~~~~~~~~~~~~~~
/home/icu/rtw88/rtw8703b.c:546:7: warning: assignment to ‘int’ from ‘const void *’ makes integer from pointer without a cast [-Wint-conversion]
  546 |   ret = of_get_mac_address(node, efuse->addr);
      |       ^
make[3]: *** [/usr/src/linux-headers-5.10.0-27-common/scripts/Makefile.build:291: /home/icu/rtw88/rtw8703b.o] Error 1
make[2]: *** [/usr/src/linux-headers-5.10.0-27-common/Makefile:1856: /home/icu/rtw88] Error 2
make[1]: *** [/usr/src/linux-headers-5.10.0-27-common/Makefile:192: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.10.0-27-amd64'
make: *** [Makefile:150: all] Error 2

fffonion, 2024-04-14:
on my 5.10.160-legacy-rk35xx I did:

```patch
diff --git a/rtw8703b.c b/rtw8703b.c
index 867e632..0e96b34 100644
--- a/rtw8703b.c
+++ b/rtw8703b.c
@@ -540,11 +540,13 @@ static void try_mac_from_devicetree(struct rtw_dev *rtwdev)
 {
        struct device_node *node = rtwdev->dev->of_node;
        struct rtw_efuse *efuse = &rtwdev->efuse;
-       int ret;
+       const void *ret;

        if (node) {
-               ret = of_get_mac_address(node, efuse->addr);
-               if (ret == 0) {
+               ret = of_get_mac_address(node);
+               if (!IS_ERR(ret))
+                       ether_addr_copy(efuse->addr, ret);
+               else {
                        rtw_dbg(rtwdev, RTW_DBG_EFUSE,
                                "got wifi mac address from DT: %pM\n",
                                efuse->addr);
```

lwfinger, 2024-04-14:
Sorry that I did not get to your problem earlier.

I used your change to fix the code to build on versions of the kernel older than 5.13.

I think you have a small error in your patch. The rtw_dbg() call should be reached if the code gets a valid address, not if the address is invalid.

Do a 'git checkout -f' to clear your changes, and a 'git pull' to get mine, and you should be OK.

closed 2024-07-25 by dubhater