lwfinger/rtw88 #190: make error

WeSourceCode, 2024-04-19:
make -C /lib/modules/5.4.0-150-generic/build M=/opt/rtw88 modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-150-generic'
  CC [M]  /opt/rtw88/rtw8703b.o
In file included from /opt/rtw88/rtw8703b.c:4:0:
./include/linux/of_net.h:17:42: warning: ‘struct device_node’ declared inside parameter list will not be visible outside of this definition or declaration
 static inline int of_get_phy_mode(struct device_node *np)
                                          ^~~~~~~~~~~
./include/linux/of_net.h: In function ‘of_get_phy_mode’:
./include/linux/of_net.h:19:10: error: ‘ENODEV’ undeclared (first use in this function)
  return -ENODEV;
          ^~~~~~
./include/linux/of_net.h:19:10: note: each undeclared identifier is reported only once for each function it appears in
./include/linux/of_net.h: At top level:
./include/linux/of_net.h:22:53: warning: ‘struct device_node’ declared inside parameter list will not be visible outside of this definition or declaration
 static inline const void *of_get_mac_address(struct device_node *np)
                                                     ^~~~~~~~~~~
./include/linux/of_net.h: In function ‘of_get_mac_address’:
./include/linux/of_net.h:24:9: error: implicit declaration of function ‘ERR_PTR’ [-Werror=implicit-function-declaration]
  return ERR_PTR(-ENODEV);
         ^~~~~~~
./include/linux/of_net.h:24:18: error: ‘ENODEV’ undeclared (first use in this function)
  return ERR_PTR(-ENODEV);
                  ^~~~~~
./include/linux/of_net.h: At top level:
./include/linux/of_net.h:27:68: warning: ‘struct device_node’ declared inside parameter list will not be visible outside of this definition or declaration
 static inline struct net_device *of_find_net_device_by_node(struct device_node *np)
                                                                    ^~~~~~~~~~~
./include/linux/of_net.h: In function ‘of_find_net_device_by_node’:
./include/linux/of_net.h:29:9: error: ‘NULL’ undeclared (first use in this function)
  return NULL;
         ^~~~
In file included from ./arch/x86/include/asm/processor.h:32:0,
                 from ./arch/x86/include/asm/cpufeature.h:5,
                 from ./arch/x86/include/asm/thread_info.h:53,
                 from ./include/linux/thread_info.h:39,
                 from ./arch/x86/include/asm/preempt.h:7,
                 from ./include/linux/preempt.h:78,
                 from ./include/linux/spinlock.h:51,
                 from ./include/linux/seqlock.h:36,
                 from ./include/linux/time.h:6,
                 from ./include/linux/skbuff.h:15,
                 from ./include/linux/if_ether.h:19,
                 from ./include/net/mac80211.h:18,
                 from /opt/rtw88/main.h:12,
                 from /opt/rtw88/rtw8703b.c:5:
./include/linux/err.h: At top level:
./include/linux/err.h:24:35: error: conflicting types for ‘ERR_PTR’
 static inline void * __must_check ERR_PTR(long error)
                                   ^~~~~~~
In file included from /opt/rtw88/rtw8703b.c:4:0:
./include/linux/of_net.h:24:9: note: previous implicit declaration of ‘ERR_PTR’ was here
  return ERR_PTR(-ENODEV);
         ^~~~~~~
/opt/rtw88/rtw8703b.c: In function ‘try_mac_from_devicetree’:
/opt/rtw88/rtw8703b.c:554:28: error: passing argument 1 of ‘of_get_mac_address’ from incompatible pointer type [-Werror=incompatible-pointer-types]
   ret = of_get_mac_address(node);
                            ^~~~
In file included from /opt/rtw88/rtw8703b.c:4:0:
./include/linux/of_net.h:22:27: note: expected ‘struct device_node *’ but argument is of type ‘struct device_node *’
 static inline const void *of_get_mac_address(struct device_node *np)
                           ^~~~~~~~~~~~~~~~~~
./include/linux/of_net.h: In function ‘of_get_mac_address’:
./include/linux/of_net.h:25:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
cc1: some warnings being treated as errors
scripts/Makefile.build:270: recipe for target '/opt/rtw88/rtw8703b.o' failed
make[2]: *** [/opt/rtw88/rtw8703b.o] Error 1
Makefile:1767: recipe for target '/opt/rtw88' failed
make[1]: *** [/opt/rtw88] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-150-generic'
Makefile:154: recipe for target 'all' failed
make: *** [all] Error 2

dubhater, 2024-07-25:
Is it still broken?

a5a5aa555oo, 2024-08-10:
I did a compile test and no error/warning apears.

OS: Debian 11.10
kernel: 5.4.0-192-generic (Manually downloaded and installed from Ubuntu's repo)

<details>
<summary>build log</summary>
<pre><code>
$ make
make -j`nproc` -C /lib/modules/5.4.0-192-generic/build M=$PWD modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-192-generic'
  CC [M]  /tmp/rtw88/main.o
  CC [M]  /tmp/rtw88/mac80211.o
  CC [M]  /tmp/rtw88/util.o
  CC [M]  /tmp/rtw88/debug.o
  CC [M]  /tmp/rtw88/tx.o
  CC [M]  /tmp/rtw88/rx.o
  CC [M]  /tmp/rtw88/mac.o
  CC [M]  /tmp/rtw88/phy.o
  CC [M]  /tmp/rtw88/coex.o
  CC [M]  /tmp/rtw88/efuse.o
  CC [M]  /tmp/rtw88/fw.o
  CC [M]  /tmp/rtw88/ps.o
  CC [M]  /tmp/rtw88/sec.o
  CC [M]  /tmp/rtw88/bf.o
  CC [M]  /tmp/rtw88/regd.o
  CC [M]  /tmp/rtw88/sar.o
  CC [M]  /tmp/rtw88/wow.o
  CC [M]  /tmp/rtw88/rtw8703b.o
  CC [M]  /tmp/rtw88/rtw8703b_tables.o
  CC [M]  /tmp/rtw88/rtw8723cs.o
  CC [M]  /tmp/rtw88/rtw8723d.o
  CC [M]  /tmp/rtw88/rtw8723d_table.o
  CC [M]  /tmp/rtw88/rtw8723de.o
  CC [M]  /tmp/rtw88/rtw8723ds.o
  CC [M]  /tmp/rtw88/rtw8723du.o
  CC [M]  /tmp/rtw88/rtw8723x.o
  CC [M]  /tmp/rtw88/rtw8812au.o
  CC [M]  /tmp/rtw88/rtw8821a.o
  CC [M]  /tmp/rtw88/rtw8821a_table.o
  CC [M]  /tmp/rtw88/rtw8812a_table.o
  CC [M]  /tmp/rtw88/rtw8821au.o
  CC [M]  /tmp/rtw88/rtw8821c.o
  CC [M]  /tmp/rtw88/rtw8821c_table.o
  CC [M]  /tmp/rtw88/rtw8821ce.o
  CC [M]  /tmp/rtw88/rtw8821cs.o
  CC [M]  /tmp/rtw88/rtw8821cu.o
  CC [M]  /tmp/rtw88/rtw8822b.o
  CC [M]  /tmp/rtw88/rtw8822b_table.o
  CC [M]  /tmp/rtw88/rtw8822be.o
  CC [M]  /tmp/rtw88/rtw8822bs.o
  CC [M]  /tmp/rtw88/rtw8822bu.o
  CC [M]  /tmp/rtw88/rtw8822c.o
  CC [M]  /tmp/rtw88/rtw8822c_table.o
  CC [M]  /tmp/rtw88/rtw8822ce.o
  CC [M]  /tmp/rtw88/rtw8822cs.o
  CC [M]  /tmp/rtw88/rtw8822cu.o
  CC [M]  /tmp/rtw88/pci.o
  CC [M]  /tmp/rtw88/sdio.o
  CC [M]  /tmp/rtw88/usb.o
  LD [M]  /tmp/rtw88/rtw_core.o
  LD [M]  /tmp/rtw88/rtw_8703b.o
  LD [M]  /tmp/rtw88/rtw_8723cs.o
  LD [M]  /tmp/rtw88/rtw_8723d.o
  LD [M]  /tmp/rtw88/rtw_8723de.o
  LD [M]  /tmp/rtw88/rtw_8723ds.o
  LD [M]  /tmp/rtw88/rtw_8723du.o
  LD [M]  /tmp/rtw88/rtw_8723x.o
  LD [M]  /tmp/rtw88/rtw_8812au.o
  LD [M]  /tmp/rtw88/rtw_8821a.o
  LD [M]  /tmp/rtw88/rtw_8821au.o
  LD [M]  /tmp/rtw88/rtw_8821c.o
  LD [M]  /tmp/rtw88/rtw_8821ce.o
  LD [M]  /tmp/rtw88/rtw_8821cs.o
  LD [M]  /tmp/rtw88/rtw_8821cu.o
  LD [M]  /tmp/rtw88/rtw_8822b.o
  LD [M]  /tmp/rtw88/rtw_8822be.o
  LD [M]  /tmp/rtw88/rtw_8822bs.o
  LD [M]  /tmp/rtw88/rtw_8822bu.o
  LD [M]  /tmp/rtw88/rtw_8822c.o
  LD [M]  /tmp/rtw88/rtw_8822ce.o
  LD [M]  /tmp/rtw88/rtw_8822cs.o
  LD [M]  /tmp/rtw88/rtw_8822cu.o
  LD [M]  /tmp/rtw88/rtw_pci.o
  LD [M]  /tmp/rtw88/rtw_sdio.o
  LD [M]  /tmp/rtw88/rtw_usb.o
  Building modules, stage 2.
  MODPOST 26 modules
  CC [M]  /tmp/rtw88/rtw_8703b.mod.o
  CC [M]  /tmp/rtw88/rtw_8723cs.mod.o
  CC [M]  /tmp/rtw88/rtw_8723d.mod.o
  CC [M]  /tmp/rtw88/rtw_8723de.mod.o
  CC [M]  /tmp/rtw88/rtw_8723ds.mod.o
  CC [M]  /tmp/rtw88/rtw_8723du.mod.o
  CC [M]  /tmp/rtw88/rtw_8723x.mod.o
  CC [M]  /tmp/rtw88/rtw_8812au.mod.o
  CC [M]  /tmp/rtw88/rtw_8821a.mod.o
  CC [M]  /tmp/rtw88/rtw_8821au.mod.o
  CC [M]  /tmp/rtw88/rtw_8821c.mod.o
  CC [M]  /tmp/rtw88/rtw_8821ce.mod.o
  CC [M]  /tmp/rtw88/rtw_8821cs.mod.o
  CC [M]  /tmp/rtw88/rtw_8821cu.mod.o
  CC [M]  /tmp/rtw88/rtw_8822b.mod.o
  CC [M]  /tmp/rtw88/rtw_8822be.mod.o
  CC [M]  /tmp/rtw88/rtw_8822bs.mod.o
  CC [M]  /tmp/rtw88/rtw_8822bu.mod.o
  CC [M]  /tmp/rtw88/rtw_8822c.mod.o
  CC [M]  /tmp/rtw88/rtw_8822ce.mod.o
  CC [M]  /tmp/rtw88/rtw_8822cs.mod.o
  CC [M]  /tmp/rtw88/rtw_8822cu.mod.o
  CC [M]  /tmp/rtw88/rtw_core.mod.o
  CC [M]  /tmp/rtw88/rtw_pci.mod.o
  CC [M]  /tmp/rtw88/rtw_sdio.mod.o
  CC [M]  /tmp/rtw88/rtw_usb.mod.o
  LD [M]  /tmp/rtw88/rtw_8703b.ko
  LD [M]  /tmp/rtw88/rtw_8723cs.ko
  LD [M]  /tmp/rtw88/rtw_8723d.ko
  LD [M]  /tmp/rtw88/rtw_8723de.ko
  LD [M]  /tmp/rtw88/rtw_8723ds.ko
  LD [M]  /tmp/rtw88/rtw_8723du.ko
  LD [M]  /tmp/rtw88/rtw_8723x.ko
  LD [M]  /tmp/rtw88/rtw_8812au.ko
  LD [M]  /tmp/rtw88/rtw_8821a.ko
  LD [M]  /tmp/rtw88/rtw_8821au.ko
  LD [M]  /tmp/rtw88/rtw_8821c.ko
  LD [M]  /tmp/rtw88/rtw_8821ce.ko
  LD [M]  /tmp/rtw88/rtw_8821cs.ko
  LD [M]  /tmp/rtw88/rtw_8821cu.ko
  LD [M]  /tmp/rtw88/rtw_8822b.ko
  LD [M]  /tmp/rtw88/rtw_8822be.ko
  LD [M]  /tmp/rtw88/rtw_8822bs.ko
  LD [M]  /tmp/rtw88/rtw_8822bu.ko
  LD [M]  /tmp/rtw88/rtw_8822c.ko
  LD [M]  /tmp/rtw88/rtw_8822ce.ko
  LD [M]  /tmp/rtw88/rtw_8822cs.ko
  LD [M]  /tmp/rtw88/rtw_8822cu.ko
  LD [M]  /tmp/rtw88/rtw_core.ko
  LD [M]  /tmp/rtw88/rtw_pci.ko
  LD [M]  /tmp/rtw88/rtw_sdio.ko
  LD [M]  /tmp/rtw88/rtw_usb.ko
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-192-generic'
$ uname -r
5.4.0-192-generic
</code></pre>
</details>