lwfinger/rtw88 #198: [rtw_8822ce] frequent disconnections

interbattles, 2024-06-01:
### over my past 3 linux installations i've used both:
- rtw_8822ce (this, built locally)
- rtw88_8822ce (built in, still unsure what the difference is)
and experienced frequent to very occasional disconnections and general instability

#### ive switched between wpa_supplicant and iwd, changed rtw88's options, etc.

through some messing around with options the only difference i've noticed is that when disable_msi is used, the interface is online yet never detects any wifi networks, my options are as follows

### - no networks being detected:
  - rtw_pci: `support_bf=Y disable_aspm=Y, disable_msi=Y`
  - rtw_core: `support_bf=Y, disable_lps_deep=Y, debug_mask=0`
  
### - networks working (but being disconnected frequently):
  - rtw_pci: `support_bf=Y disable_aspm=Y, disable_msi=N`
  - rtw_core: `support_bf=Y, disable_lps_deep=Y, debug_mask=0`

#### at different points i switched kernel versions between
- `6.8.5-301.fc40`
- `6.8.10-300.fc40`
- `6.8.11-300.fc40`
- whatever linux mint was using at the time (iirc 5.x)

using fedora40 workstation kde, linux mint, etc

### interesting logs i got in disconnects
```
[  456.737481] wlp2s0: Connection to AP 10:da:43:96:cd:1d lost
[  458.102204] wlp2s0: 80 MHz not supported, disabling VHT
[  458.304693] wlp2s0: authenticate with 10:da:43:96:cd:1d (local address=c8:94:02:05:c4:a9)
[  458.367851] wlp2s0: send auth to 10:da:43:96:cd:1d (try 1/3)
[  458.375293] wlp2s0: authenticated
[  458.375666] wlp2s0: associate with 10:da:43:96:cd:1d (try 1/3)
[  458.381633] wlp2s0: RX AssocResp from 10:da:43:96:cd:1d (capab=0x1411 status=0 aid=12)
[  458.382018] wlp2s0: associated
[  458.458488] wlp2s0: Limiting TX power to 30 (30 - 0) dBm as advertised by 10:da:43:96:cd:1d
[  475.619923] wlp2s0: Connection to AP 10:da:43:96:cd:1d lost
[  477.020238] wlp2s0: 80 MHz not supported, disabling VHT
[  477.221723] wlp2s0: authenticate with 10:da:43:96:cd:1d (local address=c8:94:02:05:c4:a9)
[  477.284976] wlp2s0: send auth to 10:da:43:96:cd:1d (try 1/3)
[  477.292366] wlp2s0: authenticated
[  477.292871] wlp2s0: associate with 10:da:43:96:cd:1d (try 1/3)
[  477.299281] wlp2s0: RX AssocResp from 10:da:43:96:cd:1d (capab=0x1411 status=0 aid=12)
[  477.299665] wlp2s0: associated
[  477.302567] wlp2s0: Limiting TX power to 30 (30 - 0) dBm as advertised by 10:da:43:96:cd:1d
```

```
[    4.490323] NET: Registered PF_QIPCRTR protocol family
[    4.728234] alg: No test for hmac(md4) (hmac(md4-generic))
[    4.756452] Generic FE-GE Realtek PHY r8169-0-100:00: attached PHY driver (mii_bus:phy_addr=r8169-0-100:00, irq=MAC)
[    4.931686] r8169 0000:01:00.0 enp1s0: Link is Down
[    5.306429] usb 1-9: set resolution quirk: cval->res = 384
[    5.306699] usbcore: registered new interface driver snd-usb-audio
[    6.208052] wlp2s0: 80 MHz not supported, disabling VHT
[    6.411522] wlp2s0: authenticate with 10:da:43:96:cd:1d (local address=c8:94:02:05:c4:a9)
[    6.584173] wlp2s0: send auth to 10:da:43:96:cd:1d (try 1/3)
[    6.587522] wlp2s0: authenticated
[    6.589564] wlp2s0: associate with 10:da:43:96:cd:1d (try 1/3)
[    6.594139] wlp2s0: RX AssocResp from 10:da:43:96:cd:1d (capab=0x1411 status=0 aid=12)
[    6.594528] wlp2s0: associated
[    6.669503] wlp2s0: Limiting TX power to 30 (30 - 0) dBm as advertised by 10:da:43:96:cd:1d
[   12.051662] rfkill: input handler disabled
```

#### if its at all important, my kernel params are here
`ro rootflags=subvol=root noapic rhgb quiet`
(note: for some odd reason i have to use noapic otherwise my boot screen will just be stuck on the HP logo)

ps: originally this was a windows desktop that i wiped to install linux, ive heard its an issue with fastboot which i probably originally had on before wiping my ssd

### if you need any extra info i'm readily available in order to try and solve this

interbattles, 2024-06-01:
hope to have this solved as soon as possible :smiley_cat:
i am super grateful for this repo, thanks in advance to maintainers :>

interbattles, 2024-06-29:
still having this ;/

dubhater, 2024-06-29:
RTL8822CE has a good chance of being fixed, if you report the problem in the right place. That would be linux-wireless [at] vger.kernel.org. Make sure you send only plain text emails. The list rejects HTML emails. The title you used here is good. Also include the output of ``iw dev wlp2s0 station dump`` in your email.

interbattles, 2024-07-01:
### after trying many solutions yesterday it seems that ive stopped getting disconnections :grin: 
i dont know which specifically did it but i *think* it's because while the power saving mode for the network adapter was off,  **i turned off power saving mode for the PCIe using TLP**
since this has been a major issue for me for months now i would like to give an extensive **list of all the things ive tried incase it helps anyone who is just as stuck as me**, hopefully somebody can fix their own aswell:

#### in tlp, the things i added were:
```ini
WIFI_PWR_ON_AC=off
WIFI_PWR_ON_BAT=off
```
and 
```ini
RUNTIME_PM_DISABLE="02:00.0" #(the network controller pcie)
```
#### the relevant kernel parameters that i'm using are `pcie_aspm=off` `pci=noacpi` and `iommu=soft` (i think these two are from my cpu because originally it only listed one core)

i do not have power_save set in network manager anymore but its defaulted to off anyway

#### another notable thing that i did was update the development package from the aur (i am using rtw88-dkms-git)  AND blacklisted additional modules to the following

```diff
-blacklist rtw88_8723de
 blacklist rtw88_8723d
-blacklist rtw88_8821ce
+blacklist rtw88_8723de
+blacklist rtw88_8723ds
+blacklist rtw88_8723du
 blacklist rtw88_8821c
-blacklist rtw88_8822be
+blacklist rtw88_8821ce
+blacklist rtw88_8821cs
+blacklist rtw88_8821cu
 blacklist rtw88_8822b
-blacklist rtw88_8822ce
+blacklist rtw88_8822be
+blacklist rtw88_8822bs
+blacklist rtw88_8822bu
 blacklist rtw88_8822c
+blacklist rtw88_8822ce
+blacklist rtw88_8822cs
+blacklist rtw88_8822cu
 blacklist rtw88_core
 blacklist rtw88_pci
+blacklist rtw88_sdio
+blacklist rtw88_usb
```

#### here is my personal modprobe config
```ini
blacklist rtw88_8821ce rtw_8821ce rtw88_core rtw88_pci

options rtw_pci disable_aspm=1
#options rtw88_pci disable_aspm=1 disable_msi=y 

options rtw_core disable_lps_deep=y 
#options rtw88_core disable_lps_deep=y 
# options rtw88_pci disable_msi=y disable_aspm=y
```

---

### i will KEEP THIS OPEN and continue to tinker with the solutions i tried to see which solution in particular was the one that worked
- **note**: i am a complete noob and have basically no idea what i'm doing so alot of this stuff probably sounds dumb 
- **another note**: those who arent aware (as i dont think anyone has mentioned it here) the legend himself [has passed away at 84](https://arstechnica.com/gadgets/2024/06/larry-finger-linux-wireless-hero-was-a-persistent-patient-coder-and-mentor/) , still dedicated to writing software to help this community, thank you for your contributions  :pray:

io-qar, 2024-07-22:
Hey, I also have RTW8822CE and had experienced disconnections and even hang of the system (I tried Ubuntu and Fedora). Then I find out everything is running smoothly with these options set in the configuration file:
```conf
options rtw_pci disable_aspm=y disable_msi=y
options rtw_core disable_lps_deep=y
```

and the blacklist configuration:
```conf
blacklist rtw88_8822ce
blacklist rtw88_pci
blacklist rtw88_core
```

dubhater, 2024-07-22:
If your system hangs and disable_aspm=y makes it stop, you should report this to the linux-wireless@vger.kernel.org mailing list. The driver can be modified so you don't have to use the module parameter anymore.

dubhater, 2025-07-11:
```
iw dev wlan0 set power_save off
```

Does that help?

DamianBossPL, 2025-07-11:
Follow up: I've tested and it doesn't work. My connection still halts.

DamianBossPL, 2025-07-11:
> Then please attach the output of dmesg. Assuming you didn't reboot since this most recent connection problem.

I can only show the relevant parts of it since I've seen that posting all of it is not too safe. Here:
```
[    3.251838] rtw_8822ce 0000:02:00.0: enabling device (0000 -> 0003)
[    3.253937] rtw_8822ce 0000:02:00.0: Firmware version 9.9.15, H2C version 15
[    3.254328] rtw_8822ce 0000:02:00.0: WOW Firmware version 9.9.4, H2C version 15
[    3.331599] rtw_8822ce 0000:02:00.0 wlo1: renamed from wlan0
[    9.663229] wlo1: authenticate with 18:35:d1:77:48:a7 (local address=cc:6b:1e:32:45:01)
[    9.880214] wlo1: send auth to 18:35:d1:77:48:a7 (try 1/3)
[    9.884439] wlo1: authenticated
[    9.885674] wlo1: associate with 18:35:d1:77:48:a7 (try 1/3)
[    9.888755] wlo1: RX AssocResp from 18:35:d1:77:48:a7 (capab=0x1511 status=0 aid=4)
[    9.890467] wlo1: associated
[    9.989993] wlo1: Limiting TX power to 23 (23 - 0) dBm as advertised by 18:35:d1:77:48:a7
[ 1395.729948] wlo1: deauthenticating from 18:35:d1:77:48:a7 by local choice (Reason: 3=DEAUTH_LEAVING)
[ 1396.589177] wlo1: authenticate with 18:35:d1:77:48:a7 (local address=cc:6b:1e:32:45:01)
[ 1396.672586] wlo1: send auth to 18:35:d1:77:48:a7 (try 1/3)
[ 1396.676394] wlo1: authenticated
[ 1396.676861] wlo1: associate with 18:35:d1:77:48:a7 (try 1/3)
[ 1396.680303] wlo1: RX AssocResp from 18:35:d1:77:48:a7 (capab=0x1511 status=0 aid=4)
[ 1396.681430] wlo1: associated
[ 1396.734227] wlo1: Limiting TX power to 23 (23 - 0) dBm as advertised by 18:35:d1:77:48:a7
[ 1601.454031] wlo1: authenticate with 18:35:d1:77:48:a7 (local address=cc:6b:1e:32:45:01)
[ 1601.538413] wlo1: send auth to 18:35:d1:77:48:a7 (try 1/3)
[ 1601.649237] wlo1: send auth to 18:35:d1:77:48:a7 (try 2/3)
[ 1601.653019] wlo1: authenticated
[ 1601.653806] wlo1: associate with 18:35:d1:77:48:a7 (try 1/3)
[ 1601.656858] wlo1: RX AssocResp from 18:35:d1:77:48:a7 (capab=0x1511 status=0 aid=4)
[ 1601.657823] wlo1: associated
[ 1601.744478] wlo1: Limiting TX power to 23 (23 - 0) dBm as advertised by 18:35:d1:77:48:a7
```
Between `1396.734227` and `1601.454031` is where the connection halts. Dmesg did not have anything logged relating to the connection halted. It was simply silent.

Also, if it helps, here are the modprobe options I tried to put. Either way it would still lead to this issue.
```
options rtw_pci disable_aspm=y disable_msi=y
options rtw_core disable_lps_deep=y

options rtw88_pci disable_aspm=y disable_msi=y
options rtw88_core disable_lps_deep=y
```

DamianBossPL, 2025-07-11:
> Try to run `iw event -T`, maybe you will see something useful. It may not print anything at first, just leave it running until the connection dies.

Alright, I ran it and this is what it gave me right to the disconnection:
```
[2025-07-11 19:03:28.377436]: wlo1: new station 02:51:1c:77:e6:06
[2025-07-11 19:03:29.028458]: wlo1 (phy #0): CQM event: RSSI (-63 dBm) went above threshold
[2025-07-11 19:04:50.584137]: wlo1: del station 02:51:1c:77:e6:06
```
The first two lines we're from a moonlight client connecting to my sunshine service. AND when my system tried to send a request out to YouTube, the third line shows up and so my connection halts.

DamianBossPL, 2025-07-11:
> Alright, I ran it and this is what it gave me right to the disconnection:
>
> ```
> [2025-07-11 19:03:28.377436]: wlo1: new station 02:51:1c:77:e6:06
> [2025-07-11 19:03:29.028458]: wlo1 (phy #0): CQM [...]

Apparantly that 'station' was my phone's network MAC, which I was using to make the connection die using moonlight. Instead, I'll try and do web browsing until my connection dies. Give me some minutes.

DamianBossPL, 2025-07-11:
> Apparantly that 'station' was my phone's network MAC, which I was using to make the connection die using moonlight. Instead, I'll try and do web browsing until my connection dies. Give me some [...]

I experimented and it is legit because of my phone. I believe it is due to KDE Connect actually, because as far as I know, the only way my phone communicates automatically with my laptop is via that, and I could halt the connection manually using moonlight+sunshine. And from looking at `iw dev wlo1 station dump`, each of those establish a peer-to-peer connection with my laptop. When my phone disconnects, my laptop deletes the station for my phone's MAC and therefore I somehow lose connection to internet until I reconnect to my AP.
```
[2025-07-11 19:50:59.634017]: wlo1: new station 02:51:1c:77:e6:06
[2025-07-11 19:51:06.790367]: wlo1: del station 02:51:1c:77:e6:06
```

DamianBossPL, 2025-07-11:
Here's some more logs that may help. I done some commands before and after the station got deleted, separated with the clear command:
```
[damian@uplink] ~  
❯ iw dev wlo1 station dump
Station 18:35:d1:77:48:a7 (on wlo1)
	inactive time:	3300 ms
	rx bytes:	654002
	rx packets:	3533
	tx bytes:	588916
	tx packets:	1464
	tx retries:	0
	tx failed:	0
	beacon loss:	0
	rx drop misc:	3
	signal:  	-62 [-62, -64] dBm
	signal avg:	-61 [-61, -64] dBm
	tx bitrate:	526.6 MBit/s VHT-MCS 6 80MHz VHT-NSS 2
	tx duration:	0 us
	rx bitrate:	351.0 MBit/s VHT-MCS 4 80MHz VHT-NSS 2
	rx duration:	0 us
	authorized:	yes
	authenticated:	yes
	associated:	yes
	preamble:	long
	WMM/WME:	yes
	MFP:		no
	TDLS peer:	no
	DTIM period:	1
	beacon interval:100
	short slot time:yes
	connected time:	206 seconds
	associated at [boottime]:	8723.567s
	associated at:	1752261006680 ms
	current time:	1752261212574 ms
Station 02:51:1c:77:e6:06 (on wlo1)
	inactive time:	1432 ms
	rx bytes:	162101
	rx packets:	1189
	tx bytes:	17635454
	tx packets:	16574
	tx retries:	0
	tx failed:	0
	beacon loss:	0
	rx drop misc:	0
	signal:  	-47 [-52, -47] dBm
	signal avg:	-44 [-49, -44] dBm
	tx bitrate:	526.6 MBit/s VHT-MCS 6 80MHz VHT-NSS 2
	tx duration:	0 us
	rx bitrate:	104.0 MBit/s VHT-MCS 5 VHT-NSS 2
	rx duration:	0 us
	authorized:	yes
	authenticated:	yes
	associated:	yes
	preamble:	long
	WMM/WME:	yes
	MFP:		no
	TDLS peer:	yes
	DTIM period:	1
	beacon interval:100
	short slot time:yes
	connected time:	56 seconds
	associated at [boottime]:	8873.431s
	associated at:	1752261156544 ms
	current time:	1752261212574 ms

[damian@uplink] ~  
❯ iw dev wlo1 info        
Interface wlo1
	ifindex 2
	wdev 0x1
	addr cc:6b:1e:32:45:01
	ssid VM0883623-5GHz
	type managed
	wiphy 0
	channel 44 (5220 MHz), width: 80 MHz, center1: 5210 MHz
	txpower 20.00 dBm
	multicast TXQ:
		qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
		0	0	0	0	0	0	0	0		0

[damian@uplink] ~  
❯ clear

[damian@uplink] ~  
❯ iw dev wlo1 station dump                                                                                                                                                        ⏎
Station 18:35:d1:77:48:a7 (on wlo1)
	inactive time:	201 ms
	rx bytes:	2513117
	rx packets:	5104
	tx bytes:	764215
	tx packets:	1808
	tx retries:	0
	tx failed:	0
	beacon loss:	0
	rx drop misc:	4
	signal:  	-64 [-64, -65] dBm
	signal avg:	-63 [-64, -65] dBm
	tx bitrate:	175.5 MBit/s VHT-MCS 4 80MHz VHT-NSS 1
	tx duration:	0 us
	rx bitrate:	468.0 MBit/s VHT-MCS 5 80MHz VHT-NSS 2
	rx duration:	0 us
	authorized:	yes
	authenticated:	yes
	associated:	yes
	preamble:	long
	WMM/WME:	yes
	MFP:		no
	TDLS peer:	no
	DTIM period:	1
	beacon interval:100
	short slot time:yes
	connected time:	256 seconds
	associated at [boottime]:	8723.567s
	associated at:	1752261006680 ms
	current time:	1752261262779 ms

[damian@uplink] ~  
❯ iw dev wlo1 info        
Interface wlo1
	ifindex 2
	wdev 0x1
	addr cc:6b:1e:32:45:01
	ssid VM0883623-5GHz
	type managed
	wiphy 0
	channel 44 (5220 MHz), width: 80 MHz, center1: 5210 MHz
	txpower 20.00 dBm
	multicast TXQ:
		qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
		0	0	0	0	0	0	0	0		0
```

dubhater, 2025-07-11:
Ah, okay. TDLS is apparently not the same thing as P2P. rtw88 in the kernel advertises support for TDLS.

Can you write down how to reproduce this problem step by step? What software to install, how to configure it, things like that.

> Weird thing is I never downloaded from this repository. I only got my networking drivers from the linux-firmware-realtek package on Arch. But now that I look, the repo's README says I can download the drivers using the linux-firmware meta package, which has linux-firmware-realtek. ChatGPT told me to turn off TDLS in modprobe using option disable_fwps=1, but it still does it and halts my connection. So now I'm not sure what to do.

The firmware and the driver are two different things. The driver is the rtw88 kernel modules, it runs on your computer. The firmware is a binary file, rtw8822c_fw.bin, it runs on the wifi chip's microcontroller.

rtw88 doesn't have an option called disable_fwps. Remember, those GPT type programs do not think, they just tell you things they read on the internet somewhere.

DamianBossPL, 2025-07-12:
I found another weird quirk to my issue. This problem occurs only under this very specific criteria:
- Both my laptop and phone must be connected to my home AP's 5GHz band.
- TDLS occurs between my laptop and phone
Otherwise, TDLS works completely fine.

I could use 2.4GHz on either device, and I'm fine with that. But I'm wondering if there's any solution to this madness. I checked my router settings and there doesn't seem to be anything that could relate.

DamianBossPL, 2025-07-13:
> So I tried this with my Moto E22. It didn't seem to work in any combination, but I didn't lose the connection.
> 
> Laptop 5 GHz, phone 5 GHz:
> 
> ```
> Jul 13 14:32:04 ideapad2 wpa_supplicant[1353]: TDLS: Creating peer entry for <Moto E22>
> Jul 13 14:32:05 ideapad2 wpa_supplicant[1353]: nl80211: kernel reports: key addition failed
> Jul 13 14:32:05 ideapad2 wpa_supplicant[1353]: TDLS: Failed to set TPK to the driver
> ```
> 
> Laptop 5 GHz, phone 2.4 GHz: Laptop 2.4 GHz, phone 2.4 GHz:
> 
> ```
> Jul 13 14:33:37 ideapad2 wpa_supplicant[1353]: TDLS: Creating peer entry for <Moto E22>
> Jul 13 14:33:42 ideapad2 wpa_supplicant[1353]: TDLS: Retrying sending of message (action_code=0)
> Jul 13 14:33:47 ideapad2 wpa_supplicant[1353]: TDLS: Retrying sending of message (action_code=0)
> Jul 13 14:33:52 ideapad2 wpa_supplicant[1353]: TDLS: Retrying sending of message (action_code=0)
> ```
> 
> Laptop 2.4 GHz, phone 5 GHz:
> 
> ```
> Jul 13 14:39:27 ideapad2 wpa_supplicant[1353]: TDLS: Creating peer entry for <Moto E22>
> Jul 13 14:39:27 ideapad2 wpa_supplicant[1353]: TDLS: Status code in TPK M2: 37
> ```
> 
> What do you see in journalctl when you try the tdls_setup?
> 
> Do I need to have some special software on my phone?

You don't need any special software on your phone, but we're getting closer and closer now. journalctl reported this with my criteria:
```
Jul 13 12:54:17 uplink wpa_supplicant[647]: TDLS: Creating peer entry for 02:51:1c:77:e6:06
Jul 13 12:54:18 uplink wpa_supplicant[647]: nl80211: kernel reports: key addition failed
Jul 13 12:54:18 uplink wpa_supplicant[647]: TDLS: Failed to set TPK to the driver
```

dubhater, 2025-07-13:
Maybe it's working, actually. When they're both on the 5 GHz network I get the "key addition failed" error, but ``iw dev wlp2s0 station dump`` shows the phone as a TDLS peer.

DamianBossPL, 2025-07-13:
> Maybe it's working, actually. When they're both on the 5 GHz network I get the "key addition failed" error, but `iw dev wlp2s0 station dump` shows the phone as a TDLS peer.

Same happens to me, it shows as a peer despite the error. Then I am able to remove this peer using `sudo wpa_cli tdls_teardown <MAC>` and network traffic halts, with no other error in the logs.

dubhater, 2025-07-13:
I get a disconnection after tdls_teardown, but only after a scan:

```
[2025-07-13 15:47:07.556918]: wlp2s0: new station <Moto E22>
[2025-07-13 15:47:12.605821]: wlp2s0: del station <Moto E22>
[2025-07-13 15:47:12.608676]: wlp2s0 (phy #6): scan started
[2025-07-13 15:47:16.372786]: wlp2s0 (phy #6): scan finished: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 2467 2472 5180 5200 5220 5240 5260 5280 5300 5320 5500 5520 5540 5560 5580 5600 5620 5640 5660 5680 5700 5720 5745 5765 5785 5805 5825, ""
[2025-07-13 15:47:16.436893]: wlp2s0: del station <AP>
[2025-07-13 15:47:16.437528]: wlp2s0 (phy #6): deauth <RTL8822CE> -> <AP> reason 4: Disassociated due to inactivity
[2025-07-13 15:47:16.437625]: wlp2s0 (phy #6): disconnected (local request) reason: 4: Disassociated due to inactivity
```

Is that how it looks in your case?